espressif/esp-idf #7491: Error: No such option: --ccache

ChrisIdema, 2021-08-30:
## Environment

- Development Kit:  na
- Kit version: na
- Module or chip used:  na
- IDF version : v4,3
- Build System:   idf.py
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows10
- environment type: PowerShell
- Using an IDE?: no
- Power Supply:        na

## Problem Description

https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-guides/build-system.html#idf-py-options
says --ccache option can speed up build time. Using the option gives an error. ccache is already installed by the esp installer: .espressif\tools\ccache

### Expected Behavior

Builds and rebuilds. Builds faster.

### Actual Behavior

idf.py build --ccache
Usage: idf.py build [OPTIONS]
Try 'idf.py build --help' for help.

Error: No such option: --ccache
### Steps to reproduce

1. idf.py build --ccache

igrr, 2021-08-30:
Hi @ChrisIdema, sorry that the docs don't make this clear enough. `--ccache` is a global option in idf.py, not an option of the `build` command (same as the other options listed there — -B, -C, -v)
That is, you should run it as `idf.py --ccache build`.

You can run `idf.py --help` to find out more about the available command line options.

By the way, on Windows `IDF_CCACHE_ENABLE` is already set in ESP-IDF command prompt by default, so you don't need to add idf.py `--ccache` argument to enable ccache.

ChrisIdema, 2021-08-30:
Thank you.

> By the way, on Windows `IDF_CCACHE_ENABLE` is already set in ESP-IDF command prompt by default, so you don't need to add idf.py `--ccache` argument to enable ccache.

You are right. I just checked it and the environment variable is set. This is odd since the documentation implies it is off by saying you can enable it: 
> --ccache flag can be used to enable CCache when compiling source files

closed 2021-08-30 by ChrisIdema