espressif/esp-idf #7492: can't definition 2 function with diferent parameters

hdpklm, 2021-08-30:
```
void intit() {}
void intit(int x) {}
```

```
../main/hello_world_main.c:17:6: error: redefinition of 'intit'
 void intit(int x) {}
      ^~~~~
../main/hello_world_main.c:16:6: note: previous definition of 'intit' was here
 void intit() {}
```

hdpklm, 2021-08-30:
and can't define a function with default parameters 

```
../main/hello_world_main.c:16:18: error: expected ';', ',' or ')' before '=' token
 void intit(int x = 0) {}
```

igrr, 2021-08-30:
Function overloading and default arguments are both features of C++, while your source file has `.c` extension.

Please rename your file to `hello_world_main.cpp` (and change the name in CMakeLists.txt accordingly), then these features will be working.

hdpklm, 2021-08-30:
Ok thank you so mutch

closed 2021-08-31 by Alvin1Zhang