espressif/esp-idf #7493: WebSocket Server Control Frames Bug

baldhead69, 2021-08-30:
Hi,

I am writing a websocket server and i discovered this option in "handle_ws_control_frames" in esp_http_server.h file.
/**
     * Flag indicating that control frames (PING, PONG, CLOSE) are also passed to the handler
     * This is used if a custom processing of the control frames is needed
     */
    bool handle_ws_control_frames; 

I enabled "handle_ws_control_frames" to receive PING frames from websockets clients.
The URI handler is being called when i get a ping from the websocket client, but an error occurs.

W (00:00:41.021) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
I (00:00:41.021) WSS_SERVER: httpd_ws_recv_frame failed with: ESP_FAIL

What am i doing wrong ?
I think there is a bug in websocket server.

URI handler register:
```c    
static const httpd_uri_t get_clients_message =
{
    .uri       = "/ws",  
    .method    = HTTP_GET,
    .handler   = get_clients_message_handler,                
    .is_websocket = true,
    .handle_ws_control_frames = true,
    .user_ctx  = NULL,
    .supported_subprotocol = NULL
};
```

URI HANDLER:
```c
static uint8_t buf[64] = { 0 };
static httpd_ws_frame_t ws_pkt;

static esp_err_t get_clients_message_handler(httpd_req_t *req)
{       
    ESP_LOGI(WSS_SERVER_TAG, "URI HANDLER:");
    ESP_LOGI(WSS_SERVER_TAG, "req->method = %d\n", req->method);

    if( req->method == HTTP_GET )
    {
        ESP_LOGI(WSS_SERVER_TAG, "Handshake done, the new connection was opened");        
        int sockfd = httpd_req_to_sockfd(req);
        ESP_LOGI(WSS_SERVER_TAG, "Handshake Socket: %d\n", sockfd);        
        return ESP_OK;
    }

    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_pkt.payload = buf;
 
    // This function return ESP_FAIL. 
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));   
    if( ret != ESP_OK )
    {   
        ESP_LOGI(WSS_SERVER_TAG, "httpd_ws_recv_frame failed with: %s\n", esp_err_to_name(ret));
        return ret;
    }
         
    if( ws_pkt.type == HTTPD_WS_TYPE_PING )
    {
        int sock = httpd_req_to_sockfd(req);
        ESP_LOGI(WSS_SERVER_TAG, "HTTPD_WS_TYPE_PING: %d\n", sock);        
        return ESP_OK;
    }
    return ESP_OK;
}
```

I tried the code below, based on "static esp_err_t ws_handler(httpd_req_t *req)" from "https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/wss_server/main/wss_server_example.c",
But no luck.
```c
I (00:00:12.750) esp_https_server: performing session handshake
I (00:00:14.725) WSS_SERVER: Handshake done, the new connection was opened
W (00:00:44.871) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
E (00:00:44.871) WSS_SERVER: httpd_ws_recv_frame failed to get frame len with -1
 ```
URI HANDLER CODE:
```c
static esp_err_t get_clients_message_handler(httpd_req_t *req)
{
    if (req->method == HTTP_GET) 
    {
        ESP_LOGI(WSS_SERVER_TAG, "Handshake done, the new connection was opened");
        return ESP_OK;
    }
    httpd_ws_frame_t ws_pkt1;
    uint8_t *buf = NULL;
    memset(&ws_pkt1, 0, sizeof(httpd_ws_frame_t));

    // First receive the full ws message
    /* Set max_len = 0 to get the frame len */
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt1, 0);
    if (ret != ESP_OK)
    {
        ESP_LOGE(WSS_SERVER_TAG, "httpd_ws_recv_frame failed to get frame len with %d", ret);
        return ret;
    }
    ESP_LOGI(WSS_SERVER_TAG, "frame len is %d", ws_pkt1.len);
    if (ws_pkt1.len)
    {
        /* ws_pkt.len + 1 is for NULL termination as we are expecting a string */
        buf = calloc(1, ws_pkt1.len + 1);
        if (buf == NULL)
        {
            ESP_LOGE(WSS_SERVER_TAG, "Failed to calloc memory for buf");
            return ESP_ERR_NO_MEM;
        }
        ws_pkt1.payload = buf;
        /* Set max_len = ws_pkt.len to get the frame payload */
        ret = httpd_ws_recv_frame(req, &ws_pkt1, ws_pkt1.len);
        if (ret != ESP_OK) 
        {
            ESP_LOGE(WSS_SERVER_TAG, "httpd_ws_recv_frame failed with %d", ret);
            free(buf);
            return ret;
        }
    }
    // If it was a PING
    if (ws_pkt1.type == HTTPD_WS_TYPE_PING)
    {
        ESP_LOGD(WSS_SERVER_TAG, "Received PING message");
        free(buf);
        int sockfd = httpd_req_to_sockfd(req); 
        ESP_LOGI(WSS_SERVER_TAG, "SOCKFD: %d", sockfd);        
        return ESP_OK;
    }    
    // If it was a BINARY message
    else if (ws_pkt1.type == HTTPD_WS_TYPE_BINARY)
    {
        int sock = httpd_req_to_sockfd(req);
        ESP_LOGI(WSS_SERVER_TAG, "HTTPD_WS_TYPE_BINARY: %d", sock);
        free(buf);
        return ESP_OK;
    }
    free(buf);
    return ESP_OK;
}
```

I thought the problem was because the client ping has no message, ie, payload size is zero, but i tested with payload and without payload and the error still exists.

I am using OKHTTP client library. 

I'm also running tests with the jmeter software.

baldhead69, 2021-08-30:
I took two tests(using ws and not wss):
* Send ping with payload and send ping without payload.

First test: Send ping with payload.

The websocket ping control frame call uri handler:
```c
I (00:00:28.062) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:28.070) WSS_SERVER: Handshake Socket: 55

I (00:00:28.162) WSS_SERVER: URI HANDLER:
I (00:00:28.166) WSS_SERVER: req->method = 0

W (00:00:28.172) httpd_txrx: httpd_sock_err: error in recv : 104
W (00:00:28.178) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
I (00:00:28.186) WSS_SERVER: httpd_ws_recv_frame failed with: ESP_FAIL
```
[image]
[image]

Second test: Send ping without payload.

The websocket ping control frame call uri handler:
```c
I (00:00:57.224) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:57.233) WSS_SERVER: Handshake Socket: 55

I (00:00:57.325) WSS_SERVER: URI HANDLER:
I (00:00:57.328) WSS_SERVER: req->method = 0

W (00:00:57.382) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
I (00:00:57.382) WSS_SERVER: httpd_ws_recv_frame failed with: ESP_FAIL
```

[image]
[image]

baldhead69, 2021-08-30:
I already took the opportunity to test the websocket close control frame.

The websocket close control frame dont call uri handler:
```c
I (00:00:23.257) WSS_SERVER: URI HANDLER:
I (00:00:23.258) WSS_SERVER: req->method = 1

I (00:00:23.258) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:23.267) WSS_SERVER: Handshake Socket: 55

I (00:00:23.359) WSS_SERVER: Closed socket handler:
I (00:00:23.363) WSS_SERVER: hd = 0x3ffd5348
I (00:00:23.369) WSS_SERVER: Closed socket = 55
```

[image]

ESP-YJM, 2021-09-01:
@baldhead69 Could you tell me the commit ID you used. I use master branch(b9ce1ed06dece87c591df6e22d73adb917fd7de3) and it is ok.

baldhead69, 2021-09-01:
> @baldhead69 Could you tell me the commit ID you used. I use master branch exampe ws_echo_server (commit id b9ce1ed06dece87c591df6e22d73adb917fd7de3) and it is ok.
> [image]
> [image]

v4.4-dev-2359-g58022f859

baldhead69, 2021-09-01:
v4.4-dev-2359-g58022f859

AxelLin, 2021-09-01:
> @baldhead69 Could you tell me the commit ID you used. I use master branch exampe ws_echo_server (commit id b9ce1ed06dece87c591df6e22d73adb917fd7de3) and it is ok.

commit b9ce1ed06dece87c591df6e22d73adb917fd7de3 does not exist in current master tree.

ESP-YJM, 2021-09-01:
> 
> 
> > @baldhead69 Could you tell me the commit ID you used. I use master branch exampe ws_echo_server (commit id b9ce1ed06dece87c591df6e22d73adb917fd7de3) and it is ok.
> 
> commit b9ce1ed06dece87c591df6e22d73adb917fd7de3 does not exist in current master tree.

Sorry for that my commit id is internal. Maybe not push to github now.

ESP-YJM, 2021-09-01:
> 
> 
> esp-idf version: v4.4-dev-2359-g58022f859
> 
> How can i check the commit ID ?

I test with your commit id, it is ok. Could you please try examples/protocols/http_server/ws_echo_server ？

baldhead69, 2021-09-01:
> I tried the code below, based on "static esp_err_t ws_handler(httpd_req_t *req)" from [...]

I tried it here, it's very similar, isn't it ?

ESP-YJM, 2021-09-01:
> I tried it here, it's very similar, isn't it ?

Yes, it's similar. Could you please open log debug from info to debug and send me the log.

baldhead69, 2021-09-01:
@ESP-YJM 

First test: 
handle_ws_control_frames = false
Ping with payload message

ESP32 LOG:
```c
I (00:00:06.847) MY_APP: MALLOC_CAP_INTERNAL = 76191
I (00:00:06.853) MY_APP: MALLOC_CAP_SPIRAM = 4132811

W (15839) wifi:<ba-add>idx:0 (ifx:0, 8c:de:f9:43:e2:27), tid:0, ssn:3, winSize:64
I (00:00:15.000) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:15.036) WSS_SERVER: Closed socket handler:
I (00:00:15.037) WSS_SERVER: hd = 0x3ffecbc0
I (00:00:15.038) WSS_SERVER: Closed socket = 55

I (00:00:15.044) WSS_SERVER: Trying to remove client from the list = 55
I (00:00:15.053) CLIENTS LIST: ******* CLIENTS LIST *******
I (00:00:15.060) CLIENTS LIST: Connected wsClients: 0
I (00:00:15.067) CLIENTS LIST: client[0] = no_socket
I (00:00:15.073) CLIENTS LIST: client[1] = no_socket
I (00:00:15.080) CLIENTS LIST: client[2] = no_socket
I (00:00:15.086) CLIENTS LIST: client[3] = no_socket
I (00:00:15.093) CLIENTS LIST: client[4] = no_socket
I (00:00:15.099) CLIENTS LIST: ****************************

I (00:00:15.107) WSS_SERVER: Set SO_LINGER success
I (00:00:15.113) WSS_SERVER: Closing Socket 55, Heap:4198335
```

[image]

[image]

Second test: 
handle_ws_control_frames = false
Ping WITHOUT payload message

ESP32 LOG:
```c
I (00:00:06.847) MY_APP: MALLOC_CAP_INTERNAL = 76215
I (00:00:06.853) MY_APP: MALLOC_CAP_SPIRAM = 4132931

W (20213) wifi:<ba-add>idx:0 (ifx:0, 8c:de:f9:43:e2:27), tid:0, ssn:2, winSize:64
I (00:00:19.237) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:19.255) WSS_SERVER: Closed socket handler:
I (00:00:19.256) WSS_SERVER: hd = 0x3ffecbc0
I (00:00:19.257) WSS_SERVER: Closed socket = 55

I (00:00:19.263) WSS_SERVER: Trying to remove client from the list = 55
I (00:00:19.271) CLIENTS LIST: ******* CLIENTS LIST *******
I (00:00:19.279) CLIENTS LIST: Connected wsClients: 0
I (00:00:19.285) CLIENTS LIST: client[0] = no_socket
I (00:00:19.292) CLIENTS LIST: client[1] = no_socket
I (00:00:19.299) CLIENTS LIST: client[2] = no_socket
I (00:00:19.305) CLIENTS LIST: client[3] = no_socket
I (00:00:19.312) CLIENTS LIST: client[4] = no_socket
I (00:00:19.318) CLIENTS LIST: ****************************

I (00:00:19.326) WSS_SERVER: Set SO_LINGER success
I (00:00:19.332) WSS_SERVER: Closing Socket 55, Heap:4198471
```

[image]

[image]

baldhead69, 2021-09-02:
Hi @ESP-YJM , 

I did some tests and verified that i am getting the pings in the uri handler now, but the payload message is coming corrupted.

When i send(client) a message in the ping, the payload len are correct but message is corrupted.
When i don't send any message in the ping, appears payload len with 8 elements and message corrupted.

In wireshark esp32 responds correctly to received pings. 
It seems to me that the problem is in the uri handler.

ESP-YJM, 2021-09-02:
@baldhead69 I can reproduce your issue. I not set `handle_ws_control_frames = true` before. And i set now, the phenomenon is the same as your. I debug it and find the PING data has been read and deal with internal in file `httpd_ws.c` function `httpd_ws_get_frame_type`. So although will call handler when receive PING, now the PING payload is empty, we will read fail. For solve this , you can try this temporary patch.
[1.zip](https://github.com/espressif/esp-idf/files/7095983/1.zip)

baldhead69, 2021-09-02:
@ESP-YJM ,

I dont know use git.

Do i need to change this manually or is there an automated way in git inside my esp-idf instalation to install this patch ?

Thank's.

baldhead69, 2021-09-02:
@ESP-YJM ,

I tried like this.

The first git apply command returned nothing.
I don't know if the changes were applied.

[image]

baldhead69, 2021-09-03:
@ESP-YJM 

The "control frame type CLOSE" is not calling the uri handler.

baldhead69, 2021-09-03:
> @baldhead69 I can reproduce your issue. I not set `handle_ws_control_frames = true` before. And i set now, the phenomenon is the same as your. I debug it and find the PING data has been read and [...]

@ESP-YJM

It is now working with my android app.

Thank`s.

baldhead69, 2021-09-03:
@ESP-YJM ,

Suggestion: https://github.com/espressif/esp-idf/issues/7027#issuecomment-905227857
Create a function that directly returns the payload size of a request for dynamic memory allocation.
size_t payload_size = httpd_ws_recv_frame_payload_size(req);

Can you help me with this feature request ?
https://github.com/espressif/esp-idf/issues/7431

Thank's

ESP-YJM, 2021-09-03:
> 
> 
> @ESP-YJM ,
> 
> I don't know how to use git.
> 
> Do i need to change this manually or is there an automated way in git inside my esp-idf installation to install this patch ?
> 
> Thank's.

I think you should merge it manually. For your feature request,  do follow code can't meet your requirement?
```
    httpd_ws_frame_t ws_pkt;    
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
           
    // Set max_len = 0 to get the frame len
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
    if (ret != ESP_OK)
    {
        ESP_LOGE(WSS_SERVER_TAG, "httpd_ws_recv_frame failed to get frame len with: %s\n", esp_err_to_name(ret));
        return ret;
    }
    
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("frame len is: %d\n"), ws_pkt.len);
```

baldhead69, 2021-09-03:
> I think you should merge it manually. For your feature request,  do follow code can't meet your requirement?
> ```
>     httpd_ws_frame_t ws_pkt;
>     memset(&ws_pkt, 0, [...]

@ESP-YJM ,

I tried this in discussion with david_cermak but dont worked.
I think that a directly function that return payload size are more elegant too.
This is only suggestion.

The feature resquest is this: #7431.

Thank's.

baldhead69, 2021-09-16:
Note to remember:

The close frame issue remains to be resolved.
The uri handler are not called when client send a close frame and the "background websocket layer" do not send a close message to do a clean close.

baldhead69, 2021-09-16:
If issue "uri handler are not called" is resolved, the user can send the close frame:

```c
#define normal_closure_code 1000  // indicates a normal closure, meaning that the purpose for which the connection was established has been fulfilled. 

static uint8_t buf[128] = { 0 };
static httpd_ws_frame_t ws_pkt;

static esp_err_t ws_uri_handler(httpd_req_t *req)
{       
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("URI HANDLER:"));
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("req->method = %d\n"), req->method);

    if( req->method == HTTP_GET )
    {
        ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Handshake done, the new connection was opened"));         
        return ESP_OK;
    }

    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_pkt.payload = buf;
  
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));   
    if( ret != ESP_OK )
    {
        ESP_LOGI(WSS_SERVER_TAG, LOG_USER("httpd_ws_recv_frame failed with: %s\n"), esp_err_to_name(ret));
        return ret;
    }

#if enable_handle_ws_control_frames
    if( ws_pkt.type == HTTPD_WS_TYPE_CLOSE )    
    {
        ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Got Control Frame Type: CLOSE"));
        int sockfd = httpd_req_to_sockfd(req);            
        ret = send_close_frame( normal_closure_code, "OK!!!", sockfd );  // send_close_frame( normal_closure, NULL, sockfd );
        ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Send Close Frame: %s\n"), esp_err_to_name(ret));
        return ret;
}                
#endif

    return ESP_OK;    
}

```

ESP-YJM, 2021-09-16:
@baldhead69 Thanks for reminding me. I think your example code can send close frame message when we receive close from peer. I also test it before which code framework is similar with yours. But i need evaluate where to place this code function, in internal or userspace handler callback. If in userspace, it will increase the amount of code development for users. So i choose to move it internal and user no need response control frame in user handle callback. I would like to hear your views. Feel free to share your idea.

baldhead69, 2021-09-16:
> @baldhead69 Thanks for reminding me. I think your example code can send close frame message when we receive close from peer. I also test it before which code framework is similar with yours. But i [...]

I think user need to read the close status code of client peer and can call a "internal" function and pass the close code("user", server), because there are several close status codes in rfc6455.

Something like that:
```c
    if( ws_pkt.type == HTTPD_WS_TYPE_CLOSE )    
    {
        ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Got Control Frame Type: CLOSE"));
        int sockfd = httpd_req_to_sockfd(req);

        if( received client close status code == 1000 )  // internal func. int16_t rcv_status code = get_client_close_status_code( );
        {    
            ret = send_close_frame( 1000, "OK!!!", sockfd );  // send_close_frame( normal_closure, NULL, sockfd );
            ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Send Close Frame: %s\n"), esp_err_to_name(ret));
            return ret;
        }          
```

Maksons, 2021-11-12:
Hi, I use websocket server much with:
- handle_ws_control_frames = true 
- with implemented keep alive based on sending PING and awaiting PONG responses and it works Ok.

From esp_http_server websocket server component I can see : 
`       ret = httpd_ws_get_frame_type(r);
        ESP_LOGD(TAG, LOG_FMT("New WS request from existing socket, ws_type=%d"), ra->ws_type);

        /*  Stop and return here immediately if it's a CLOSE frame */
        if (ra->ws_type == HTTPD_WS_TYPE_CLOSE) {
            sd->ws_close = true;
            return ret;
        }

        if (ra->ws_type == HTTPD_WS_TYPE_PONG) {
            /* Pass the PONG frames to the handler as well, as user app might send PINGs */
            ESP_LOGD(TAG, LOG_FMT("Received PONG frame"));
        }

        /* Call handler if it's a non-control frame (or if handler requests control frames, as well) */
        if (ret == ESP_OK &&
            (ra->ws_type < HTTPD_WS_TYPE_CLOSE || sd->ws_control_frames)) {
            ret = sd->ws_handler(r);
        }
`
From there I can see: 
HTTPD_WS_TYPE_CLOSE -  can not come to ws_handler , it's ok because we have server client close event ..
HTTPD_WS_TYPE_PONG  - condition is only for debug message output .
HTTPD_WS_TYPE_PONG and HTTPD_WS_TYPE_PING going to be passed to ws_handler if handle_ws_control_frames = true , also Ok.

!! BUT !!! 
 HTTPD_WS_TYPE_PING - content is received and is replied at begin of given code in "ret = httpd_ws_get_frame_type(r);" function in httpd_ws.c : 
`
    /* Reply to PING. For PONG and CLOSE, it will be handled elsewhere. */
    if(aux->ws_type == HTTPD_WS_TYPE_PING) {
        ESP_LOGD(TAG, LOG_FMT("Got a WS PING frame, Replying PONG..."));

        /* Read the rest of the PING frame, for PONG to reply back. */
        /* Please refer to RFC6455 Section 5.5.2 for more details */
        httpd_ws_frame_t frame;
        uint8_t frame_buf[128] = { 0 };
        memset(&frame, 0, sizeof(httpd_ws_frame_t));
        frame.payload = frame_buf;

        if(httpd_ws_recv_frame(req, &frame, 126) != ESP_OK) {
            ESP_LOGD(TAG, LOG_FMT("Cannot receive the full PING frame"));
            return ESP_ERR_INVALID_STATE;
        }

        /* Now turn the frame to PONG */
        frame.type = HTTPD_WS_TYPE_PONG;
        return httpd_ws_send_frame(req, &frame);
    }
`
For me it is correct behavior in global so PING and CLOSE handled by server and only PONG messages should be passed to user handler (when we speak about control frames)
`/* Pass the PONG frames to the handler as well, as user app might send PINGs */`

Another problem is about HTTPD_WS_TYPE_CLOSE !
It is the same as in #7572  , 
HTTPD_WS_TYPE_CLOSE is not handled anywhere .
But standard ( RFC 6455  5.5.1.  Close)  request: 
`
The Close frame MAY contain a body  ...
As the data is not guaranteed to be human readable, clients MUST NOT show it to end users.
...
If an endpoint receives a Close frame and did not previously send a Close frame, the endpoint MUST send a Close frame in response. 
`

baldhead69, 2021-11-12:
HTTPD_WS_TYPE_PONG and HTTPD_WS_TYPE_PING going to be passed to ws_handler if 'handle_ws_control_frames = true;' could be Ok but I think only PONG should be passed to ws_handler .

No, i use ping frame.
My keep alive is based on client send ping and i need reset a counter when client ping arise in server.

baldhead69, 2021-11-12:
@Maksons,

>For me it is correct behavior in global so PING and CLOSE handled by server and only PONG messages should be passed to >user handler (when we speak about control frames).

I don't agree, because in the rfc6455 the application can, if you want, respond with the reason for closing.

Maksons, 2021-11-14:
@baldhead69 
> No, i use ping frame. My keep alive is based on client send ping and i need reset a counter when client ping arise in server.

Oh, you can do so, but it's abnormal use case because : 
- you do not check bidirectional communication, but only relay on stable receive path nor know if client receive your reply 
- you lose additional information like delay in communication ( which is available in normal Ping-Pong, time sent/received)
- you can still see "PING" was received in current server implementation but without it content (content is sent back), do you really need PING content ?? 
- and one more negative (in my opinion) is that big clients like JavaScript in web browsers Chrome/Firefox/etc. do not support sending PING and seeing PONG/PING. In the same time JS websocket client automatically reply to server PING requests.
 
So, I highly recommend you to switch your server to normal PING/PONG implementation and in addition there is ready example of it : 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server/wss_server/main
it could be implemented same way on non ssl http server.

baldhead69, 2021-11-14:
@Maksons 

I implemented my keep alive before espressif team implemented this keep alive mechanism. 
For now i will use like this. 
I use a cellular app, so i have control over client and server aplication. 
If server wont respond with pong, client close connection too, after a programable time.

Maksons, 2021-11-14:
@baldhead69 

First of all, yes, I'd like to see reason/content of "Close" message too, but it is not mandatory.  

> I don't agree, because in the rfc6455 the application can, if you want, respond with the reason for closing.

Sorry, respond (reply) with reason , you mean different from received close reason??
I see in RFC 6455:
    (When sending a Close frame in response, the endpoint typically echos the status code it received.)

and also : 
    If a client and server both send a Close message at the same time, both endpoints will have sent and received a Close message and should consider the WebSocket connection closed and close the underlying TCP connection.
 
So I see cases:
- Sent our "Close" content is equal to later received "Close" reply content, normal close (and more data could arrive before "Close" reply). Content not interested because it is our one we sent. 
- Received "Close" with different content than sent out, content could be needed to see reason (client initialize close before receive our one). Could mean that some data packet could come to endpoint after it sent "Close".
- Received "Close" without any sent out, content also could be interested. But we should reply with same content ! Could mean that some data packet could come to endpoint after it sent "Close" too.

baldhead69, 2021-11-14:
@Maksons 

Server can init close connection, for whatever reason, so, the server can inform what close reason and in this way the client knows the reason for the server to close the connection.

baldhead69, 2021-11-21:
> @baldhead69 I can reproduce your issue. I not set `handle_ws_control_frames = true` before. And i set now, the phenomenon is the same as your. I debug it and find the PING data has been read and [...]

Hi @ESP-YJM and @david-cermak 

I installed a new esp-idf and the problem is the same.

Has this patch been merged ?

I am using esp-idf version: v4.4-dev-3657-gdf0929009a

Thank's.

ESP-YJM, 2021-11-22:
@baldhead69 Sorry, it has not been merged yet. I need to reorganize the code structure of this patch.

baldhead69, 2021-11-22:
@ESP-YJM ,

take the opportunity to fix the close frame too.

ESP-YJM, 2021-11-22:
@baldhead69 Thanks for reminding me that. We will consider that.

robert-alfaro, 2022-02-02:
I am using `ESP-IDF: v4.4-dev-2313-gc69f0ec329` and I determined the error is in `esp_http_server` component.  I fixed the issue by removing a return statement that doesn't allow the following conditional (checks if control frames are enabled for the URI handler) to execute. See diff below:
```diff
diff --git a/components/esp_http_server/src/httpd_parse.c b/components/esp_http_server/src/httpd_parse.c
index 9fe7907591..6f133902a6 100644
--- a/components/esp_http_server/src/httpd_parse.c
+++ b/components/esp_http_server/src/httpd_parse.c
@@ -756,10 +756,10 @@ esp_err_t httpd_req_new(struct httpd_data *hd, struct sock_db *sd)
         ret = httpd_ws_get_frame_type(r);
         ESP_LOGD(TAG, LOG_FMT("New WS request from existing socket, ws_type=%d"), ra->ws_type);
 
-        /*  Stop and return here immediately if it's a CLOSE frame */
+        /* If it's a CLOSE frame, mark it here. May be passed to handler if
+         * ws_control_frames is true. */
         if (ra->ws_type == HTTPD_WS_TYPE_CLOSE) {
             sd->ws_close = true;
-            return ret;
         }
 
         if (ra->ws_type == HTTPD_WS_TYPE_PONG) {
```

robert-alfaro, 2022-02-17:
Thank you @ESP-YJM ! (and thank you @ameyinamdar)

baldhead69, 2022-02-17:
The problem was solved ?

Incorporated the solution in the esp idf ?

ESP-YJM, 2022-02-17:
> The problem was solved ?
> 
> Incorporated the solution in the esp idf ?

Now callback handler function can be called by CLOSE frame. Userspace can know when to send CLOSE frame. But still not support parse PING frame payload data. I think this usage is not common, so will not implement it in this fix.

Spritetm, 2022-04-20:
> But still not support parse PING frame payload data. I think this usage is not common, so will not implement it in this fix.

If you do have an use case for this, suggest to open up a new issue describing where it occurs.

closed 2022-04-20 by Spritetm

baldhead69, 2022-04-20:
Payload is 0 bytes.

Spritetm, 2022-04-20:
Still, suggest you open a new issue rather than re-using this one.

baldhead69, 2022-04-20:
Why ?

The problem is the same when i open this issue.

See first post

baldhead69, 2022-04-20:
Maybe this patch was not applied in any esp-idf.

https://github.com/espressif/esp-idf/issues/7493#issuecomment-911143105

I will test this patch tomorrow.