espressif/esp-idf #7495: Custom BL

fbucafusco, 2021-08-30:
## Environment

- Development Kit:      [none]
- Module or chip used:  ESP32-WROOM-32
- IDF version ESP-IDF v4.3
- Build System:      idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    Windows
- (Windows only) environment type:ESP Command Prompt
- Using an IDE?: no
- Power Supply:  USB

## Problem Description

Hi!

I need to replace the bootloader to make some modifications that are needed in a project. 
According to this [link ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html#custom-bootloader)copying the folder bootloader from components to the user project folder, also under components should allow to modify the behavior for the bl. 

So I chose an example, I compile it without any mod, and was OK. Then, I copied the bootloader folder to example/components and tried to compile again. 

This is the error that throws

-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.peripherals.ld
CMake Error at components/bootloader/CMakeLists.txt:15 (esptool_py_flash_project_args):
  Unknown CMake command "esptool_py_flash_project_args".
-- Configuring incomplete, errors occurred!

### Code to reproduce this issue
example: esp-idf-master\examples\wifi\fast_scan
+ components/bootloader folder from esp-idf-master\components\bootloader
 
I tried also the custom BL from the [example](https://github.com/espressif/esp-idf/tree/master/examples/custom_bootloader/bootloader_override). It compiled OK, but then I realized that c files from bootloader_components/main folder are not being included in the compilation (I've inserted a couple of syntax errors in the c file, and all the project compile the same).  
 
Thanks in advance!

AxelLin, 2021-08-31:
> ## Environment
> * Development Kit:      [none]
> * Module or chip used:  ESP32-WROOM-32
> * IDF version ESP-IDF v4.3
> 
> ## Problem Description
> Hi!
> 
> I need to replace the bootloader to make some modifications that are needed in a project.
> According to this link titled [Custom bootloader](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html#custom-bootloader)copying the folder bootloader from components to the user project folder, also under components should allow to modify the behavior for the bl.

Above link is for master tree.
For v4.3, please reference https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-guides/bootloader.html#custom-bootloader

fbucafusco, 2021-08-31:
Hi! Thanks for you fast response. 
You are right ! I think I copied the wrong link to add to the issue.   However, the link you provided describes the copy/paste procedure that I've tried unsuccessfully.

ESP-Marius, 2022-08-31:
@fbucafusco Sorry for the late response.

I've just tested on v4.3 according to what the docs says (copy the whole component to `/your_project/components/bootloader/`) and everything seems OK. It compiles and flash the "custom" bootloader as expected.

Note that customizing the bootloader should be even easier on the latest IDF master, including two new examples showing two different approaches to achieving this: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html#custom-bootloader

ESP-Marius, 2022-09-08:
Closing this one due to lack of activity, but feel free to reopen if my answer didnt fully solve your issue.

closed 2022-09-08 by ESP-Marius