espressif/esp-idf #7496: Repeated ESP-NOW transmissions failure

ynormend, 2021-08-31:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used: ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.3-356-g48ae2309fd
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System: macOS
- (Windows only) environment type: -
- Using an IDE?: No
- Power Supply: USB

## Problem Description

When transmitting repeatedly ESP-NOW packets, the send callback is not called after 1 or 2 transmissions (ESP32-WROOM-32). 

### Expected Behavior

- When the periodic timer goes off, if the previous frame has been sent, `esp_now_send` is called and returns success
- The ESP-NOW packet is sent
- The ESP-NOW send callback is called

### Actual Behavior

The ESP-NOW send callback is not called after 1 or 2 transmissions.

### Steps to reproduce

- Connect to an access point
- Start a periodic timer with a 50-millisecond period
- In the timer callback, if the previous frame has been transmitted, send another frame
- Count the send and send callback calls

### Code to reproduce this issue

https://gist.github.com/ynormend/12204530fc7c5c5325212ed6e27d4701

## Debug Logs

```
I (28) boot: ESP-IDF v4.3-356-g48ae2309fd-dirty 2nd stage bootloader
I (29) boot: compile time 16:43:35
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=13868h ( 79976) map
I (141) esp_image: segment 1: paddr=00023890 vaddr=3ffb0000 size=0414ch ( 16716) load
I (148) esp_image: segment 2: paddr=000279e4 vaddr=40080000 size=08634h ( 34356) load
I (163) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6c328h (443176) map
I (323) esp_image: segment 4: paddr=0009c350 vaddr=40088634 size=0c4f4h ( 50420) load
I (344) esp_image: segment 5: paddr=000a884c vaddr=50000000 size=00010h (    16) load
I (355) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (366) cpu_start: Pro cpu up.
I (366) cpu_start: Starting app cpu, entry point is 0x40081220
0x40081220: call_start_cpu1 at /Users/yoan/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (381) cpu_start: Pro cpu start user code
I (381) cpu_start: cpu freq: 160000000
I (381) cpu_start: Application information:
I (385) cpu_start: Project name:     nami-esp-csi
I (391) cpu_start: App version:      2780407-dirty
I (396) cpu_start: Compile time:     Aug 31 2021 16:43:30
I (402) cpu_start: ELF file SHA256:  db0fcb032c994f7a...
I (408) cpu_start: ESP-IDF:          v4.3-356-g48ae2309fd-dirty
I (415) heap_init: Initializing. RAM available for dynamic allocation:
I (422) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (428) heap_init: At 3FFB7EE0 len 00028120 (160 KiB): DRAM
I (434) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (441) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (447) heap_init: At 40094B28 len 0000B4D8 (45 KiB): IRAM
I (454) spi_flash: detected chip: generic
I (458) spi_flash: flash io: dio
I (463) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (539) wifi:wifi driver task: 3ffc1960, prio:23, stack:6656, core=0
I (539) system_api: Base MAC address is not set
I (539) system_api: read default base MAC address from EFUSE
I (549) wifi:wifi firmware version: 42b8729
I (549) wifi:wifi certification version: v7.0
I (559) wifi:config NVS flash: enabled
I (559) wifi:config nano formating: disabled
I (559) wifi:Init data frame dynamic rx buffer num: 32
I (559) wifi:Init management frame dynamic rx buffer num: 32
I (569) wifi:Init management short buffer num: 32
I (569) wifi:Init dynamic tx buffer num: 32
I (579) wifi:Init static rx buffer size: 2212
I (579) wifi:Init static rx buffer num: 10
I (589) wifi:Init dynamic rx buffer num: 32
I (589) wifi_init: tcpip mbox: 32
I (589) wifi_init: udp mbox: 6
I (599) wifi_init: tcp mbox: 6
I (599) wifi_init: tcp tx win: 5744
I (609) wifi_init: tcp rx win: 5744
I (609) wifi_init: tcp mss: 1440
I (609) wifi_init: WiFi IRAM OP enabled
I (619) wifi_init: WiFi RX IRAM OP enabled
I (619) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (729) wifi:mode : sta (24:0a:c4:f5:4b:f8)
I (729) wifi:enable tsf
I (729) main: wifi started
I (739) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (739) wifi:state: init -> auth (b0)
I (749) wifi:state: auth -> assoc (0)
I (749) wifi:state: assoc -> run (10)
I (759) wifi:connected with nami, aid = 4, channel 11, BW20, bssid = f4:92:bf:84:23:54
I (759) wifi:security: WPA2-PSK, phy: bgn, rssi: -64
I (759) wifi:pm start, type: 1

I (769) main: Connected to nami (bssid: f4:92:bf:84:23:54, channel: 11)
I (769) esp_netif_handlers: sta ip: 192.168.1.6, mask: 255.255.255.0, gw: 192.168.1.1
I (779) main: got ip: 192.168.1.6
I (779) ESPNOW: espnow [version: 1.0] init
I (839) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (1789) main: send = 2, sent = 1, missed = 17
I (2789) main: send = 2, sent = 1, missed = 37
I (3789) main: send = 2, sent = 1, missed = 57
I (4789) main: send = 2, sent = 1, missed = 77
I (5789) main: send = 2, sent = 1, missed = 97
I (6789) main: send = 2, sent = 1, missed = 117
I (7789) main: send = 2, sent = 1, missed = 137
I (8789) main: send = 2, sent = 1, missed = 157
...
```

After the second transmission, the send callback was never called.

MaxwellAlan, 2021-09-06:
hi @ynormend thanks for reporting this issue, we are debuging now.
BTW, you can just call esp_wifi_set_ps(0) to disable modem sleep in your application to workaround for this issue.

jack0c, 2022-05-20:
@ynormend Can you try latest v4.3 b137ae42592ef67a4f145c3d067dafca0d6bf612 with esp_wifi_set_ps(1).

closed 2022-07-13 by Alvin1Zhang