espressif/esp-idf #7499: idf.py: command not found

mianqi2016, 2021-09-01:
----------------------------- Delete below -----------------------------

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [安信可ESP32-Audio-Kit音频开发板]
- Module or chip used:  [ESP32-A1S]
- IDF version (run ``git describe --tags`` to find it):esp-idf-v4.3
- Build System:         [CMake|idf.py]
- Compiler version (xtensa-esp32-elf-gcc --version: command not found)
- Operating System:     [Ubuntu 20.04]
- Using an IDE?: [Yes (eclipse 2020-12)]
- Power Supply:         [USB]

## Problem Description

//The board I tried was a demo board for Audio AI based on ESP32-A1S, since I found the software env was difficult to be setup, so I decided to try a lower step first - esp-idf, then step up to esp-adf. 

I installed esp-idf as it was told here(https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#get-started-get-prerequisites), but there were to many 
"Failed to clone  Retry scheduled
fatal:unable to access
gnutls_handshake() failed:", even when I tried in a some internet free time(half past 7 AM), so I thought to try eclipse with esp-idf Plugin.

I installed esp-idf plug-in and tools in eclipse,and tried the example "hellow_world", the button "Build the configurations of selected projects" was disable, while the button "Build All" was OK, I clicked "Build All" no error prompted, and I click "Run", it prompted "Serial port not found", so I think I should use "idf.py menuconfig", but when I typed "idf.py menuconfig" into terminal, it prompted "idf.py: command not found", although I could see "idf.py" in my "tools" directory and I have added both IED_PATH and idf.py PATH into .profile as it was told here(https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started-cmake/add-idf_path-to-profile.html) , I verified it by "echo $PATH".

Then I tried "cmake", while when I typed "cmake menuconfig", it was prompted:"The source directory "/home/mq/menuconfig" does not exist."

### Expected Behavior
I expect the menuconfig interface would shown, and I could config the serial port "ttyUSB0".

### Actual Behavior

when typed"idf.py menuconfig", it was prompted:
"idf.py: command not found"

when typed "cmake menuconfig", it was prompted:
"The source directory "/home/mq/menuconfig" does not exist."
""
### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.
x-special/nautilus-clipboard
copy
file:///home/mq/Pictures/Screenshot%20from%202021-09-01%2007-29-40.png

the screenshot is also here - https://ibb.co/wYNkgtW

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dobairoland, 2021-09-01:
Hi @mianqi2016. 

First of all, you are browsing the documentation for the master (v4.4) branch. This is the right page for your version:
https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/get-started/index.html#step-1-install-prerequisites
(different versions of ESP-IDF could have very different prerequisites)

Please don't update your PATH. The documentation page is for old version of ESP-IDF (v3.3). The `. export.sh` command outlined in the getting started guide should set your PATH. This way it won't be permanent and your global PATH won't be affected.

After you completed the steps of the getting started guide and run `. export.sh`, you should be able to run for example `idf.py --version`.

If you completed that successfully, you can move to the Eclipse setup guide: https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf and use the "To configure an existing ESP-IDF" step from the guide.

mianqi2016, 2021-09-01:
I tried this:https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/get-started/index.html#step-1-install-prerequisites
it was  failed, how can I install it in eclipse, I found use eclipse plugin more realistic。
https://ibb.co/yPh2NYG
https://ibb.co/Mgqrz2d

dobairoland, 2021-09-01:
Those messages are because of poor Internet connection. You can either retry many times or can download a ZIP-ed environment from the https://github.com/espressif/esp-idf/releases page, for example
https://github.com/espressif/esp-idf/releases/download/v4.3/esp-idf-v4.3.zip.

mianqi2016, 2021-09-02:
I saw lots of "Retry scheduled", will they be retried by the software automatically in the background, or should I manually retry by "git clone ..." again and again?

dobairoland, 2021-09-02:
> I saw lots of "Retry scheduled", will they be retried by the software automatically in the background, 

No, it won't be. 

> or should I manually retry by "git clone ..." again and again?

I'm not sure if that would ever be successful. I would suggest to download the ZIP or contact the company directly for an alternative way of getting ESP-IDF.

closed 2021-09-02 by dobairoland

Benedito821, 2023-01-31:
I am facing similar problem on windows 11. When I try to run the example advanced_https_ota idf.py command is not found.  On another hand idf.py runs properly when I use the blink example. I tried to run the export.bat file to set the env variables automatically but without any effect. Any tip @dobairoland ?

fjmsouza, 2024-07-25:
Before every _idf.py menuconfig_ try run:

```
cd ~/esp/esp-idf
./install.sh
```

More infos [here](https://docs.franzininho.com.br/docs/franzininho-wifi/exemplos-espidf/primeiros-passos/).