espressif/esp-idf #7502: Problem with external flash larger than 128Mbit

errorer, 2021-09-01:
Hello there,
I recently attached a second flash to ESP32's SPI port because the integrated flash size on ESP32-WROOM-32
 module can't meet my requirements. I am using `spi_bus_add_flash_device`, `esp_flash_init` and `esp_partition_register_external` and tried both littlefs and spiffs. Here is the initialization code (for littlefs):

`int littlefs_init(void)
{
    esp_err_t ret;
    size_t tb, ub;

    ESP_LOGI(TAG, "Initializing filesystem");

    if(spi_bus_initialize(VSPI_HOST, &bus_conf, 1) != ESP_OK)
    {
        ESP_LOGE(TAG, "SPI Flash port init failed");
        return -1;
    }

    if(spi_bus_add_flash_device(&ext_flash, &device_conf) != ESP_OK)
    {
        ESP_LOGE(TAG, "SPI bus add flash device failed");
        return -1;        
    }

    if(esp_flash_init(ext_flash) != ESP_OK)
    {
        ESP_LOGE(TAG, "SPI Flash initialize failed");
        return -1;  
    }

    uint32_t size;
    esp_flash_get_size(ext_flash, &size);

    ESP_LOGI(TAG, "External flash size: %d", ext_flash->size);
    ESP_LOGI(TAG, "External flash size(read): %d", size);

    uint32_t id;
    esp_flash_read_id(ext_flash, &id);
    ESP_LOGI(TAG, "External flash ID: 0x%08x", id);

    if(esp_partition_register_external( ext_flash, 
                                        0, 
                                        ext_flash->size, 
                                        littlefs_conf.partition_label, 
                                        ESP_PARTITION_TYPE_DATA,
                                        ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL) != ESP_OK)
    {
        ESP_LOGE(TAG, "SPI Flash data partition registration failed");
        return -1;      
    }

    ret = esp_vfs_littlefs_register(&littlefs_conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAG, "Failed to find LittleFS partition");
        } else {
            ESP_LOGE(TAG, "Failed to initialize LittleFS (%s)", esp_err_to_name(ret));
        }
        return -1;
    }
    esp_littlefs_info(littlefs_conf.partition_label, &tb, &ub);
    ESP_LOGI(TAG, "File system usage: %d/%d", ub, tb);
	return 0;
}`

The initialization is successful, and FS will format the flash for the first time. Writing and reading with small files have no problem. But when writing large files (eg, file exceeding 4MB), it seems the dir table will get corrupted and the whole file system will become unusable, triggering a 'mount fail' and formatting on the next initialization.
I'm guessing there's something wrong with esp_partition or esp_flash module, which will overwrite the flash from the beginning for large files(or high capacity usage) on larger flash chips. I've tried W25Q512 from Winbond and 25L256 from MXIC, both have the same issue. However the W25Q128 has no such problem.
The IDF version is release/4.2.
Anyone's help is appreciated, and I'll also look into the library to see if I can find any problem.
Thanks!

igrr, 2021-09-01:
Hi @errorer,

Initial support for external flash larger than 16 MB was added in https://github.com/espressif/esp-idf/commit/3bacf35310d5325d480972ccea9e29161e19861d, this commit is part of v4.3 release.

Recently support for large capacity GD flash chips was added, this will be part of the upcoming v4.4 release: https://github.com/espressif/esp-idf/commit/fecf27e54c318dc3bd2707486529a213ce8f6823.

Could you please upgrade your IDF copy to v4.3 or later, and try again with W25Q512?

errorer, 2021-09-02:
Hi @igrr 
Thank you a lot for your quick reply.
I've tried both of your suggestions, and here are the results:
Firstly I tried idf v4.3  3bacf35 as you suggested. It didn't work on MXIC 25L256. The file system gets corrupted after writing a 6MB file. I didn't test it with Winbond W25Q512.
Then I tried idf v4.4 fecf27e . When using  MXIC 25L256, there's a new warning output on idf monitor never seen before:
```
...
I (2981) spi_flash: detected chip: mxic
W (2981) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (2981) spi_flash: flash io: dio
I (2981) FS: External flash size: 33554432
I (2981) FS: External flash ID: 0x00c22019
...
```
And the file system still gets corrupted after writing a big file.
Then I tried Winbond W25Q512. This chip is little bit tricky. The JEDEC ID is incorrect. The last two hex digits are supposed to be 1Ah for a 512Mbit chip, but for this chip it's 20h. I guess the engineers at Winbond messed up hexadecimal with decimal.
[image]
As a result, the esp-flash driver can't recognize the capacity correctly. I have to manually set `ext_flash->size = 0x4000000;` for the chip to work. This time writing big files seems to have no problem. I've tried files as big as 8MB. Further tests still waiting to be done.
Since the IC supply chain is crazy these days, we can't stick on certain model of flash chip. I'll try other models we can get and give you more reports if possible.
Thanks again!

mythbuster5, 2021-09-02:
@errorer Thanks and Yes, you are right!

Since there are sooo many flash vendors in the world, each of them may have little difference behavior, it's hard for us to support all of the chips in our IDF repo. Even for same vendor, it's also hard for us to consider the circumstance for every single chip.

Therefore, we have a little example for users to configure/complement their own flash chips, see https://github.com/espressif/esp-idf/blob/master/examples/storage/custom_flash_driver/components/custom_chip_driver/spi_flash_chip_eon.c (and other files under this repo) as an example, change the functions there and give it the function pointer, our chip driver will auto detect the flash chip and behave in right way.

For mxic or winbond, maybe much easier, modify directly in their files if needed.

Alvin1Zhang, 2021-09-03:
Thanks for reporting and sharing updates, feel free to reopen.

closed 2021-09-03 by Alvin1Zhang

errorer, 2021-09-03:
@mythbuster5 @Alvin1Zhang Thank you all for your replies. I should have done more research before. Will also check out the custom flash driver. Thanks again.