espressif/esp-idf #7503: ESP32-C3 Wakeup Stub does not work on second wakeup

helmut64, 2021-09-01:
I have a wakeup_stub on the ESP32-C3, I simplified the code which does not work on the C3. Basically I like to re-enter after the deep sleep into deep sleep period again. This works after the first wakeup of my 10 secs specified wakeup time, it goes into the second deep sleep via the stub code. When the second wakeup expires after another 10 secs the ESP32-C3 boots without running into my wakeup_stub code.
It looks like something with the "REG_WRITE(RTC_ENTRY_ADDR_REG, (uint32_t)&wake_stub);" does not work.
Any ideas?
PS: The C3 RISC-V ESP is a great device.
```
static void RTC_IRAM_ATTR wake_stub()
{
	esp_default_wake_deep_sleep();
	
	REG_WRITE(TIMG_WDTFEED_REG(0), 1); // feed the watchdog

	SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE);

	uint64_t t_now = READ_PERI_REG(RTC_CNTL_TIME0_REG);
	t_now |= ((uint64_t) READ_PERI_REG(RTC_CNTL_TIME1_REG)) << 32;

	uint64_t t = t_now + _deepsleepTicks; // calculated in deep sleep to avoid 64 mul/divs here

	rtc_cntl_ll_set_wakeup_timer(t); // can be called because it is static inline (not in flash)
    
	// Set the pointer of the wake stub function.
	REG_WRITE(RTC_ENTRY_ADDR_REG, (uint32_t)&wake_stub);

	// Go to sleep.
	CLEAR_PERI_REG_MASK(RTC_CNTL_STATE0_REG, RTC_CNTL_SLEEP_EN);
	SET_PERI_REG_MASK(RTC_CNTL_STATE0_REG, RTC_CNTL_SLEEP_EN);
	// A few CPU cycles may be necessary for the sleep to start...
	while (true) {
		;
	}
	// never reaches here.
}
```
This is a simplified C3 wakeup stub, I use a different one for the ESP32 and ESP32-S2 which works, only the C3 does not work.

xiaojie76, 2022-03-28:
@helmut64  You can try to call function ‘set_rtc_memory_crc(); ’ before enter sleep.

helmut64, 2022-03-28:
Dear @xiaojie76, perfect this is the solution, I call the set_rtc_memory_crc() before entering into sleep within the wakeup stub. Now the wakeup stub works on C3, I tested it as well on the regular ESP32, perfect.
Thank you so much.
Helmut
PS: Should I close this issue.

xiaojie76, 2022-03-28:
@helmut64,  When chip wakes up from deep sleep, the RTC fast memory will be validated with CRC by RTC_MEMORY_CRC_REG. If the validation passes, the wake stubs code will be executed.  So need updating RTC_MEMORY_CRC_REG register via set_rtc_memory_crc(). 
But it confuses me why wakeup_stub works the first time if RTC_MEMORY_CRC_REG is not updated.

You can do this check on the original code (without set_rtc_memory_crc), before entering into sleep.
uint32_t crc = cacl_rtc_memory_crc(0, 0x7ff);
if (crc != REG_READ(RTC_MEMORY_CRC_REG)){
//crc validation failed
}

xiaojie76, 2022-03-29:
@helmut64 Oh, this is the wrong name defined in esp32c3/rom/rtc.h,  it should be “cacl_rtc_memory_crc”. You can temporarily use
extern uint32_t cacl_rtc_memory_crc(uint32_t start_addr, uint32_t crc_len);

helmut64, 2022-03-29:
@xiaojie76 cacl_rtc_memory_crc is still not working, even with the extern declaration. I have an idea about it:
I filed the original bug under the ESP32 Arduino support, it has been moved to IDF. I am using the Arduino ESP32 2.0.2 release and this has no such function in the IDF library.
Can you provide  a copy of the cacl_rtc_memory_crc code, then I can compile it locally (RTC_IRAM_ATTR) to do the test.

xiaojie76, 2022-03-29:
@helmut64  You can just check the value of RTC_MEMORY_CRC_REG, it should be 0 if set_rtc_memory_crc is not executed.
If needed,， the implementation cacl_rtc_memory_crc of c3 is as follows:
```c
uint32_t cacl_rtc_memory_crc(uint32_t start_addr, uint32_t crc_len)
{
        /* set rtc mem crc start addr */
    REG_SET_FIELD(SYSTEM_RTC_FASTMEM_CONFIG_REG, SYSTEM_RTC_MEM_CRC_ADDR, start_addr);
    /* set rtc mem crc len */
    REG_SET_FIELD(SYSTEM_RTC_FASTMEM_CONFIG_REG, SYSTEM_RTC_MEM_CRC_LEN, crc_len);

    /* start rtc mem crc */
    REG_SET_BIT(SYSTEM_RTC_FASTMEM_CONFIG_REG, SYSTEM_RTC_MEM_CRC_START);
    /* wait rtc mem crc finish */
    while(!REG_GET_BIT(SYSTEM_RTC_FASTMEM_CONFIG_REG, SYSTEM_RTC_MEM_CRC_FINISH));
    /* clear rtc mem crc start bit */
    REG_CLR_BIT(SYSTEM_RTC_FASTMEM_CONFIG_REG, SYSTEM_RTC_MEM_CRC_START);

    return REG_READ(SYSTEM_RTC_FASTMEM_CRC_REG);
}
```

helmut64, 2022-03-29:
@xiaojie76 I get a panic at:
REG_SET_BIT(SYSTEM_RTC_FASTMEM_CONFIG_REG, SYSTEM_RTC_MEM_CRC_START);
Is the start address really 0 and length 0x7ff?

xiaojie76, 2022-03-30:
@helmut64 yes, start address 0 and length 0x7ff, crc = cacl_rtc_memory_crc(0, 0x7ff); What problem do you meet?

xiaojie76, 2022-03-31:
@helmut64 I think it is not allowed to execute a program from rtc mem while calculating rtc mem crc. So you can not use RTC_IRAM_ATTR. 
Which version of idf are you using, check if there is a definition of cacl_rtc_memory_crc or calc_rtc_memory_crc in components/esp_rom/esp32c3/ld/esp32c3.rom.ld. If there is, extern this function should be able to find the definition.

helmut64, 2022-03-31:
@xiaojie76 I found the .ld file which is somehow now available in Arduino linking, I used a function pointer instead:
```
	uint32_t (*cacl_rtc_memory_crc)(uint32_t start_addr, uint32_t crc_len);
	cacl_rtc_memory_crc = (uint32_t(*)(uint32_t start_addr, uint32_t crc_len))0x40000038;
	
    static const char RTC_RODATA_ATTR mem_status_str[] = "cacl_rtc_memory_crc status: %d\n";
 	uint32_t crc = (*cacl_rtc_memory_crc)(0, 0x7ff);
	if (crc != REG_READ(RTC_MEMORY_CRC_REG)) {
		ets_printf(mem_status_str, -1);
	} else {
		ets_printf(mem_status_str, 0);
	}
	ets_delay_us(20000)
```
The result is that the CRC is wrong but it re-enters into a second deep sleep, after the second wakeup it wakes up without calling the stub code.
See my log from the Arduino console here (see cacl_rtc_memory_crc status):
```
cacl_rtc_memory_crc = 0x40000038;

10:41:39.260 -> ESP-ROM:esp32c3-api1-20210207
10:41:39.260 -> Build:Feb  7 2021
10:41:39.260 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
10:41:39.260 -> SPIWP:0xee
10:41:39.260 -> mode:DIO, clock div:1
10:41:39.260 -> load:0x3fcd6100,len:0x420
10:41:39.260 -> load:0x403ce000,len:0x90c
10:41:39.260 -> load:0x403d0000,len:0x236c
10:41:39.260 -> SHA-256 comparison failed:
10:41:39.298 -> Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71
10:41:39.298 -> Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7
10:41:39.298 -> Attempting to boot anyway...
10:41:39.298 -> entry 0x403ce000
10:41:39.437 -> 10:37:37.000042  Boot: CPU(Pro): Vbat power on reset
10:41:39.437 -> 10:37:37.000264  ESP32-C3: Rev: 3, 160 MHz, IDF(v4.4-beta1-189-ga79dc75f0a)
10:41:41.469 -> 10:37:39.024269  Entering deepsleep
10:41:41.469 -> E (4250) gpio: GPIO 9 does not support deep sleep wakeup
10:41:51.467 -> ESP-ROM:esp32c3-api1-20210207
10:41:51.505 -> Build:Feb  7 2021
10:41:51.505 -> rst:0x5 (DSLEEP),boot:0xc (SPI_FAST_FLASH_BOOT)
10:41:51.505 -> wakeup deepsleep gpio_low status: 0x33f340
10:41:51.505 -> re-enter into deepsleep
10:41:51.505 -> cacl_rtc_memory_crc status: -1
10:42:01.503 -> ESP-ROM:esp32c3-api1-20210207
10:42:01.541 -> Build:Feb  7 2021
10:42:01.541 -> rst:0x5 (DSLEEP),boot:0xc (SPI_FAST_FLASH_BOOT)
10:42:01.541 -> SPIWP:0xee
10:42:01.541 -> mode:DIO, clock div:1
10:42:01.541 -> load:0x3fcd6100,len:0x420
10:42:01.541 -> load:0x403ce000,len:0x90c
10:42:01.541 -> load:0x403d0000,len:0x236c
10:42:01.541 -> SHA-256 comparison failed:
10:42:01.541 -> Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71
10:42:01.541 -> Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7
10:42:01.541 -> Attempting to boot anyway...
10:42:01.541 -> entry 0x403ce000
10:42:01.727 -> 10:37:59.249373  Boot: Wakeup caused by timer (bootCount: 1)
10:42:01.727 -> 10:37:59.249620  Boot: CPU(Pro): Deep Sleep reset digital core
10:42:01.727 -> 10:37:59.249770  ESP32-C3: Rev: 3, 160 MHz, IDF(v4.4-beta1-189-ga79dc75f0a)
10:42:01.832 -> 10:37:59.362718  Entering deepsleep
```
Of course the set_rtc_memory_crc() overcomes this problem

xiaojie76, 2022-04-01:
@helmut64 So the first wakeup from deep sleep is not through wakeup stub? It seems I have always misunderstood. Then there is no doubt now. The crc check error will enter the normal boot process instead of wakeup stub, so you must set_rtc_memory_crc before using wakeup_stub, I think you can close this issue now. Thanks for your feedback.

helmut64, 2022-04-01:
@xiaojie76 perfect, the C3 is a very fine chip, I am looking forward to versions with more IOs. Thank you for your support.
Helmut

helmut64, 2022-04-03:
Maybe one last question, the wakeup stub wakeup us busy firmware message printing for about 15ms, I tried already GPIO 8 (high or low) at boot, but no difference.
Is there any way to avoid the 15ms wakeup busy duration.?

helmut64, 2022-04-03:
[image]

helmut64, 2022-04-03:
Green is the console TX write, yellow VDD, orange the wakeup stub setting GPIO 2 high

xiaojie76, 2022-04-04:
@helmut64 you can suppress rom printing with rtc_suppress_rom_log().

helmut64, 2022-04-04:
@xiaojie76 Thank you so much.

closed 2022-04-04 by helmut64