espressif/esp-idf #7510: Modbus RTU on, but TCP off fails

franz-ms-muc, 2021-09-03:
if i set: 
# CONFIG_FMB_COMM_MODE_TCP_EN is not set 
CONFIG_FMB_COMM_MODE_RTU_EN=y 
# CONFIG_FMB_COMM_MODE_ASCII_EN is not set 

i get this Error and more. 
it seems it still compiles in the Modbus TCP even if disabled. 

C:/Users/holznan/esp-idf-v4.3/components/freemodbus/tcp_slave/modbus_controller/mbc_tcp_slave.c: In function 'mbc_tcp_slave_destroy':
C:/Users/holznan/esp-idf-v4.3/components/freemodbus/tcp_slave/modbus_controller/mbc_tcp_slave.c:118:11: error: implicit declaration of function 'vMBTCPPortClose'; did you mean 'vMBPortClose'? [-Werror=implicit-function-declaration]
     (void)vMBTCPPortClose();
           ^~~~~~~~~~~~~~~
           vMBPortClose

alisitsyn, 2021-09-03:
Hello @franz-ms-muc ,

As I think it is known issue and was fixed in commit a20df743f1c51e6d65b021ed2ffd3081a2feec64. However the backport for this PR was not merged to v4.3 yet. Could you check if your code can be compiled correctly on latest master?

AxelLin, 2021-09-03:
@alisitsyn 
FYI: **v4.3.1** does **not** include the fix.

alisitsyn, 2021-09-03:
Yes, should be merged soon I think.

josesimoes, 2021-12-15:
> Yes, should be merged soon I think. Update: is planned to be merged in 4.3.2

Any date for 4.3.2?

alisitsyn, 2021-12-15:
@josesimoes, 
The fix for master has been merged in commit a20df743f1c51e6d65b021ed2ffd3081a2feec64
Backport v4.3 has been merged in commit 48997e82c00f745afa7da0b254511cb40d04de9b

josesimoes, 2021-12-15:
> @josesimoes, The fix for master has been merged in commit [a20df74](https://github.com/espressif/esp-idf/commit/a20df743f1c51e6d65b021ed2ffd3081a2feec64) Backport v4.3 has been merged in commit [48997e8](https://github.com/espressif/esp-idf/commit/48997e82c00f745afa7da0b254511cb40d04de9b)

Thanks for the confirmation. My question remains unanswered. As we're using github tags (and not the branch) for our CI-CD I would like to have an idea on when to expect  4.3.2. Thanks!

alisitsyn, 2021-12-15:
@josesimoes , The IDF v4.3.2 is ready for release today. However I don't know when it will be released.

Alvin1Zhang, 2021-12-16:
@josesimoes Thanks for sharing the updates, will update here once IDF v4.3.2 releases.

Alvin1Zhang, 2022-03-15:
Thanks for reporting, sorry for slow turnaround, IDF v4.3.2 has released https://github.com/espressif/esp-idf/releases/tag/v4.3.2, feel free to reopen if the issue still happens. Thanks.

closed 2022-03-15 by Alvin1Zhang