espressif/esp-idf #7512: Bluedroid 4.2 feature support KConfig has error, build fails as it cannot find common BT elements.

AliAkdurak, 2021-09-03:
## Environment

- Development Kit:      Custom Card,
- Kit version 
- Module or chip used:  ESP32-WROOM-32E
- IDF version:v4.4-dev-2915-gc8aab00fd
- Build System:idf.py
- Compiler version:xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:Linux
- Using an IDE?: Yes Eclipse

## Problem Description

Can't enable BLE_42_FEATURE_SUPPORT from menuconfig as bluedroid will not let me enable it under its menu. components/bt/host/bluedroid/Kconfig.in for the feature is as follow which doesn't allow for activation if target is not ESP32C3 or ESP32S3. I have been using all bluetooth functionality with ESP32-WROOM-32E until upgrade to a recent master branch requiring support for W5500

config BT_BLE_42_FEATURES_SUPPORTED
    bool "Enable BLE 4.2 features"
    depends on (BT_BLUEDROID_ENABLED && (IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3))
    default n
    help
        This enables BLE 4.2 features.

### Expected Behavior

Being able to enable BLE4.2 features. Our project has stopped being able to be built after upgrade to latest master branch. 

### Actual Behavior

No menu element will show for ESP32 target can't enable define BLE_42_FEATURE_SUPPORT.

### Steps to reproduce

I am not sure why my spesific project of ESP32-WROOM-32E does not BLE_42_FEATURE_SUPPORT. I am unable to offer suggestions for reproduce. I can build examples under the latest master branch.

### Code to reproduce this issue

Use anything from esp_gap_ble_api.h such as 

esp_ble_adv_data_t
esp_ble_adv_params_t

while not having enabled BLE_42_FEATURE_SUPPORT

## Debug Logs

## Other items if possible

AliAkdurak, 2021-09-03:
Problem can be solved with
`config BT_BLE_42_FEATURES_SUPPORTED
bool "Enable BLE 4.2 features"
depends on (BT_BLUEDROID_ENABLED && (IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3))
default n
help
This enables BLE 4.2 features.`

changing this to 
`config BT_BLE_42_FEATURES_SUPPORTED
bool "Enable BLE 4.2 features"
depends on (BT_BLUEDROID_ENABLED)
default n
help
This enables BLE 4.2 features.`

xiewenxiang, 2021-09-04:
If the chip is ESP32, There would be no configuration items `CONFIG_BT_BLE_50_FEATURES_SUPPORTED` and `CONFIG_BT_BLE_42_FEATURES_SUPPORTED`.

```c
#if (UC_BT_BLE_42_FEATURES_SUPPORTED == TRUE || BLE_50_FEATURE_SUPPORT == FALSE)
#define BLE_42_FEATURE_SUPPORT   TRUE
#else
#define BLE_42_FEATURE_SUPPORT   FALSE
#endif
```

Then, the BLE 4.2 macro will open, So, compilation errors will not occur.

AliAkdurak, 2021-09-05:
> If the chip is ESP32, There would be no configuration items `CONFIG_BT_BLE_50_FEATURES_SUPPORTED` and `CONFIG_BT_BLE_42_FEATURES_SUPPORTED`.
>
> ```c
> #if (UC_BT_BLE_42_FEATURES_SUPPORTED == TRUE || [...]

Is this already in the code or another nice solution to the problem my friend.

xiewenxiang, 2021-09-12:
@AliAkdurak 

Yes, It's already done in IDF. No matter which chip you use(ESP32 or ESP32C3), you won't meet the  compile issues.

If you do have this problem, please provide your `sdkconfig` file.

nicklasb, 2022-04-03:
Hi @xiewenxiang!
Sorry, but I am not sure I understand this one.
Is this a problem, fixed, or not?
I cannot get the ble50_security_server example to compile due to this.

nicklasb, 2022-04-03:
Hi again, solved, for me at least.

Basically, I hadn't been able to properly select the ESP32 S3 target in Platform IO....because it doesn't seem to support it.
Made it work using `idf.py set-target esp32s3` and not using Platform IO, only command line.

Only to realize I have one S3 too few to make my idea work anyway. :-)

(The gist of this thread is that only S3 and C3 supports BLE 5)