espressif/esp-idf #7513: increase FF_VOLUMES

chegewara, 2021-09-03:
I am playing a bit with USB host and MSC now and i think that `FF_VOLUMES` set to 2 is not enough. Can we increase it to at least 4?

atanisoft, 2021-09-03:
Perhaps move it to a KConfig setting?

chegewara, 2021-12-11:
@o-marshmallow  Ive found there is few issues related to `FF_VOLUMES > 2`. 
One was easy to find:
- this multi-partition array is still filled for 2 volumes only and should be expanded up to 10 elements, which means for volumes 2+ will cause memory corruption and crash; i confirmed it do the trick:
https://github.com/espressif/esp-idf/blob/master/components/fatfs/diskio/diskio.c#L21-L26

Second issue is harder to track. Long story short, for all volumes 2+ mutex in fs object is NULL: `fs->sobj`, which cause assert crash when freertos is trying to take that mutex:
```
assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

Backtrace:0x40023736:0x3ffc29900x40026cb5:0x3ffc29b0 0x4002c7e5:0x3ffc29d0 0x40027916:0x3ffc2af0 0x40089839:0x3ffc2b30 0x40087ac9:0x3ffc2b50 0x40087c04:0x3ffc2b70 0x40089303:0x3ffc2bc0 0x40086414:0x3ffc2c10 0x40086796:0x3ffc2c40 0x400a5727:0x3ffc2c80 0x400293a9:0x3ffc2ca0 
0x40023736: panic_abort at /home/chegewara/esp/master/components/esp_system/panic.c:403

0x40026cb5: esp_system_abort at /home/chegewara/esp/master/components/esp_system/esp_system.c:123

0x4002c7e5: __assert_func at /home/chegewara/esp/master/components/newlib/assert.c:85

0x40027916: xQueueSemaphoreTake at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/queue.c:1549 (discriminator 1)

0x40089839: ff_req_grant at /home/chegewara/esp/master/components/fatfs/port/freertos/ffsystem.c:91

0x40087ac9: lock_fs at /home/chegewara/esp/master/components/fatfs/src/ff.c:911

0x40087c04: find_volume at /home/chegewara/esp/master/components/fatfs/src/ff.c:3238

0x40089303: f_getfree at /home/chegewara/esp/master/components/fatfs/src/ff.c:4648
```

Thanks

chegewara, 2021-12-17:
The second one was my issue. I thought i can re-use the same `pdrv` and `f_mount` will take care of destroying old semaphore and create neww, but for some reason it is not working. Adding `f_mount(0, drv, 0)` in unmount procedure fixed the issue.

Thamks