espressif/esp-idf #7514: The select function times out too early

Emill, 2021-09-03:
The `select` function seems to round down the supplied timeout value to the granularity supported by the system. The POSIX standard says that the value should be rounded up instead. This makes timeouts time out too early with esp-idf. For example if you want to do something every 9 ms and the system's tick rate is set to 10 ms, then your loop will become a busyloop since 9 is rounded down to 0, making `select` return immediately.

Similar issue for `pthread_cond_timedwait`: https://github.com/espressif/esp-idf/issues/6901.

In components/vfs/vfs.c we see this:

```c
        TickType_t ticks_to_wait = portMAX_DELAY;
        if (timeout) {
            uint32_t timeout_ms = timeout->tv_sec * 1000 + timeout->tv_usec / 1000;
            ticks_to_wait = timeout_ms / portTICK_PERIOD_MS;
            ESP_LOGD(TAG, "timeout is %dms", timeout_ms);
        }
        ESP_LOGD(TAG, "waiting without calling socket_select");
        xSemaphoreTake(sel_sem.sem, ticks_to_wait);
```

Similar code can be found in components/lwip/port/esp32/freertos/sys_arch.c.

The code above clearly rounds down, while it should round up.

Sample code:

```c
    struct timespec ts1;
    clock_gettime(CLOCK_MONOTONIC, &ts1);

    int efd = eventfd(0, 0);

    fd_set s1;
    FD_ZERO(&s1);
    FD_SET(efd, &s1);

    struct timeval tv;
    tv.tv_sec = 0;
    tv.tv_usec = 9999;
    select(1, &s1, NULL, NULL, &tv);

    struct timespec ts2;
    clock_gettime(CLOCK_MONOTONIC, &ts2);

    printf("ts1: %d %d, ts2: %d %d, diff: %d ns\n", (int)ts1.tv_sec, (int)ts1.tv_nsec, (int)ts2.tv_sec, (int)ts2.tv_nsec,
            (int)(ts2.tv_sec * 1000000000ULL + ts2.tv_nsec - (ts1.tv_sec * 1000000000ULL + ts1.tv_nsec)));
```

Output:
```
ts1: 2 99243000, ts2: 2 99478000, diff: 235000 ns
```

To be POSIX compliant, `diff` should have been at least 9999000 ns.

o-marshmallow, 2021-09-29:
Hello @Emill ,
Thanks for your message, I am currently investigating and managing this issue.

o-marshmallow, 2021-11-01:
Hello @Emill ,
This bug has been fixed and merged internally, it will be synchronized to Github very soon.

closed 2021-11-03 by espressif-bot

Emill, 2021-11-03:
Thanks for fixing this!

But as mentioned in the topic description, there is a similar place here: https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/freertos/sys_arch.c#L187. This function is called by lwip_select here: https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/api/sockets.c#L2153, which is in turned called from esp_vfs_select here: https://github.com/espressif/esp-idf/blob/7bd93e3362e212b1e27fbd19a7a45cbe8e0b46ff/components/vfs/vfs.c#L1005

Could that result in the same issue as well?

o-marshmallow, 2021-11-04:
@Emill Indeed, thanks for pointing this out. It is a different (sub)project, so it will be part of another commit.

AxelLin, 2021-11-09:
> @Emill Indeed, thanks for pointing this out. It is a different (sub)project, so it will be part of another commit.

@o-marshmallow 
The fix for https://github.com/espressif/esp-idf/issues/7514#issuecomment-960240182 is in components/lwip/port/esp32/freertos/sys_arch.c which is part of esp-idf, which is **not** in sub project.

AxelLin, 2022-01-24:
> @Emill Indeed, thanks for pointing this out. It is a different (sub)project, so it will be part of another commit.

@o-marshmallow 
Just remind that the fix for https://github.com/espressif/esp-idf/issues/7514#issuecomment-960240182 is not yet available.