espressif/esp-idf #7515: Wifi Heap Error after Few hours of oprating

sowmimalu, 2021-09-03:
Hai,

I'm trying to figure out why the wifi driver is running out of memory after 3 hours .cant find where the log gets 

tried out by reinit wifi and reinitialized again and traced out my free heap size of application  ....but once this gets then frequently getting the wifi disconnects due to this error...

W (7535336) wifi: alloc eb len=36 type=3 fail, heap:3563020 
W (7535336) wifi: mem fail

here is my sdkconfig option for wifi :

#
# Wi-Fi
#
CONFIG_SW_COEXIST_ENABLE=y
CONFIG_SW_COEXIST_PREFERENCE_WIFI=
CONFIG_SW_COEXIST_PREFERENCE_BT=
CONFIG_SW_COEXIST_PREFERENCE_BALANCE=y
CONFIG_SW_COEXIST_PREFERENCE_VALUE=2
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=25
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=0
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_CSI_ENABLED=
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=6
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=16
CONFIG_ESP32_WIFI_NVS_ENABLED=y
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1=y
CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN=1024
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=32
CONFIG_ESP32_WIFI_DEBUG_LOG_ENABLE=
CONFIG_ESP32_WIFI_IRAM_OPT=y

SPIRAM config

CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y

if gets restarted again working fine then few hours later getting the same
how to resolve this issue ?

MaxwellAlan, 2021-09-06:
hi @sowmimalu , do you also use Bluetooth in your application?

sowmimalu, 2021-09-06:
Hai @MaxwellAlan 

yes using DUAL MODE both BLE and Bluetooth. and enabled BLUETOOTH AND wifi CO-existence option as well.

nishanth-radja, 2021-09-15:
Hi @sowmimalu,
In what security is the wifi connected in (wpa3 or wpa2)? any traffic is being run on the wifi and BT ? and which example are you using for this exp ?

sowmimalu, 2021-09-15:
hi @nishanth-radja  i am using BLE gattc client example and Wifi station example in my application and for wifi connection its with wpa2 security WIFI_AUTH_WPA2_PSK mode.

nishanth-radja, 2021-09-16:
@sowmimalu what is the IDF version you are using?

sowmimalu, 2021-09-16:
I am using idf 4.0

AxelLin, 2022-01-16:
> I am using idf 4.0 dev

The esp-idf-v4.0 is EOL. 
https://www.espressif.com/sites/default/files/advisory_downloads/AR2021-007%20End-of-Life%20Advisory%20for%20ESP-IDF%20v4.0%20Release%20Branch%20EN.pdf
Could you try reproduce the issue on newer esp-idf version?
Also please provide the exactly esp-idf version by running "git describe --tags".

sowmimalu, 2022-01-18:
Hi @AxelLin '
 i have checked in idf 4.3 , i am getting the error sometimes while downloading the file using http
W (86048) wifi:<ba-del>idx
W (86048) wifi:mem fail

AxelLin, 2022-02-06:
@sowmimalu 

So the symptom on v4.3 is different from what you initially reported on v4.0 (OOM after 3hrs)?
You probably need to create a new issue instead. (Is it possible to reproduce it with simple snip code?)

AxelLin, 2022-12-18:
@sowmimalu 
Do you still hit the "wifi: m f null" issue with latest 4.3.4 release?

closed 2024-02-27 by Alvin1Zhang