espressif/esp-idf #7520: Latest (V4.3) SmartConfig Example Not working with EspTouchV2 (Works with V1)

FalconBFR, 2021-09-04:
- Development Kit:      ESP32-Wrover-Kit
- Kit version:  v1
- Module or chip used:  ESP32-WROOM-32
- IDF version: ESP-IDF v4.3-dirty
- Build System:  CMake or idf.py???
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     macOS
- Using an IDE?: VSCode
- Power Supply:         USB

## Problem Description

Latest Smart Config Example contains code for support for V2 but only works with ESPTouchV1 and not V2

### Expected Behavior

SmartConfig example should work for both V2 and V1. 

### Actual Behavior

Connects to Wifi successfully when using EspTouchV1 on IOS iPhone 12.
ESP32 does not do anything upon activation of EspTouchV2 on IOS iPhone 12

### Steps to reproduce

1. Create a new project using the latest version of the smart config example
2. Build, Flash and Monitor and use the app as instructed by the documentation
3. Select EspTouch V2, leave AES Key as blank, enter the correct password and click enter

// If possible, attach a picture of your setup/wiring here.
[https://drive.google.com/file/d/13lFB0rF89w4OWkLqfJtbWJ7Rh7skkRn3/view?usp=sharing](https://drive.google.com/file/d/13lFB0rF89w4OWkLqfJtbWJ7Rh7skkRn3/view?usp=sharing)

### Code to reproduce this issue

The example as instructed above

## Debug Logs

### When using EspTouchV2
```
entry 0x400806b4
I (26) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (27) boot: compile time 21:18:55
D (27) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (33) boot: chip revision: 1
I (36) boot_comm: chip revision: 1, min. bootloader chip revision: 0
D (43) boot.esp32: magic e9
D (46) boot.esp32: segments 03
D (49) boot.esp32: spi_mode 02
D (53) boot.esp32: spi_speed 00
D (56) boot.esp32: spi_size 02
I (59) boot.esp32: SPI Speed      : 40MHz
I (64) boot.esp32: SPI Mode       : DIO
I (68) boot.esp32: SPI Flash Size : 4MB
D (73) boot: Enabling RTCWDT(9000 ms)
I (76) boot: Enabling RNG early entropy source...
D (82) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (91) boot: mapped partition table 0x8000 at 0x3f408000
D (96) flash_parts: partition table verified, 4 entries
I (101) boot: Partition Table:
I (105) boot: ## Label            Usage          Type ST Offset   Length
D (112) boot: load partition table entry 0x3f408000
D (117) boot: type=1 subtype=2
I (121) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (128) boot: load partition table entry 0x3f408020
D (133) boot: type=1 subtype=1
I (136) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (144) boot: load partition table entry 0x3f408040
D (149) boot: type=0 subtype=0
I (152) boot:  2 factory          factory app      00 00 00010000 00100000
[... 226 lines cut ...]
I (769673) smartconfig: Found channel on 4-0. Start to get ssid and password...
I (769673) smartconfig_example: Found channel
I (774183) smartconfig: T|pswd: koda2kko
I (774183) smartconfig: T|ssid: ETHomeMesh
I (774183) wifi:ic_disable_sniffer
I (774183) smartconfig_example: Got SSID and password
I (774183) smartconfig_example: SSID:ETHomeMesh
I (774193) smartconfig_example: PASSWORD:koda2kko
I (774223) wifi:new:<4,0>, old:<4,0>, ap:<255,255>, sta:<4,0>, prof:1
I (774223) wifi:state: init -> auth (b0)
I (774273) wifi:state: auth -> assoc (0)
I (774323) wifi:state: assoc -> run (10)
I (774373) wifi:connected with ETHomeMesh, aid = 3, channel 4, BW20, bssid = 50:64:2b:b6:1d:39
I (774373) wifi:security: WPA2-PSK, phy: bgn, rssi: -66
I (774383) wifi:pm start, type: 1

W (774413) wifi:<ba-add>idx:0 (ifx:0, 50:64:2b:b6:1d:39), tid:6, ssn:2, winSize:64
I (774433) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (775243) wifi:<ba-add>idx:1 (ifx:0, 50:64:2b:b6:1d:39), tid:0, ssn:5, winSize:64
I (775853) esp_netif_handlers: sta ip: 192.168.31.212, mask: 255.255.255.0, gw: 192.168.31.1
I (775853) smartconfig_example: WiFi Connected to ap
I (778903) smartconfig_example: smartconfig over
```

Any help will be appreciated! Thanks!

mhdong, 2021-09-06:
hello, when you Selected EspTouch V2, did you enter any reserved data?

FalconBFR, 2021-09-06:
Hi @mhdong! Thanks for the reply! Do you mean the custom data field provided by the demo app? If so, I tried entering both something and nothing as shown in the previously attached screenshot (reattached: https://drive.google.com/file/d/13lFB0rF89w4OWkLqfJtbWJ7Rh7skkRn3/view?usp=sharing)
If you mean anything on the code side, I haven't made any changes!

Thanks so much for your help in advance and I look forward to your reply!

FalconBFR, 2021-09-06:
Also, just for info, I left the AES key blank. Thanks!

mhdong, 2021-09-06:
hi @FalconBFR ,   I have tested that  SmartConfig example work for both V2 and V1 on android OPPO A83. 
When you selected EspTouch V2, did you set smartconfig type according in you application.
like this : esp_smartconfig_set_type(SC_TYPE_ESPTOUCH_V2).