espressif/esp-idf #7522: nvs err code 110c,how to fix it

qpmc1s, 2021-09-05:
typedef struct __my_dev_t
{
    uint8_t ass_mode ;

    float auto_humi01 ;
    float auto_speed01 ;
    int32_t auto_tim01 ;
    float auto_humi02 ;
    float auto_speed02 ;
    int32_t auto_tim02 ;

    struct {
        bool power_switch ;
        bool low_power_event_flag ;
        uint8_t pump01_switch ;
        uint8_t pump02_switch ;
        uint16_t run_mode ;
        uint16_t sub_dev_nbr ;
        float pump01_speed ;
        float pump02_speed ;
        float soil01_humidity ;
        float soil02_humidity ;
        float battery_level ;
    } ;
    
}my_dev_t ;

my_dev_t g_dev ;

read code:
    HAL_Kv_Get("HUM_01", &(g_dev.auto_humi01), &val_len) ;   
    HAL_Kv_Get("PUMP_01", &(g_dev.auto_speed01), &val_len) ;  
    HAL_Kv_Get("TIM_01", &(g_dev.auto_tim01), &val_len) ;
    HAL_Kv_Get("SWITCH_01", &(g_dev.pump01_switch), &val_len) ;
    HAL_Kv_Get("SPEE_01", &(g_dev.pump01_speed), &val_len) ;

write code:
    HAL_Kv_Set("HUM_01", &(g_dev.auto_humi01), sizeof(float), 0) ;
    HAL_Kv_Set("PUMP_01", &(g_dev.auto_speed01), sizeof(float), 0) ;
    HAL_Kv_Set("TIM_01", &(g_dev.auto_tim01), sizeof(uint32_t), 0) ;
    HAL_Kv_Set("SWITCH_01", &(g_dev.pump01_switch), sizeof(uint8_t), 0) ;
    HAL_Kv_Set("SPEE_01", &(g_dev.pump01_speed), sizeof(float), 0) ;

err:
W (808) wrapper_kv: nvs get blob HUM_01 failed with 110c
W (808) wrapper_kv: nvs get blob PUMP_01 failed with 110c
W (808) wrapper_kv: nvs get blob TIM_01 failed with 110c
W (818) wrapper_kv: nvs get blob SWITCH_01 failed with 110c

igrr, 2021-09-07:
Hi @qpmc1s, can you also post the source of HAL_Kv_Get, plus the rest of the information requested in the issue template you were presented with when opening the issue?

qpmc1s, 2021-09-07:
thanks！ here is the source code。 i used the function in my past project，and it always worked。however，recently ，its not stabilize ，sometimes I read the data succed，sometims I fail。when i call the function repeated like below it worked。i get the data finally。 and i don't know why 。

HAL_Kv_Get("HUM_01", &(g_dev.auto_humi01), &val_len) ;
HAL_Kv_Get("PUMP_01", &(g_dev.auto_speed01), &val_len) ;
HAL_Kv_Get("TIM_01", &(g_dev.auto_tim01), &val_len) ;
HAL_Kv_Get("SWITCH_01", &(g_dev.pump01_switch), &val_len) ;
HAL_Kv_Get("SPEE_01", &(g_dev.pump01_speed), &val_len) ;
**HAL_Kv_Get("HUM_01", &(g_dev.auto_humi01), &val_len) ;  
HAL_Kv_Get("PUMP_01", &(g_dev.auto_speed01), &val_len) ;
HAL_Kv_Get("TIM_01", &(g_dev.auto_tim01), &val_len) ;**

`int HAL_Kv_Get(const char *key, void *val, int *buffer_len)
{
    nvs_handle handle;
    esp_err_t ret;

    char key_name[16] = {0};

    if (key == NULL || val == NULL || buffer_len == NULL) {
        ESP_LOGE(TAG, "HAL_Kv_Get Null params");
        return ESP_FAIL;
    }

    if (HAL_Kv_Init() != ESP_OK) {
        return ESP_FAIL;
    }

    ret = nvs_open_from_partition(NVS_PARTITION_NAME, NVS_KV, NVS_READONLY, &handle);

    if (ret != ESP_OK) {
        ESP_LOGW(TAG, "nvs open %s failed with %x", NVS_KV, ret);
        return ESP_FAIL;
    }
    /*max key name is 15UL*/
    memcpy(key_name, key, sizeof(key_name) - 1);

    ret = nvs_get_blob(handle, key_name, val, (size_t *) buffer_len);

    if (ret != ESP_OK) {
        ESP_LOGW(TAG, "nvs get blob %s failed with %x", key_name, ret);
    }

    nvs_close(handle);

    return ret;
}

int HAL_Kv_Set(const char *key, const void *val, int len, int sync)
{
    nvs_handle handle;
    esp_err_t ret;

    char key_name[16] = {0};

    if (key == NULL || val == NULL || len <= 0) {
        ESP_LOGE(TAG, "HAL_Kv_Set NULL params");
        return ESP_FAIL;
    }

    if (HAL_Kv_Init() != ESP_OK) {
        return ESP_FAIL;
    }

    ret = nvs_open_from_partition(NVS_PARTITION_NAME, NVS_KV, NVS_READWRITE, &handle);

    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "nvs open %s failed with %x", NVS_KV, ret);
        return ESP_FAIL;
    }
    /*max key name is 15UL*/
    memcpy(key_name, key, sizeof(key_name) - 1);
    ESP_LOGE(TAG, "Set %s blob value", key_name);
    ret = nvs_set_blob(handle, key_name, val, len);

    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "nvs erase key %s failed with %x", key_name, ret);
    } else {
        nvs_commit(handle);
    }

    nvs_close(handle);

    return ret;
}`

igrr, 2021-09-07:
The last argument of nvs_get_blob has the following meaning:
- before the call, it should be set to the number of bytes available in the output buffer
- after the call, it gets set to the number of bytes written to the buffer (blob size)

In your snippet, you are reusing the same val_len variable for each call. So if one nvs_get_blob call sets val_len=10 (because the blob size was 10) the subsequent call with the same val_len will assume that there is only 10 bytes available in the output buffer, so the operation may fail if the blob is larger.

You need to do something along these lines:
```cpp
val_len = sizeof(g_dev.auto_humi01);
HAL_Kv_Get("HUM_01", &(g_dev.auto_humi01), &val_len);
val_len = sizeof(g_dev.auto_speed01);
HAL_Kv_Get("PUMP_01", &(g_dev.auto_speed01), &val_len);
```

qpmc1s, 2021-09-07:
thanks！ I‘ll test it later。thanks！

qpmc1s, 2021-09-08:
it works，thank you.

closed 2021-09-08 by qpmc1s