espressif/esp-idf #7524: Unable to build micropython env with C++ for ESP32

Friday8229, 2021-09-05:
## Environment

- Development Kit:   [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROVER]
- IDF version (v4.3):
- Build System:         [CMake]
- Compiler version (9.3.0):
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I am following [this](https://docs.micropython.org/en/latest/develop/cmodules.html) guide to compile micropython for esp32 board with custom C module. I can use the "make" command to build the basic example they provide but I can't build my own code. The cpp code works fine when build for the unix port but ESP32 port gives error. 

### Expected Behavior

A Clean Build 

### Actual Behavior
```

/tmp/ccetUhYg.s:529330: Error: operand 2 of 'l32r' has out of range value '4294120976'
/tmp/ccetUhYg.s:529340: Error: operand 2 of 'l32r' has out of range value '4294120964'
/tmp/ccetUhYg.s:529345: Error: operand 2 of 'l32r' has out of range value '4294120956'
/tmp/ccetUhYg.s:529366: Error: operand 2 of 'l32r' has out of range value '4294120924'
/tmp/ccetUhYg.s:529371: Error: operand 2 of 'l32r' has out of range value '4294120916'
/tmp/ccetUhYg.s:529381: Error: operand 2 of 'l32r' has out of range value '4294120904'
/tmp/ccetUhYg.s:529386: Error: operand 2 of 'l32r' has out of range value '4294120896'
/tmp/ccetUhYg.s:529396: Error: operand 2 of 'l32r' has out of range value '4294120880'
/tmp/ccetUhYg.s:529401: Error: operand 2 of 'l32r' has out of range value '4294120872'
/tmp/ccetUhYg.s:529412: Error: operand 2 of 'l32r' has out of range value '4294120856'
/tmp/ccetUhYg.s:529417: Error: operand 2 of 'l32r' has out of range value '4294120848'
/tmp/ccetUhYg.s:529427: Error: operand 2 of 'l32r' has out of range value '4294120836'
/tmp/ccetUhYg.s:529432: Error: operand 2 of 'l32r' has out of range value '4294120828'
/tmp/ccetUhYg.s:529440: Error: operand 2 of 'l32r' has out of range value '4294120812'
/tmp/ccetUhYg.s:529445: Error: operand 2 of 'l32r' has out of range value '4294120804'
/tmp/ccetUhYg.s:529455: Error: operand 2 of 'l32r' has out of range value '4294120792'
/tmp/ccetUhYg.s:529460: Error: operand 2 of 'l32r' has out of range value '4294120784'
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [Makefile:36: all] Error 2
```

### Steps to reproduce

1. make the micropython environment
2. cd to the ESP32 port 
3. make clean
4. make make USER_C_MODULES="Path to the cmake file"

### Code to reproduce this issue

[Here](https://drive.google.com/drive/folders/12hR6_e9kxTK3_Y4ppqIsFfAERjPK3WAn?usp=sharing) are the files, i uploaded it in my google drive as i was having issues with the preferred way.

```
I can't find any logs 
```

## Other items if possible
All the files I can provide are in the drive

igrr, 2021-09-07:
Have you tried asking this question in Micropython project issue tracker? Given the errors my guess is that some architecture specific compiler arguments are not being added when compiling C++ source files. This is something to troubleshoot in Micropython build system, unless you can reproduce the same issue with ESP-IDF, I'm afraid.

Friday8229, 2021-09-09:
Hi, i resolved this issue by breaking the loop in smaller pieces. I got a new error now, I can't mount SPIFFS.

The error that I'm getting

```
Terminal ready

>>> import cppexample
>>> cppexample.cppfunc()
E (32581) SPIFFS: mount failed, -10025
Failed to mount or format filesystem

abort() was called at PC 0x402ad94a on core 0

Backtrace:0x400912c6:0x3ffce020 0x40091989:0x3ffce040 0x400956ae:0x3ffce060 0x402ad94a:0x3ffce0d0 0x4039e8af:0x3ffce0f0 0x402a8c7b:0x3ffce390 0x402a87d5:0x3ffce3c0 0x402a810f:0x3ffce3f0 0x402b4fed:0x3ffce680 0x402bb0c5:0x3ffce6a0 0x402bb1f2:0x3ffce6c0 0x402bd2a9:0x3ffce6e0 0x402b51c0:0x3ffce780 0x402bb0c5:0x3ffce7d0 0x402bb0ee:0x3ffce7f0 0x402e730f:0x3ffce810 0x402e7661:0x3ffce8a0 0x402cc464:0x3ffce8e0

ELF file SHA256: 43f09eee77945144

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5572
load:0x40078000,len:12696
load:0x40080400,len:4292
entry 0x40080414
MicroPython v1.16-243-g8c4ba575f-dirty on 2021-09-10; ESP32 module with ESP32
Type "help()" for more information.

```

My code:
```
/**
 * @brief this function mount the virtual file system so we can able access the csv file we can need using the stream
 * 
 */
static void mount_file_system()
{
	esp_vfs_spiffs_conf_t conf = {.base_path = "/spiffs",
								  .partition_label = NULL,
								  .max_files = 5,
								  .format_if_mount_failed = false};

	// Use settings defined above to initialize and mount SPIFFS filesystem.

	esp_err_t ret = esp_vfs_spiffs_register(&conf);

	if (ret != ESP_OK)
	{
		if (ret == ESP_FAIL){
      std::string ab1 = "";
			ab1 = "Failed to mount or format filesystem";
      cout << ab1 << endl;
		}
		else if (ret == ESP_ERR_NOT_FOUND){
      std::string ab2 = "";
			ab2 = "Failed to find SPIFFS partition";
      cout << ab2 << endl;
		}
		else{
      std::string ab3 = "";
			ab3 = "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret);
      cout << ab3 << endl;
		}
		return;
	}

}

vector<vector<double>> read_csv(const char *filename)
{

	mount_file_system(); // mound the spiffs before using the input output stream

	ifstream read_file(filename);
	vector<vector<double>> csv_data;
	if (read_file.is_open())
	{

		std::string lineStr;
		while (getline(read_file, lineStr))
		{
			stringstream ss(lineStr);
			std::string d;
			vector<double> a_line_data;
			while (getline(ss, d, ','))
			{
				a_line_data.push_back(atof(d.c_str()));
			}
			csv_data.push_back(a_line_data);
		}
    std::string ab4 = "";
		ab4 = "rows of csv file:%d.\n", csv_data.size();
    cout << ab4 << endl;
	}
	esp_vfs_spiffs_unregister(NULL); // unmount the spiffs 
	return csv_data;
}
```

This  is how i read the file:
`vector<vector<double> > csv_data = read_csv("/spiffs/AA00068.csv");`

The [issue](https://github.com/micropython/micropython/issues/7705) that I opened from my side in the micropython repo

o-marshmallow, 2021-09-10:
Hello @Friday8229 ,
I see that there is an error when mounting the SPIFFS partition, error -10025 is `SPIFFS_ERR_NOT_A_FS`.
Have you specified your spiffs storage partition in the partition table for your chip? Have a look at `partitions_example.csv` file in `examples/storage/spiffsgen`. When you have such file, you also have to specify its path/name in the `menuconfig` (`Partition Table -->` submenu)
If you have activated encryption on your chip, you have to manually specify that spiffs partition should **not** be encrypted

Friday8229, 2021-09-10:
i have specified the name of the partition.csv file in menuconfig. Contents of my partition.csv are provided below.
```

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap,,,,
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 4M,
# created the spiffs partition for the storage of cvs file
storage,  data, spiffs,  ,        1M, 
```

This is what i get after using the make command, in case if that's helpful 

```
Project build complete. To flash, run this command:
/home/friday/.espressif/python_env/idf4.2_py3.8_env/bin/python ../../../components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build-GENERIC/bootloader/bootloader.bin 0x8000 build-GENERIC/partition_table/partition-table.bin 0x10000 build-GENERIC/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    22656  (    6016 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  3735840  (  458464 remaining)
total                  3801376

```

o-marshmallow, 2021-09-24:
@Friday8229 For some reasons, it seems like the spiffs storage partition is not taken into account. Here is the output I get when compiling an spiffs example:
```
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory ~/esp-idf/examples/storage/spiffsgen/build
Executing "ninja all"...
[28/969] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
storage,data,spiffs,0x110000,960K,
*******************************************************************************
```

I noticed that you said in the first message that you are using IDF version v4.3. But your previous message it shows that you are using IDF v4.2 tools (`idf4.2_py3.8_env`). It surely won't fix the problem you are getting but it is always a good habit to keep the tools synced

o-marshmallow, 2022-03-17:
Closing this issue for inactivity

closed 2022-03-17 by o-marshmallow