espressif/esp-idf #7525: [v4.4] [a2dp_sink and avrcp] Volume change notifications not coming through while a2dp sink stream is active

EricMiddleton, 2021-09-05:
In example project a2dp_sink, volume changes by host do not come through while an audio stream is active.

Repro steps:

- Connect to ESP32 running a2dp_sink example project with host device that supports avrcp volume (e.g. iPhone 12)
- Adjust volume and see log statements for each volume change on ESP32 device
- Start audio stream
- Adjust volume during stream and see no log statements indicating that volume has changed
- Stop audio stream
- See volume change notifications that were made during streaming all show up as soon as the stream is paused

This is a regression and was working as expected on previous version of esp-idf. I based a custom project on this example code and was working as expected (not sure the esp-idf version, but I started the project in March 2021). When I loaded the code on a new computer with new v4.4 esp-idf, this bug presented.

Log snippet from repro:

> _[--NOTE:_ iPhone connected--]
W (132258) BT_APPL: new conn_srvc id:19, app_id:1
I (132268) BT_AV: A2DP audio state: Started
I (132358) RCCT: AVRC event notification: 1
I (132358) BT_AV: Playback status changed: 0x1
I (132378) RCTG: AVRC set absolute volume: 11%
I (132378) RCTG: Volume is set by remote controller 11%
_[--Note:_ I'm adjusting the volume up and down on the iPhone while audio is streaming]
I (134918) BT_AV: Audio packet count 100
I (137238) BT_AV: Audio packet count 200
I (139568) BT_AV: Audio packet count 300
_[--Note:_ I paused the stream and now all of the volume changes I made earlier are coming through--]
I (141498) RCTG: AVRC set absolute volume: 18%
I (141498) RCTG: Volume is set by remote controller 18%
I (141498) RCTG: AVRC set absolute volume: 24%
I (141498) RCTG: Volume is set by remote controller 24%
I (141508) RCTG: AVRC set absolute volume: 30%
I (141508) RCTG: Volume is set by remote controller 30%
I (141518) RCTG: AVRC set absolute volume: 37%
I (141518) RCTG: Volume is set by remote controller 37%
I (141528) RCTG: AVRC set absolute volume: 43%
I (141538) RCTG: Volume is set by remote controller 43%
I (141538) RCTG: AVRC set absolute volume: 48%
I (141548) RCTG: Volume is set by remote controller 48%
I (141548) RCTG: AVRC set absolute volume: 55%
I (141558) RCTG: Volume is set by remote controller 55%
I (141558) RCTG: AVRC set absolute volume: 61%
I (141568) RCTG: Volume is set by remote controller 61%
I (141578) RCTG: AVRC set absolute volume: 67%
I (141578) RCTG: Volume is set by remote controller 67%
I (141588) RCTG: AVRC set absolute volume: 74%
I (141588) RCTG: Volume is set by remote controller 74%
I (141598) RCTG: AVRC set absolute volume: 80%
I (141598) RCTG: Volume is set by remote controller 80%
I (141608) RCTG: AVRC set absolute volume: 86%
I (141608) RCTG: Volume is set by remote controller 86%
I (141618) RCCT: AVRC event notification: 1
I (141618) BT_AV: Playback status changed: 0x2
I (141628) BT_AV: A2DP audio state: Suspended
I (144398) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode:2
_[--Note:_ Sometime later. Connected to Windows PC--]
I (167478) RCCT: AVRC conn_state evt: state 0, [00:00:00:00:00:00]
I (167488) RCTG: AVRC conn_state evt: state 0, [00:00:00:00:00:00]
I (167488) RCTG: Stop volume change simulation
I (167498) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode:0
E (189248) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (189258) BT_AV: A2DP connection state: Connecting, [18:ff:0f:33:a2:8e]
I (189288) BT_AV: A2DP audio stream configuration, codec type 0
I (189288) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, divb: 11
I (189298) BT_AV: Configure audio player 21-15-2-35
I (189298) BT_AV: Audio player configured, sample rate=44100
W (189328) BT_APPL: new conn_srvc id:19, app_id:0
I (189328) RCTG: AVRC conn_state evt: state 1, [18:ff:0f:33:a2:8e]
I (189338) RCTG: start volume change simulation
I (189338) BT_AV: A2DP connection state: Connected, [18:ff:0f:33:a2:8e]
I (189348) RCTG: AVRC register event notification: 13, param: 0x0
I (189348) RCCT: AVRC conn_state evt: state 1, [18:ff:0f:33:a2:8e]
I (189358) RCCT: AVRC remote features 24b, TG features 11
I (189358) RCTG: AVRC remote features 24b, CT features 2
I (189368) RCCT: remote rn_cap: count 6, bitmask 0xf06
I (189428) RCCT: AVRC metadata rsp: attribute id 0x1, Billie Jean
I (189428) RCCT: AVRC metadata rsp: attribute id 0x2, Michael Jackson
I (189428) RCCT: AVRC metadata rsp: attribute id 0x4, Thriller 25 Super Deluxe Edition
E (192548) BT_APPL: bta_dm_pm_btm_status hci_status=36
I (192548) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode:0
_[--Note:_ Adjusting volume before starting any streams--]
I (192718) RCTG: AVRC set absolute volume: 97%
I (192718) RCTG: Volume is set by remote controller 97%
I (193598) RCTG: AVRC set absolute volume: 96%
I (193598) RCTG: Volume is set by remote controller 96%
I (193818) RCTG: AVRC set absolute volume: 93%
I (193818) RCTG: Volume is set by remote controller 93%
I (193948) RCTG: AVRC set absolute volume: 92%
I (193948) RCTG: Volume is set by remote controller 92%
I (194148) RCTG: AVRC set absolute volume: 89%
I (194148) RCTG: Volume is set by remote controller 89%
I (194238) RCTG: AVRC set absolute volume: 88%
I (194238) RCTG: Volume is set by remote controller 88%
I (194348) RCTG: AVRC set absolute volume: 85%
I (194348) RCTG: Volume is set by remote controller 85%
I (194448) RCTG: AVRC set absolute volume: 84%
I (194448) RCTG: Volume is set by remote controller 84%
I (194458) RCTG: AVRC set absolute volume: 81%
I (194458) RCTG: Volume is set by remote controller 81%
I (194528) RCTG: AVRC set absolute volume: 80%
I (194528) RCTG: Volume is set by remote controller 80%
W (204858) BT_BTC: handle_rc_notification_rsp unsupported notification on CT, event id 0xf
I (206288) RCCT: AVRC event notification: 1
I (206288) BT_AV: Playback status changed: 0x1
I (206338) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b
W (206338) BT_APPL: new conn_srvc id:19, app_id:1
I (206338) BT_AV: A2DP audio state: Started
_[--Note:_ I'm adjusting the volume during audio streaming--]
I (208998) BT_AV: Audio packet count 100
I (211318) BT_AV: Audio packet count 200
I (213638) BT_AV: Audio packet count 300
I (215968) BT_AV: Audio packet count 400
I (218288) BT_AV: Audio packet count 500
I (220608) BT_AV: Audio packet count 600
I (222928) BT_AV: Audio packet count 700
I (225258) BT_AV: Audio packet count 800
I (227578) BT_AV: Audio packet count 900
I (229898) BT_AV: Audio packet count 1000
I (232228) BT_AV: Audio packet count 1100
_[--Note:_ As soon as stream is closed, some of the volume changes I had made are coming through--]
I (232808) RCTG: AVRC set absolute volume: 77%
I (232808) RCTG: Volume is set by remote controller 77%
I (232808) RCTG: AVRC set absolute volume: 85%
I (232808) RCTG: Volume is set by remote controller 85%
I (232818) RCTG: AVRC set absolute volume: 74%
I (232818) RCTG: Volume is set by remote controller 74%
I (232828) RCTG: AVRC set absolute volume: 32%
I (232828) RCTG: Volume is set by remote controller 32%
I (232838) RCTG: AVRC set absolute volume: 44%
I (232848) RCTG: Volume is set by remote controller 44%
I (232848) RCCT: AVRC event notification: 1
I (232858) BT_AV: Playback status changed: 0x2
I (232858) BT_AV: A2DP audio state: Suspended
E (235728) BT_APPL: bta_dm_pm_btm_status hci_status=31
I (235728) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode:0

Wth-Esp, 2021-09-06:
Hi，Erric

Could you please provide the commit information?

Thanks

EricMiddleton, 2021-09-06:
Can you clarify what you mean by “commit information”? This repros on release v4.4 with the stock example project

BetterJincheng, 2021-09-06:
Hi, @EricMiddleton
 
There is no such a bug on branch 'master'. Please use 'master' instead of 'v4.4-dev'.

Thanks

EricMiddleton, 2021-09-06:
Confirmed that the bug does not exist on master. Thanks @BetterJincheng !

closed 2021-09-06 by EricMiddleton