espressif/esp-idf #7527: MODBUS Serial example not working

rjrajbir, 2021-09-06:
Hi newbie here,

I am working on MODBUS Serial example of esp-idf,I simply just connected the hardware as shown in the example.

ESP32-1(Master) and ESP32-2(Slave) by using two TTL to RS485 converters.

But I am not able to get the expected output. I am gettinfg the 0x108 Err that is invalid slave response.

Master output:
`
E (65579) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (65579) MASTER_TEST: Characteristic #0 (Data_channel_0) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (65599) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE). 
E (65609) MASTER_TEST: Characteristic #1 (Humidity_1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (65619) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (65629) MASTER_TEST: Characteristic #2 (Temperature_1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (65649) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE). 
E (65649) MASTER_TEST: Characteristic #3 (Humidity_2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (65669) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (65679) MASTER_TEST: Characteristic #4 (Temperature_2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (65699) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (65699) MASTER_TEST: Characteristic #5 (Humidity_3) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (65719) MB_CONTROLLER_MASTER: mbc_master_get_parameter(111): SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (65729) MASTER_TEST: Characteristic #6 (Test_regs) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
I (65739) MASTER_TEST: Characteristic #7 RelayP1 (on/off) value = OFF (0x55) read successful.
I (65749) MASTER_TEST: Characteristic #8 RelayP2 (on/off) value = OFF (0xaa) read successful.
E (67749) MASTER_TEST: Alarm is not triggered after 30 retries.
I (67749) MASTER_TEST: Destroy master...`

Slave Response:
`[0;32mI (97984) SLAVE_TEST: COILS READ (97681730 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (97994) SLAVE_TEST: COILS READ (97690818 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (100154) SLAVE_TEST: COILS READ (99851729 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (100164) SLAVE_TEST: COILS READ (99860956 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (102324) SLAVE_TEST: COILS READ (102021659 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (102334) SLAVE_TEST: COILS READ (102030886 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (104494) SLAVE_TEST: COILS READ (104191526 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (104504) SLAVE_TEST: COILS READ (104200746 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (106664) SLAVE_TEST: COILS READ (106361451 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (106674) SLAVE_TEST: COILS READ (106370676 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (108834) SLAVE_TEST: COILS READ (108532558 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (108844) SLAVE_TEST: COILS READ (108541925 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (111004) SLAVE_TEST: COILS READ (110702488 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (111014) SLAVE_TEST: COILS READ (110711785 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (113174) SLAVE_TEST: COILS READ (112872418 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (113184) SLAVE_TEST: COILS READ (112881716 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (115344) SLAVE_TEST: COILS READ (115042417 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (115354) SLAVE_TEST: COILS READ (115051715 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (117514) SLAVE_TEST: COILS READ (117212208 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (117524) SLAVE_TEST: COILS READ (117221581 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (119684) SLAVE_TEST: COILS READ (119382138 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (119694) SLAVE_TEST: COILS READ (119391435 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (121854) SLAVE_TEST: COILS READ (121552068 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (121864) SLAVE_TEST: COILS READ (121561365 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (124024) SLAVE_TEST: COILS READ (123722067 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (124034) SLAVE_TEST: COILS READ (123731364 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (126194) SLAVE_TEST: COILS READ (125891858 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (126204) SLAVE_TEST: COILS READ (125901224 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (128364) SLAVE_TEST: COILS READ (128061788 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (128374) SLAVE_TEST: COILS READ (128071085 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (130534) SLAVE_TEST: COILS READ (130231718 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (130544) SLAVE_TEST: COILS READ (130241014 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (132704) SLAVE_TEST: COILS READ (132401717 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (132714) SLAVE_TEST: COILS READ (132411014 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (134874) SLAVE_TEST: COILS READ (134571511 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (134884) SLAVE_TEST: COILS READ (134580874 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (137044) SLAVE_TEST: COILS READ (136741438 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (137054) SLAVE_TEST: COILS READ (136750734 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (139214) SLAVE_TEST: COILS READ (138911367 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (139224) SLAVE_TEST: COILS READ (138920664 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[... 3 lines cut ...]
[0;32mI (143564) SLAVE_TEST: COILS READ (143260524 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (145724) SLAVE_TEST: COILS READ (145421087 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (145734) SLAVE_TEST: COILS READ (145430384 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (147894) SLAVE_TEST: COILS READ (147591017 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (147904) SLAVE_TEST: COILS READ (147600314 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (150064) SLAVE_TEST: COILS READ (149761016 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (150074) SLAVE_TEST: COILS READ (149770313 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (152234) SLAVE_TEST: COILS READ (151930807 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (152244) SLAVE_TEST: COILS READ (151940174 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (154404) SLAVE_TEST: COILS READ (154100737 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (154414) SLAVE_TEST: COILS READ (154110034 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (156574) SLAVE_TEST: COILS READ (156270667 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (156584) SLAVE_TEST: COILS READ (156279964 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (158744) SLAVE_TEST: COILS READ (158440666 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (158754) SLAVE_TEST: COILS READ (158449963 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (160914) SLAVE_TEST: COILS READ (160610527 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (160924) SLAVE_TEST: COILS READ (160619893 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[0;32mI (163084) SLAVE_TEST: COILS READ (162780387 us), ADDR:0, TYPE:32, INST_ADDR:0x3ffb26a4, SIZE:8[0m
[0;32mI (163094) SLAVE_TEST: COILS READ (162789685 us), ADDR:8, TYPE:32, INST_ADDR:0x3ffb26a5, SIZE:8[0m
[image]
`

Please help to solve this issue.

Thank you in advance.

alisitsyn, 2021-09-06:
Hi @rjrajbir ,

Your connection looks correct but we need to check if your RS485 tranceivers can work reliable. There may different types of chips with different voltage (3.3V or 5V). Which one do you have on the boards?

1. I would recommend you to check your connection using 
[rs485 echo example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485)

2. Do you have any change in the  
[master Data Dictionary](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_master/main/master.c#L88) or  [area descriptors of slave](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_slave/main/slave.c#L118) ?
According to log your slave device do not recognize requests from master to read holding registers and input registers.

Please also increase the response time in the master configuration to = 400 ms instead of 150 ms as on the picture below and recompile.
`cd $IDF_PATH
idf.py menuconfig
idf.py fullclean
idf.py build
idf.py -p PORT -b 921600 flash monitor
`
[image]

Let me know if you have any issues according to instruction above.

rjrajbir, 2021-09-06:
Thank you for the reply, i have tried this but haven't got the results.
I think its the  esp-idf version issue, I have installed new version i.e. 4.3 in my another PC and its working fine with the same setup
and not working with the 4.2.2 version.

alisitsyn, 2021-09-06:
I can check if the v4.2.2 has issues but recommend you to work with the latest master.

alisitsyn, 2021-09-06:
Please also take a look to extended documentation for API usage which recently merged in commit 2146b5b291e1b067908998f83cf563aa220eadb7.

rjrajbir, 2021-09-06:
I have made some codes with 4.2.2 so i have integrate it with them.
Could you please tell me the codes made in previous versions can work in the new verions or not.

alisitsyn, 2021-09-06:
Yes. They should work just fine without changes.

rjrajbir, 2021-09-07:
Yes, I recommend to check the issue, because maxium work has been done on 4.2,2

alisitsyn, 2021-09-07:
@rjrajbir ,
As I described above I have checked the v4.2.2 tag and it works just fine. I guess you have some preliminary commit of release v4.2 and this is why the bug is there. Please use the latest official releases. You do not need to update your Modbus application.

rjrajbir, 2021-09-07:
Ok Thank you for the help.
I want to make the cpp lib of MODBUS SERIAL example. But the code you made is so complicated. Can you please give me simple version of modbus serial.

alisitsyn, 2021-09-07:
@rjrajbir ,

Please consider the [simple Modbus master code](https://github.com/alisitsyn/modbus_support/tree/mb_support/simple_modbus_master).
This example is just to read/write couple of registers in slave.

Let me know if you need more help.

rjrajbir, 2021-09-08:
Thank you for the help, i'll check and get back to you

rjrajbir, 2021-09-08:
[master.zip](https://github.com/espressif/esp-idf/files/7126550/master.zip)
Here is my code but its showing many errors, Please help

alisitsyn, 2021-09-08:
Your Arduino library duplicates the basic functionality of low layer ESP_Modbus stack. The ESP_Modbus API is extended library compare to Arduino and they are very different.
What is your goal? Do you want to update the code to do the same readings/writings with the ESP_Modbus library instead?
Please provide more information or questions to explain what is your project. For now I can give you the code which will be able to read/write any registers of any slave using the ESP_Modbus. Actually it is provided in the simple Modbus example above and can be used together with your other CPP code. If absolutely required the CPP interface wrapper class can be created for ESP_Modbus.

rjrajbir, 2021-09-08:
Actually i want to create a c++ wrapper for esp-idf modbus-serial example. I am new with c++.
That its simple for user to just call the function for reading and writing different registers in modbus.

alisitsyn, 2021-09-08:
I propose you to start this and learn how to use ESP_Modbus and CPP. The functionality of the basic Modbus read/write can be implemented using the `mbc_serial_master_send_request()` function call of the library.
My preliminary instructions on how to start from scratch to create wrapper are below:
1. Use your ModbusMaster class template to modify. Include the required header files of ESP_Modbus. The ESP_Modbus headers include the required CPP guards.
2. Its begin method will get two parameters: 1: address of slave being accessed 2: mb_communication_info_t type as second parameter. The method saves the communication info and address of slave into corresponding class properties for future use in read/write methods.
This method calls the initialization functions similar to what is done in `master_init()` function of the example and uses saved communication parameters.
3. The corresponded methods `ModbusMaster::readCoils`, `ModbusMaster::readHoldingRegisters`, `ModbusMaster::readInputRegisters` of the class shall be translated to calls of 
[mbc_serial_master_send_request function](https://github.com/espressif/esp-idf/blob/master/components/freemodbus/serial_master/modbus_controller/mbc_serial_master.c#L174) with prefilled `mb_param_request_t` structure. It uses the saved slave address property. 
The input parameters: register offset to read/write, number of registers, pointer to data buffer to read/write to/from the slave.

See the [example on how to fill the request structure](https://github.com/espressif/esp-idf/blob/master/components/freemodbus/serial_master/modbus_controller/mbc_serial_master.c#L370)

Each method handles the Modbus transaction reads/writes data to/from buffer provided as data pointer and returns the error code to application.

See the [simple Modbus master code](https://github.com/alisitsyn/modbus_support/tree/mb_support/simple_modbus_master)

4. The destroy method of the class to free all resources and destroy all active objects. See the example to implement.

The [API usage information](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/modbus.html?highlight=modbus)

rjrajbir, 2021-09-13:
Hi, Sorry for the late reply, but have made just simple code to test but I am getting some errors.

`too many initializers for 'mb_parameter_opt_t'` this error in below section
`
const mb_parameter_descriptor_t device_parameters[500] = {
    // { CID, Param Name, Units, Modbus Slave Addr, Modbus Reg Type, Reg Start, Reg Size, Instance Offset, Data Type, Data Size, Parameter Options, Access Mode}
    { CID_INP_DATA_0, STR("Data_channel_0"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 0, 2,
                    INPUT_OFFSET(input_data0), PARAM_TYPE_FLOAT, 4, OPTS( -10, 10, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_0, STR("Humidity_1"), STR("%rH"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 0, 2,
            HOLD_OFFSET(holding_data0), PARAM_TYPE_FLOAT, 4, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_INP_DATA_1, STR("Temperature_1"), STR("C"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 2, 2,
            INPUT_OFFSET(input_data1), PARAM_TYPE_FLOAT, 4, OPTS( -40, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_1, STR("Humidity_2"), STR("%rH"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 2, 2,
            HOLD_OFFSET(holding_data1), PARAM_TYPE_FLOAT, 4, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_INP_DATA_2, STR("Temperature_2"), STR("C"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 4, 2,
            INPUT_OFFSET(input_data2), PARAM_TYPE_FLOAT, 4, OPTS( -40, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_DATA_2, STR("Humidity_3"), STR("%rH"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 4, 2,
            HOLD_OFFSET(holding_data2), PARAM_TYPE_FLOAT, 4, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_HOLD_TEST_REG, STR("Test_regs"), STR("__"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING, 10, 58,
            HOLD_OFFSET(test_regs), PARAM_TYPE_ASCII, 116, OPTS( 0, 100, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_RELAY_P1, STR("RelayP1"), STR("on/off"), MB_DEVICE_ADDR1, MB_PARAM_COIL, 0, 8,
            COIL_OFFSET(coils_port0), PARAM_TYPE_U16, 2, OPTS( BIT1, 0, 0 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_RELAY_P2, STR("RelayP2"), STR("on/off"), MB_DEVICE_ADDR1, MB_PARAM_COIL, 8, 8,
            COIL_OFFSET(coils_port1), PARAM_TYPE_U16, 2, OPTS( BIT0, 0, 0 ), PAR_PERMS_READ_WRITE_TRIGGER }
};
`
Please help me solve this, and I would be great if you could make a simple c++ template for modbus, so that i can use it complete the wrapper.

rjrajbir, 2021-09-13:
I am confused with your code, its just too complicated

alisitsyn, 2021-09-15:
Hi,

> Please help me solve this.

Please use the modified master code to check  it under CPP:
[master.zip](https://github.com/espressif/esp-idf/files/7170949/master.zip)

> I would be great if you could make a simple c++ template for modbus, so that i can use it complete the wrapper.

The wrapper the data dictionary is not required because the function `mbc_master_send_request()` does not use this.
I will send you the wrapper according to my plans.

rjrajbir, 2021-10-01:
Hi, Sorry for the late reply, I was busy with something else, By the way thank you so much for the wrapper, I just checked it, Further I am also planning for slave CPP wrapper, If you can help with that, please do reply, I will be so thankful to you for this.

rjrajbir, 2021-10-04:
Awaiting for the response.

alisitsyn, 2021-10-05:
Hi,

Sorry for delay. The code you provided has very different interface and the wrapper for slave shall have different interface. 

Some preliminary comments are below:

1. slave.setBaud() -> [setup slave](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_slave/main/slave.c#L110)
2. regBank.add() method can be changed to call of [mbc_slave_set_descriptor() function](https://github.com/espressif/esp-idf/blob/master/components/freemodbus/common/esp_modbus_slave.c#L207) which will define whole addressable register area in one call.
3. `regBank.get(reg_addr)` to find the instance address which corresponds to register address  and return corresponded value of register from register area defined on previous step.
4. The `regBank.set(reg_addr, value)` will find the instance address which corresponds to register address  and then set the data of register to the value. The [critical section](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_slave/main/slave.c#L192) is required to set the data in the register area.
5.   [the code here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_slave/main/slave.c#L176) can be used to block task until slave registers are not accessed.

Try to implement the slave wrapper by yourself. I will help once possible.

rjrajbir, 2021-10-06:
Thank you so much for your help, I will try to do this, but I am also waiting from your version of code

alisitsyn, 2021-10-12:
Sorry for the late reply. Do not have update for now.

rjrajbir, 2021-10-13:
Alright, Thank you still waiting

alisitsyn, 2021-10-19:
Hi,

The [ModbusSlave wrapper update here](https://github.com/alisitsyn/modbus_support/tree/mb_support/modbus_add_cpp_wrapper/modbus_simple_slave).

rjrajbir, 2022-02-11:
Which energy meter are you using and could you please share the errors and code

HEMILSHAH007, 2022-02-11:
SELEC MFM384-R-C energy meter 
and I am using a simple Modbus master code for reading INPUT REGISTERS

CODE:-
#include "mbcontroller.h"   // for common Modbus defines
#include "string.h"
#include "esp_log.h"

#define MASTER_MAX_CIDS 2
#define MASTER_MAX_RETRY 100
#define MASTER_PORT_NUM 2
#define MASTER_SPEED 115200
#define MASTER_TAG "MODBUS_MASTER"
#define MB_UART_RXD_PIN 16
#define MB_UART_TXD_PIN 17
#define MB_UART_RTS_PIN 18

#define MASTER_CHECK(a, ret_val, str, ...) \
    if (!(a)) { \
        ESP_LOGE(MASTER_TAG, "%s(%u): " str, __FUNCTION__, __LINE__, ##__VA_ARGS__); \
        return (ret_val); \
    }

#define STR(fieldname) ((const char*)( fieldname ))
#define OPTS(min_val, max_val, step_val) { .opt1 = min_val, .opt2 = max_val, .opt3 = step_val }

// Enumeration of modbus slave addresses accessed by master device
enum {
    MB_DEVICE_ADDR1 = 1
};

// Enumeration of all supported CIDs for device (used in parameter definition table)
enum {
    CID_DEV_REG0 = 0,
    CID_DEV_REG1
};

extern "C" {

// Example Data (Object) Dictionary for Modbus parameters
const mb_parameter_descriptor_t device_parameters[2] = {
    // CID, Name, Units, Modbus addr, register type, Modbus Reg Start Addr, Modbus Reg read length, 
    // Instance offset (NA), Instance type, Instance length (bytes), Options (NA), Permissions
    { CID_DEV_REG0, STR("MB_input_reg-0"), STR("Data"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 0, 1,
                    0, PARAM_TYPE_U16, 2, OPTS( 0,0,0 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_DEV_REG1, STR("MB_input_reg-1"), STR("Data"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 2, 1,
                    0, PARAM_TYPE_U16, 2, OPTS( 0,0,0 ), PAR_PERMS_READ_WRITE_TRIGGER }
};

// Calculate number of parameters in the table
const uint16_t num_device_parameters = (sizeof(device_parameters)/sizeof(device_parameters[0]));

static esp_err_t read_modbus_parameter(uint16_t cid, uint16_t *par_data)
{
    const mb_parameter_descriptor_t* param_descriptor = NULL;
    
    esp_err_t err = mbc_master_get_cid_info(cid, &param_descriptor);
    if ((err != ESP_ERR_NOT_FOUND) && (param_descriptor != NULL)) {
        uint8_t type = 0;
        err = mbc_master_get_parameter(cid, (char*)param_descriptor->param_key,
                                                        (uint8_t*)par_data, &type);
        if (err == ESP_OK) {
            ESP_LOGI(MASTER_TAG, "Characteristic #%d %s (%s) value = (0x%x) parameter read successful.",
                                     param_descriptor->cid,
                                     (char*)param_descriptor->param_key,
                                     (char*)param_descriptor->param_units,
                                     *(uint16_t*)par_data);
        }
        else 
        {
[... 107 lines cut ...]
    return err;
}

void app_main()
{
    // Initialization of device peripheral and objects
    ESP_ERROR_CHECK(master_init());
    vTaskDelay(10);
    
    // Write registers to predefined state

    write_modbus_parameter(CID_DEV_REG0, &register_data);
    
    write_modbus_parameter(CID_DEV_REG1, &register_data);

    master_read_write_func(NULL);
}

}

ERRORS:-
 SERIAL master get parameter failure error=(0x108) (ESP_ERR_INVALID_RESPONSE)
Bunch of this type of errors coming

can you pls suggest any code I am new to this...Thanks in advance

rjrajbir, 2022-02-11:
You shold use this [https://github.com/alisitsyn/modbus_support/tree/mb_support/simple_modbus_master](code) to test and at first check if the data is coming on [https://www.win-tech.com/demos/modscan64.zip](MODSACN) Software.

HEMILSHAH007, 2022-02-12:
ANYONE can help ??

HEMILSHAH007, 2022-02-12:
Anyone can help???

alisitsyn, 2022-02-15:
@HEMILSHAH007 ,

The [simple modbus example](https://github.com/alisitsyn/modbus_support/tree/mb_support/simple_modbus_master) demonstrates the read/write of modbus holding registers. In your case you are going to access INPUT registers in read only area. You can use  `read_modbus_parameter(CID_DEV_REGX, &register_data);` function to read data in the `static void master_read_write_func(void *arg);` function and remove all calls like `write_modbus_parameter(CID_DEV_REGX, &register_data);` from your code. This should help to avoid errors reading the registers if you configured the data dictionary for your device correctly. You need the Modbus registers mapping table for your device to configure the data dictionary accordingly. Please refer to [ESP_Modbus official documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/modbus.html#messaging-model-and-data-mapping) for more information on how to do this. If you still have issues with configuration and data access please provide the data mapping for your device and I can help you to configure it.

Thanks.

HEMILSHAH007, 2022-02-15:
@alisitsyn Thanks for the reply 
Are you asking about this mapping table ??
[image]
[image]
pls help me regarding this I am totally new to this and very keen to work on this more now...

alisitsyn, 2022-02-15:
@HEMILSHAH007,

Please use the data dictionary configured as below in [simple modbus master](https://github.com/alisitsyn/modbus_support/blob/mb_support/simple_modbus_master/main/master.c#L29):
``` CPP

enum {
    CID_V_1ST_PHASE = 0,
    CID_V_2ND_PHASE,
    CID_V_3RD_PHASE,
    CID_V_AVERAGE_LN,
    CID_V12,
    CID_V23,
    CID_V31,
    CID_V_AVERAGE_LL,
    CID_CURRENT_I1,
    CID_CURRENT_I2,
    CID_CURRENT_I3,
    CID_CURRENT_AVERAGE,
    CID_KW1,
    CID_KW2,
    CID_KW3,
    CID_HOLD_TEST_REG,
    CID_COUNT
};

const mb_parameter_descriptor_t device_parameters[] = {
    // { CID, Param Name, Units, Modbus Slave Addr, Modbus Reg Type, Reg Start, Reg Size, Instance Offset, Data Type, Data Size, Parameter Options, Access Mode}
    { CID_V_1ST_PHASE, STR("Voltage first phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 0, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_2ND_PHASE, STR("Voltage second phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 2, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_3RD_PHASE, STR("Voltage third phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 4, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_AVERAGE_LN, STR("Average voltage LN"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 6, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V12, STR("Voltage_V12"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 8, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V23, STR("Voltage_V23"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 10, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V31, STR("Voltage_V31"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 12, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_AVERAGE_LL, STR("Voltage_AVERAGE_LL"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 14, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_CURRENT_I1, STR("Current_I1"), STR("A"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 16, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_CURRENT_I2, STR("Current_I2"), STR("A"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 18, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_CURRENT_AVERAGE, STR("Current_Average"), STR("A"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 22, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER } 
};

```
Then you can read each parameter using the function:
```
static esp_err_t read_modbus_parameter(uint16_t cid, uint32_t par_data)
{
const mb_parameter_descriptor_t param_descriptor = NULL;

esp_err_t err = mbc_master_get_cid_info(cid, &param_descriptor);
if ((err != ESP_ERR_NOT_FOUND) && (param_descriptor != NULL)) {
    uint8_t type = 0;
    err = mbc_master_get_parameter(cid, (char*)param_descriptor->param_key,
                                                    (uint8_t*)par_data, &type);
    if (err == ESP_OK) {
        ESP_LOGI(MASTER_TAG, "Characteristic #%d %s (%s) value = (0x%x) parameter read successful.",
                                 param_descriptor->cid,
                                 (char*)param_descriptor->param_key,
                                 (char*)param_descriptor->param_units,
                                 *(uint32_t*)par_data);
    }
    else 
    {
        ESP_LOGE(MASTER_TAG, "Characteristic #%d %s (%s), parameter read fail.", 
                                param_descriptor->cid,
                                (char*)param_descriptor->param_key,
                                (char*)param_descriptor->param_units);
    }
}
return err;  

}    

// The code below can be used to read every parameter from device with slave address == 1
    uint32_t parameter_data;
    err = read_modbus_parameter(CID_V23, &parameter_data);   
    if (err == ESP_OK) {
        // if read successful then the parameter_data contains the data read from device

    }
```

HEMILSHAH007, 2022-02-16:
I modified this table accordingly still
It shows these errors:-
E (407) MB_CONTROLLER_MASTER: mbc_serial_master_set_descriptor(162): mb descriptor cid field is incorrect.
E (407) MB_CONTROLLER_MASTER: mbc_master_set_descriptor(124): Master set descriptor failure, error=(0x102) (ESP_ERR_INVALID_ARG).
E (417) MODBUS_MASTER: master_init(234): mb controller set descriptor fail, returns(0x102).
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40086af4
I don't know why it is showing CID incorrect.

HEMILSHAH007, 2022-02-16:
Yes CID error solved now.Thanks.But I am getting master get parameter failure and parameter failure. Can you help regarding this?

alisitsyn, 2022-02-16:
Yes, Please provide the log of the communication.

HEMILSHAH007, 2022-02-16:
#include "mbcontroller.h"   // for common Modbus defines
#include "string.h"
#include "esp_log.h"

#define MASTER_MAX_CIDS 2
#define MASTER_MAX_RETRY 100
#define MASTER_PORT_NUM 2
#define MASTER_SPEED 115200
#define MASTER_TAG "MODBUS_MASTER"
#define MB_UART_RXD_PIN 22
#define MB_UART_TXD_PIN 23
#define MB_UART_RTS_PIN 18

#define MASTER_CHECK(a, ret_val, str, ...) \
    if (!(a)) { \
        ESP_LOGE(MASTER_TAG, "%s(%u): " str, __FUNCTION__, __LINE__, ##__VA_ARGS__); \
        return (ret_val); \
    }

#define STR(fieldname) ((const char*)( fieldname ))
#define OPTS(min_val, max_val, step_val) { .opt1 = min_val, .opt2 = max_val, .opt3 = step_val }

// Enumeration of modbus slave addresses accessed by master device
enum {
    MB_DEVICE_ADDR1 = 1
};

// Enumeration of all supported CIDs for device (used in parameter definition table)

enum {
    CID_V_1ST_PHASE = 0,
    CID_V_2ND_PHASE,
    CID_V_3RD_PHASE,
    CID_V_AVERAGE_LN,
    CID_V12,
    CID_V23,
    CID_V31,
    CID_V_AVERAGE_LL,
    CID_CURRENT_I1,
    CID_CURRENT_I2,
    CID_CURRENT_I3,
    CID_CURRENT_AVERAGE,
    CID_KW1,
    CID_KW2,
    CID_KW3,
    CID_HOLD_TEST_REG,
    CID_COUNT
};

const mb_parameter_descriptor_t device_parameters[] = {
    // { CID, Param Name, Units, Modbus Slave Addr, Modbus Reg Type, Reg Start, Reg Size, Instance Offset, Data Type, Data Size, Parameter Options, Access Mode}
    { CID_V_1ST_PHASE, STR("Voltage first phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 0, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_2ND_PHASE, STR("Voltage second phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 2, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_3RD_PHASE, STR("Voltage third phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 4, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_AVERAGE_LN, STR("Average voltage LN"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 6, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V12, STR("Voltage_V12"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 8, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V23, STR("Voltage_V23"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 10, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V31, STR("Voltage_V31"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 12, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_AVERAGE_LL, STR("Voltage_AVERAGE_LL"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 14, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_CURRENT_I1, STR("Current_I1"), STR("A"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 16, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_CURRENT_I2, STR("Current_I2"), STR("A"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 18, 2,
[... 153 lines cut ...]
            "mb serial set mode failure, uart_set_mode() returned (0x%x).", (uint32_t)err);

    vTaskDelay(5);
    err = mbc_master_set_descriptor(&device_parameters[0], num_device_parameters);
    MASTER_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
                                "mb controller set descriptor fail, returns(0x%x).",
                                (uint32_t)err);
    ESP_LOGI(MASTER_TAG, "Modbus master stack initialized...");
    return err;
}

void app_main(void)
{
    // Initialization of device peripheral and objects
    ESP_ERROR_CHECK(master_init());
    vTaskDelay(10);
    
    // Write registers to predefined state
    /*uint16_t register_data = 0x1234;
    write_modbus_parameter(CID_DEV_REG0, &register_data);
    register_data = 0x5678;
    write_modbus_parameter(CID_DEV_REG1, &register_data);*/

    master_read_func(NULL);
}

HEMILSHAH007, 2022-02-16:
#include "mbcontroller.h"   // for common Modbus defines
#include "string.h"
#include "esp_log.h"

#define MASTER_MAX_CIDS 2
#define MASTER_MAX_RETRY 100
#define MASTER_PORT_NUM 2
#define MASTER_SPEED 115200
#define MASTER_TAG "MODBUS_MASTER"
#define MB_UART_RXD_PIN 22
#define MB_UART_TXD_PIN 23
#define MB_UART_RTS_PIN 18

#define MASTER_CHECK(a, ret_val, str, ...) \
    if (!(a)) { \
        ESP_LOGE(MASTER_TAG, "%s(%u): " str, __FUNCTION__, __LINE__, ##__VA_ARGS__); \
        return (ret_val); \
    }

#define STR(fieldname) ((const char*)( fieldname ))
#define OPTS(min_val, max_val, step_val) { .opt1 = min_val, .opt2 = max_val, .opt3 = step_val }

// Enumeration of modbus slave addresses accessed by master device
enum {
    MB_DEVICE_ADDR1 = 1
};

// Enumeration of all supported CIDs for device (used in parameter definition table)

enum {
    CID_V_1ST_PHASE = 0,
    CID_V_2ND_PHASE,
    CID_V_3RD_PHASE,
    CID_V_AVERAGE_LN,
    CID_V12,
    CID_V23,
    CID_V31,
    CID_V_AVERAGE_LL,
    CID_CURRENT_I1,
    CID_CURRENT_I2,
    CID_CURRENT_I3,
    CID_CURRENT_AVERAGE,
    CID_KW1,
    CID_KW2,
    CID_KW3,
    CID_HOLD_TEST_REG,
    CID_COUNT
};

const mb_parameter_descriptor_t device_parameters[] = {
    // { CID, Param Name, Units, Modbus Slave Addr, Modbus Reg Type, Reg Start, Reg Size, Instance Offset, Data Type, Data Size, Parameter Options, Access Mode}
    { CID_V_1ST_PHASE, STR("Voltage first phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 0, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_2ND_PHASE, STR("Voltage second phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 2, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_3RD_PHASE, STR("Voltage third phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 4, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_AVERAGE_LN, STR("Average voltage LN"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 6, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V12, STR("Voltage_V12"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 8, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V23, STR("Voltage_V23"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 10, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V31, STR("Voltage_V31"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 12, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_V_AVERAGE_LL, STR("Voltage_AVERAGE_LL"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 14, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_CURRENT_I1, STR("Current_I1"), STR("A"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 16, 2,
            0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },
    { CID_CURRENT_I2, STR("Current_I2"), STR("A"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 18, 2,
[... 89 lines cut ...]
    MASTER_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
            "mb serial set pin failure, uart_set_pin() returned (0x%x).", (uint32_t)err);
    // Set driver mode to Half Duplex
    err = uart_set_mode(MASTER_PORT_NUM, UART_MODE_RS485_HALF_DUPLEX);
    MASTER_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
            "mb serial set mode failure, uart_set_mode() returned (0x%x).", (uint32_t)err);

    vTaskDelay(5);
    err = mbc_master_set_descriptor(&device_parameters[0], num_device_parameters);
    MASTER_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
                                "mb controller set descriptor fail, returns(0x%x).",
                                (uint32_t)err);
    ESP_LOGI(MASTER_TAG, "Modbus master stack initialized...");
    return err;
}

void app_main(void)
{
    // Initialization of device peripheral and objects
    ESP_ERROR_CHECK(master_init());
    vTaskDelay(10);
    

    master_read_func(NULL);
}

HEMILSHAH007, 2022-02-16:
This above is my modified code ...

alisitsyn, 2022-02-16:
Please attach the code as archive here because the github removes some details. And please include your error log.

HEMILSHAH007, 2022-02-16:
[SMM.txt](https://github.com/espressif/esp-idf/files/8079736/SMM.txt)
[image]

HEMILSHAH007, 2022-02-16:
I had attached code in these SMM file and this errors I am getting in my serial monitor.

alisitsyn, 2022-02-16:
This means that your slave device does not respond.

1. Please make sure the device is connected to the communication interface correctly and is able to respond to the master simulator.
2. Make sure the address of the slave in the network segment is configured == MB_DEVICE_ADDR1 and the baud rate is set to the same value as in the master example.

HEMILSHAH007, 2022-02-16:
Yes, my connection is correct as I am successfully running the UART ECHO code.
My slave address is 1 and the baud rate is the same too, still I am checking this all once again...

alisitsyn, 2022-02-16:
All slave devices of different manufacturers sometimes have different Modbus address notations.  Please also try to change register addresses in the data dictionary like below:
```
{ CID_V_1ST_PHASE, STR("Voltage first phase"), STR("Volts"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 30000, 2,
0, PARAM_TYPE_U32, 4, OPTS(0, 65535, 1 ), PAR_PERMS_READ_WRITE_TRIGGER },

.... other changes here...

```

HEMILSHAH007, 2022-02-16:
Yes I am checking by changing values..

HEMILSHAH007, 2022-02-17:
I changed the values of registers constantly but I don't know why these errors are still coming....something is still not configured.

alisitsyn, 2022-02-17:
This error just means that your slave does not recognize the request and does not respond to the master.
In order to find the reason for the error I propose you configure the  [Modbus Slave Simulator](https://www.modbustools.com/modbus_slave.html) tool. Add there the registers of your slave device and try to communicate from the master with the Slave simulator first. This will allow us to record the communication traffic and find the reason for the issue. 

Please:

1. install the simulator.
2. create the input register configuration window with the registers as per your slave register map table.
3. Start your master and then slave simulator and record the traffic.

Thanks.

HEMILSHAH007, 2022-02-18:
I had tried and got a 
[Uploading Communication log.txt…]()
communication log I am attaching it...

HEMILSHAH007, 2022-02-18:
I had tried this and got a communication log in MODPOLL
[Communication log.txt](https://github.com/espressif/esp-idf/files/8094853/Communication.log.txt)

alisitsyn, 2022-02-18:
@HEMILSHAH007,

The log looks good and your master should work the same way. Please try to increase slave response time in kconfig menu (see above in the topic). Could you connect your simple modbus master to Modbus Slave simulator and also record the traffic?

alisitsyn, 2022-02-18:
Please connect your devices like below: 

```
                                                      Modbus Slave SW
                                                      Record MB Master Traffic from ESP
    -------------                 -------------        -------------
    |   ESP     |-TX--@---- +--RX-|           |        |           |
    |  MODBUS   |---<>---+  |     |TTL to USB |--------|PC_MB_SLAVE|
    |  MASTER   |-RX-+   |  @  TX-|  adapter  |        |           |
    -------------    |  RTS |     -------------        -------------
                     |   |  |
                     |   |  |
                  ---+---+--+--             
                  |RO RE/DE DI| 
                  | RS485     |
                  |           |
                  -------------                  
                    |    |
                    A    B
    -------------   |    |
    |           |---+    |
    |METER RS485|        |
    |           |--------+
    -------------
```

HEMILSHAH007, 2022-02-22:
In your given schematic diagram RE/DE means to short them or take any one of them because I am shorting both of them always to RTS.

alisitsyn, 2022-02-22:
> In your given schematic diagram RE/DE means to short them or take any one of them because I am shorting both of them always to RTS.

HEMILSHAH007, 2022-02-22:
@alisitsyn I tried this way it didn't work. I dont know why traffic is not showing in slave simulator.

alisitsyn, 2022-02-22:
I can not check this on your side but in fact, it should work just fine if you follow the connection diagram. Please try just connection to serial simulator using your USB-TTL adapter. A better way is to connect the simple Modbus master to your meter and record the traffic using the logic analyzer or oscilloscope.
```
    -------------                 -------------        -------------
    |   ESP     |-TX--@-----+--RX-|           |        |           |
    |  MODBUS   |---              |TTL to USB |--------|PC_MB_SLAVE|
    |  MASTER   |-RX-+---------TX-|  adapter  |        |           |
    -------------                 -------------        -------------
```

HEMILSHAH007, 2022-03-02:
Connection:- 
 RS485 USB TTL(COM 6) - TO ->MAX485 ->ESP32 (COM 4)
Now from the same PC, I am trying to send data(HELLO) from TX OF USB TTL COM 6 to RX OF UART OF ESP32.  I had already done reverse connection such as sending data from ESP TO USB TTL and it succeeds but when I am doing USB TTL TO ESP32 data send it is not showing data on serial monitor of esp32 ... I am sending code hereby for both Proceses.

CODE:-

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_log.h"
#include "driver/uart.h"
#include "string.h"
#include "driver/gpio.h"

static const int RX_BUF_SIZE = 1024;

#define TXD_PIN (GPIO_NUM_17)
#define RXD_PIN (GPIO_NUM_16)
#define RTS_PIN (GPIO_NUM_18)

void init(void) {
    const uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_APB,
    };
    // We won't use a buffer for sending data.
    uart_driver_install(UART_NUM_1, RX_BUF_SIZE * 2, 0, 0, NULL, 0);
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, TXD_PIN, RXD_PIN, RTS_PIN , UART_PIN_NO_CHANGE);
}

int sendData(const char* logName, char* data)
{
    const int len = strlen(data);
    const int txBytes = uart_write_bytes(UART_NUM_1, data, len);
    ESP_LOGI(logName, "Wrote %d bytes", txBytes);
    return txBytes;
}

/*static void tx_task(void *arg)
{
    static const char *TX_TASK_TAG = "TX_TASK";
    esp_log_level_set(TX_TASK_TAG, ESP_LOG_INFO);
    while (1) {
        sendData(TX_TASK_TAG, "HELLO HEMIL\n");
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
}*/

static void rx_task(void *arg)
{
  {
    static const char *RX_TASK_TAG = "RX_TASK";
    esp_log_level_set(RX_TASK_TAG, ESP_LOG_INFO);
    uint8_t* data = (uint8_t*) malloc(RX_BUF_SIZE+1);
    while (1) {
        const int rxBytes = uart_read_bytes(UART_NUM_1, data, RX_BUF_SIZE, 1000 / portTICK_PERIOD_MS);
        if (rxBytes > 0) {
            ESP_LOGI(RX_TASK_TAG, "Read %d bytes: '%s'", rxBytes, data);
             printf("%s", data);
            
      }
    }
  }
}

void app_main(void)
{
    init();

   xTaskCreate(rx_task, "uart_rx_task", 1024*2, NULL, configMAX_PRIORITIES, NULL);
 //  xTaskCreate(tx_task, "uart_tx_task", 1024*2, NULL, configMAX_PRIORITIES-1, NULL);
}

HEMILSHAH007, 2022-03-02:
I want to check whether UART RX is receiving something or not so I did this,  UART Tx successfully prints data on RX OF USB TTL so it's done.

alisitsyn, 2022-03-03:
> I want to check whether UART RX is receiving something or not so I did this, UART Tx successfully prints data on RX OF USB TTL so it's done.

Ok, thank you for the results, but I was going to make sure that modbus master stack sends a correct request over UART and the opposite side (PC) is able to receive it and  respond to master. This is why I asked you to install the Modbus slave SW on PC side just to record the Modbus log. 
The RTS pin is not controlled by ESP32 if the [HALF DUPLEX MODE](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/uart/uart_echo_rs485/main/rs485_example.c#L90) is not set. So, your code is able to work in one direction.

Please use the [echo rs485 example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485) to check the RS485 interface connection instead of your code. The [uart echo](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo) can be used to check USB-TTL to ESP32 connection.

See the readme.md of the example for more information.

HEMILSHAH007, 2022-03-03:
I tried the ECHO RS485 code in Inspite of sending data from USB TTL to ESP32 I didn't receive any data on my ESP-IDF Serial monitor.

[46 more replies]