espressif/esp-idf #7529: I2S output broken after 'refactor/i2s_driver' merge commit

EricMiddleton, 2021-09-06:
I2S output in project based on a2dp_sink example is broken on commit "2991eea97db8302d0523f7dad402dda9ae8bf645" (merge branch refactor/i2s_driver into master) and works on previous commit.

I am initializing the i2s driver like this:
    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_TX,                                  // Only TX
        .sample_rate = 44100,
        .bits_per_sample = 16,
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,                           //2-channels
        .communication_format = I2S_COMM_FORMAT_STAND_MSB,
        .dma_buf_count = 4,
        .dma_buf_len = 1024,
        .intr_alloc_flags = 0,                                                  //Default interrupt priority
        .tx_desc_auto_clear = true,
        .use_apll = true,
        .fixed_mclk = 256*44100                                              //Auto clear tx descriptor on underflow
    };

The audio out of my DAC is broken (super glitchy and periodic, kind of sounds as though a significant percent of the audio samples are never making it out to the DAC).

L-KAYA, 2021-09-07:
Sorry for the issue, may I ask if you are using built-in DAC or DAC in external codec? And what chip you're using? I'd like to reproduce this issue, could you offer more details?

L-KAYA, 2021-09-07:
This issue is caused by wrong configuration of APLL, it will be fixed in few days, you can use following configuration to have a temporary quick fix
```
i2s_config_t i2s_config = {
    .mode = I2S_MODE_MASTER | I2S_MODE_TX, // Only TX
    .sample_rate = 44100,
    .bits_per_sample = 16,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT, //2-channels
    .communication_format = I2S_COMM_FORMAT_STAND_MSB,
    .dma_buf_count = 6,
    .dma_buf_len = 64,
    .intr_alloc_flags = 0, //Default interrupt priority
    .tx_desc_auto_clear = true,
    .use_apll = false,
    .mclk_multiple = I2S_MCLK_MULTIPLE_256
};
```

Sorry for this issue again and thank you for your feedback!

meltdown03, 2021-09-10:
Could this be causing my issue with the ESP-ADF's `record_raw_http` example? It used to work but now the recordings play back like they're in slow motion or fast-forward depending on the bitrate I set. I tried setting the APLL to off and it seems better at 32k hz but not others. If i switch to the v4.3 branch of IDF and rebuild, it works fine.

L-KAYA, 2021-09-13:
Probably, the apll seems not the critical issue that cause these glitches. I2S works well at all the bit rate in I2S examples, I guess there might be some conflicts to WIFI/BLE. Need more time to find the true problem out.

L-KAYA, 2021-09-14:
@meltdown03 Humm, for IDFGH-5824, the glitches disappeared after APLL clock fixed. So could we wait for this bugfix get merged and see if it can also help with your issue? If not, you can start a new issue thread to describe it.

meltdown03, 2021-09-14:
My issue seemed to be related to recording from the i2s device (ESP-LyraT-8311A v1.2). I accidentally destroyed the board though so I can't test anything now.

L-KAYA, 2021-09-14:
@meltdown03 Never mind, I'm using ESP-LyraT-8311A v1.2 too, so I can tell I2S can work well on it. Another news is there will be an new i2s_es8311 example in IDF soon, you can test I2S directly in IDF instead of ADF, in case of any incompatible version.

meltdown03, 2021-09-14:
I was trying to use the ES7243 ADC to work so ce it's there but had no luck. Do you know of anyone getting it to work on that board?

L-KAYA, 2021-09-14:
Sorry, I'm not familiar with ADF, I suggest you create a new issue, other people may help~

closed 2021-09-24 by espressif-bot