espressif/esp-idf #7532: USB CDC setting in sdk configuration doesnt do anything (gets overwritten with default on compilation)

xBiohazardx, 2021-09-07:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version v3
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.3
- Build System:         idf.py
- Operating System: Ubuntu
- Using an IDE?: VS-Code with esp idf extension

## Problem Description

USB CDC option gets overwritten by default value when recompiling

### Expected Behavior

USB CDC option to stay in sdkconfig file for compilation

dobairoland, 2021-09-07:
Hi @xBiohazardx. If you have an ESP32 then this is expected behavior. ESP32 doesn't have a native USB. It has serial port which is connect with an UART-to-USB bridge to an USB connector. Look for "serial" connection in the documentation/datasheet for whatever you want to achieve.

xBiohazardx, 2021-09-07:
I have got an ESP32-c3-wroom-02. It has builtin usb and i get the bootloader output to the acm port. Its just the (usb-cdc) sdk setting (to pipe the standard output to the builtin usb instead of uart) which gets overwritten.

igrr, 2021-09-07:
For ESP32-C3, you need to enable `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG ` option in sdkconfig (see https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html#software-configuration).

`CONFIG_ESP_CONSOLE_USB_CDC` is only valid for chips with a USB_OTG peripheral (i.e. a fully programmable USB Device/Host controller). This option is mentioned in ESP32-S2 documentation, for example: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/usb-otg-console.html#software-configuration.

The reason why these are different is that `CONFIG_ESP_CONSOLE_USB_CDC` relies on software CDC implementation, while `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` uses the hardware one.

xBiohazardx, 2021-09-10:
The CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG options seems to be correct. However this option is not available in menuconfig. (I tried it with version 4.3.1 and the latest one)

igrr, 2021-09-10:
@xBiohazardx have you done `idf.py set-target esp32c3` before running `idf.py menuconfig`?

If yes, you can find it under Component config > ESP System Settings > Channel for console output. You can use the search feature — press `/` and then start typing the option name (or part of it) — this will bring up this option in the list.

[image]
[image]

xBiohazardx, 2021-09-10:
Thank you for your reply and sorry for my bad english. I have already set the target to esp32c3.

There is no setting "Channel for console output" in ESP System Settings.
[image]

The setting is in "Common ESP-related"
[image]

However there is no "USB Serial/JTAG Controller" option:
[image]

igrr, 2021-09-10:
@xBiohazardx I'm sorry I missed the fact that you are using v4.3.1. You are right, v4.3.1 doesn't include "usb serial JTAG" feature. It is only available in latest master branch.

xBiohazardx, 2021-09-10:
Switched to the master branch and now its working properly. Thank you for your help.

closed 2021-09-10 by xBiohazardx