espressif/esp-idf #7533: esp_http_client https handshake takes over 15s and really slow in async mode

rucebee, 2021-09-07:
I recovered my old esp32 project to connect to telegram bot api. I year ago it worked fine. But with new esp-idf esp_http_client https became slow and unstable.

Please, refer https://github.com/rucebee/esp-telegram-bot/blob/main/main/app.cpp for full code and https://github.com/rucebee/esp-telegram-bot/blob/main/sdkconfig for config

esp_http_client http sync work fine, btw

Async esp_http_client_*

I (2533) app: https://httpbin.org/get?test=1
D (2533) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (2533) esp-tls: host:httpbin.org: strlen 11
D (2563) esp-tls: connecting...
D (2763) esp-tls: handshake in progress...
D (2773) HTTP_CLIENT: Connection not yet established
I (4773) app: esp_http_client_perform delay 2s
D (4773) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (4773) esp-tls: handshake in progress...
D (4793) HTTP_CLIENT: Connection not yet established
I (6793) app: esp_http_client_perform delay 2s
D (6793) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (6793) esp-tls: handshake in progress...
D (6803) HTTP_CLIENT: Connection not yet established
I (8803) app: esp_http_client_perform delay 2s
D (8803) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (8803) esp-tls: handshake in progress...
D (8813) HTTP_CLIENT: Connection not yet established
I (10813) app: esp_http_client_perform delay 2s
D (10813) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (10813) esp-tls: handshake in progress...
D (11453) HTTP_CLIENT: Connection not yet established
I (13453) app: esp_http_client_perform delay 2s
D (13453) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (13453) esp-tls: handshake in progress...
D (13463) HTTP_CLIENT: Connection not yet established
I (15463) app: esp_http_client_perform delay 2s
D (15463) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (15463) esp-tls: handshake in progress...
D (15473) HTTP_CLIENT: Connection not yet established
I (17473) app: esp_http_client_perform delay 2s
D (17473) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (17473) esp-tls: handshake in progress...
D (17483) app: HTTP_EVENT_ON_CONNECTED
D (17483) HTTP_CLIENT: Write header[3]: GET /get?test=1 HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0

D (17503) app: HTTP_EVENT_HEADER_SENT
I (24513) app: esp_http_client_perform delay 2s
D (28153) HTTP_CLIENT: on_message_begin
D (28163) HTTP_CLIENT: HEADER=Date:Tue, 07 Sep 2021 11:49:47 GMT
D (28163) app: HTTP_EVENT_ON_HEADER, key=Date, value=Tue, 07 Sep 2021 11:49:47 GMT
D (28173) HTTP_CLIENT: HEADER=Content-Type:application/json
D (28173) app: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/json
D (28183) HTTP_CLIENT: HEADER=Content-Length:299
D (28183) app: HTTP_EVENT_ON_HEADER, key=Content-Length, value=299
D (28193) HTTP_CLIENT: HEADER=Connection:keep-alive
D (28193) app: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
D (28213) HTTP_CLIENT: HEADER=Server:gunicorn/19.9.0
D (28213) app: HTTP_EVENT_ON_HEADER, key=Server, value=gunicorn/19.9.0
D (28223) HTTP_CLIENT: HEADER=Access-Control-Allow-Origin:*
D (28223) app: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
D (28233) HTTP_CLIENT: HEADER=Access-Control-Allow-Credentials:true
D (28233) app: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Credentials, value=true
D (28243) HTTP_CLIENT: http_on_headers_complete, status=200, offset=230, nread=230
D (28243) HTTP_CLIENT: content_length = 299
D (28253) HTTP_CLIENT: data_process=0, content_length=299
D (28263) HTTP_CLIENT: http_on_body 299
D (28263) app: HTTP_EVENT_ON_DATA, len=299
E (28263) app: not parsed {
  "args": {
    "test": "1"
  }, 
  "headers": {
    "Content-Length": "0", 
    "Host": "httpbin.org", 
    "User-Agent": "ESP32 HTTP Client/1.0", 
    "X-Amzn-Trace-Id": "Root=1-613751db-55ed0de3139530ce5745643a"
  }, 
  "origin": "178.122.207.101", 
  "url": "https://httpbin.org/get?test=1"
}

D (28303) HTTP_CLIENT: http_on_message_complete, parser=3ffc6544
D (28303) app: HTTP_EVENT_ON_FINISH
I (28313) app: HTTPS Status = 200, content_length = 299
D (28313) app: HTTP_EVENT_DISCONNECTED

Sync esp_http_client_*

D (64153) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (64153) esp-tls: host:httpbin.org: strlen 11
D (64463) esp-tls: handshake in progress...
W (77033) esp-tls: Failed to open new connection in specified timeout
E (77033) TRANS_SSL: Failed to open a new connection
E (77033) HTTP_CLIENT: Connection failed, sock < 0
E (77033) app: Error perform http request ESP_ERR_HTTP_CONNECT
D (77053) app: HTTP_EVENT_DISCONNECTED
I (77053) app: Last esp error code: 0x8012
I (77063) app: Last mbedtls failure: 0x0
I (80063) app: https://httpbin.org/get?test=1
D (80063) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (80063) esp-tls: host:httpbin.org: strlen 11
D (80233) esp-tls: handshake in progress...
D (90163) app: HTTP_EVENT_ON_CONNECTED
D (90163) HTTP_CLIENT: Write header[3]: GET /get?test=1 HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0

D (90183) app: HTTP_EVENT_HEADER_SENT
E (95183) app: Error perform http request ESP_ERR_HTTP_FETCH_HEADER
D (95183) app: HTTP_EVENT_DISCONNECTED
I (98183) app: https://httpbin.org/get?test=1
D (98183) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (98183) esp-tls: host:httpbin.org: strlen 11
D (98353) esp-tls: handshake in progress...
D (108293) app: HTTP_EVENT_ON_CONNECTED
D (108293) HTTP_CLIENT: Write header[3]: GET /get?test=1 HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0

D (108303) app: HTTP_EVENT_HEADER_SENT
E (113313) app: Error perform http request ESP_ERR_HTTP_FETCH_HEADER
D (113313) app: HTTP_EVENT_DISCONNECTED
I (116313) app: https://httpbin.org/get?test=1
D (116313) HTTP_CLIENT: Begin connect to: https://httpbin.org:443
D (116313) esp-tls: host:httpbin.org: strlen 11
D (116583) esp-tls: handshake in progress...
D (120773) app: HTTP_EVENT_ON_CONNECTED
D (120783) HTTP_CLIENT: Write header[3]: GET /get?test=1 HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: httpbin.org
Content-Length: 0

D (120793) app: HTTP_EVENT_HEADER_SENT
D (124063) HTTP_CLIENT: on_message_begin
D (124063) HTTP_CLIENT: HEADER=Date:Tue, 07 Sep 2021 11:47:05 GMT
D (124063) app: HTTP_EVENT_ON_HEADER, key=Date, value=Tue, 07 Sep 2021 11:47:05 GMT
D (124073) HTTP_CLIENT: HEADER=Content-Type:application/json
D (124073) app: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/json
D (124083) HTTP_CLIENT: HEADER=Content-Length:299
D (124083) app: HTTP_EVENT_ON_HEADER, key=Content-Length, value=299
D (124093) HTTP_CLIENT: HEADER=Connection:keep-alive
D (124093) app: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
D (124113) HTTP_CLIENT: HEADER=Server:gunicorn/19.9.0
D (124113) app: HTTP_EVENT_ON_HEADER, key=Server, value=gunicorn/19.9.0
D (124123) HTTP_CLIENT: HEADER=Access-Control-Allow-Origin:*
D (124123) app: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
D (124133) HTTP_CLIENT: HEADER=Access-Control-Allow-Credentials:true
D (124133) app: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Credentials, value=true
D (124143) HTTP_CLIENT: http_on_headers_complete, status=200, offset=230, nread=230
D (124153) HTTP_CLIENT: http_on_body 282
D (124163) app: HTTP_EVENT_ON_DATA, len=282
D (124163) app: HTTP_EVENT_ON_DATA, waiting 17 bytes more
D (124163) HTTP_CLIENT: content_length = 299
D (124173) HTTP_CLIENT: data_process=282, content_length=299
D (124173) TRANS_SSL: remain data in cache, need to read again
D (124183) HTTP_CLIENT: http_on_body 17
D (124183) app: HTTP_EVENT_ON_DATA, len=17
E (124183) app: not parsed {
  "args": {
    "test": "1"
  }, 
  "headers": {
    "Content-Length": "0", 
    "Host": "httpbin.org", 
    "User-Agent": "ESP32 HTTP Client/1.0", 
    "X-Amzn-Trace-Id": "Root=1-61375139-1f1a213671c0d43447347658"
  }, 
  "origin": "178.122.207.101", 
  "url": "https://httpbin.org/get?test=1"
}

D (124223) HTTP_CLIENT: http_on_message_complete, parser=3ffb9c40
D (124223) app: HTTP_EVENT_ON_FINISH
I (124233) app: HTTPS Status = 200, content_length = 299
D (124233) app: HTTP_EVENT_DISCONNECTED

shubhamkulkarni97, 2021-11-01:
Hi @rucebee, thanks for reporting the issue.

Can you reproduce same issue with esp_http_client example provided in IDF?

If not, please share a small example so we can reproduce this issue at our end.

rucebee, 2021-11-03:
> Hi @rucebee, thanks for reporting the issue.
> 
> Can you reproduce same issue with esp_http_client example provided in IDF?
> 
> If not, please share a small example so we can reproduce this issue at our end.

Thank you for reply, 

The problem is not in IDF itself. I installed the latest from github and now it works fine.

So, the steps a got into the trouble:

1. Install fresh Visual Studio Code
2. Install IDF plugin
3. Let IDF plugin to download the latest version of IDF while configuring
And then you get IDF configured for self debugging and as a result working terribly slow

Now, I have the latest IDF installed manually from github and all is fine

BTW, "a small example" is in the first post and the further link to repo

shubhamkulkarni97, 2021-11-11:
@rucebee Great to hear that you are no longer observing this issue. Feel free to reopen.

closed 2021-11-11 by shubhamkulkarni97