espressif/esp-idf #7537: Flash encryption - Using Host Generated Key not working in production

roysG, 2021-09-08:
Hi,
I generated a key and burned it to device.
When i enabled flash encryption, set it to development mode and run the command: idf.py flash monitor

I get errors that i burned regular plaintext(not encrypted).
When i run the command "idf.py encrypted-flash monitor" then it run ok.

Then I tried to run the flash encryption in production,
I changed it in the menu to production and run the command "idf.py flash monitor".

But i got errors (running without ending):

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

What do i miss?

I follow this document:
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/security/flash-encryption.html#flash-encryption-algorithm

Also it is important to mention,
I succeed to burn the key with this lines:

espsecure.py generate_flash_encryption_key my_flash_encryption_key.bin
espefuse.py --port PORT burn_key flash_encryption my_flash_encryption_key.bin

But, i could not understand how to run the last line code line:
espefuse.py --port /dev/usb-0001 burn_key BLOCK my_flash_encryption_key.bin XTS_AES_128_KEY

What should i set instead the "BLOCK" and "XTS_AES_128_KEY"?

ESP-Marius, 2021-09-09:
Hi @roysG 

It sounds like you might be trying to flash a device where flash encryption is already enabled.  You can check this with `espefuse.py summary`, if `FLASH_CRYPT_CNT`'s value is an odd number of bits, then flash encryption is enabled and the method described here will not work. 

If flash encryption is already enabled in release mode and OTA is not possible then the only option is to pre-encrypt the binary on your PC before flashing it (assuming you have the key). See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#manual-encryption for how to do this.

> But, i could not understand how to run the last line code line:
espefuse.py --port /dev/usb-0001 burn_key BLOCK my_flash_encryption_key.bin XTS_AES_128_KEY
>  What should i set instead the "BLOCK" and "XTS_AES_128_KEY"?
Sorry, this is our mistake. Seems like some commands that are only used for ESP32-S2 is present in the ESP32 documentation. This line is not needed on ESP32. I'll make sure we update the docs here.

roysG, 2021-09-09:
Hi,
Thank you for your reply, i have a key for flash encryption.

But the encryption still does not work. Can you tell me the commands for
generate the encrypted code, that works for you?

Thanks

ESP-Marius, 2021-09-09:
You will have to explain a bit more in details what the status is.

Is flash encryption already enabled? (value of the efuse FLASH_CRYPT_CNT)? 
If yes then follow the instructions here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#manually-encrypting-files

roysG, 2021-09-09:
This is the summary of the device:

espefuse.py -p /dev/cu.usbserial-3 summary
Connecting........_____.
Detecting chip type... ESP32
espefuse.py v3.2-dev
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1128 R/W (0b00100)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 51 R/W (0x33)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 128 R/W (0x0080)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 1 R/W (0x1)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = ac:67:b2:2c:0e:20 (CRC 0x27 OK) R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 39 R/W (0x27)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = False R/W (0b0)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 0 R/W (0b000)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 3 R/W (0b0000011)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                         Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

As it looks the flash encryption already enabled

roysG, 2021-09-09:
I tried to run this commands as follow your link:

`
espsecure.py encrypt_flash_data --keyfile flash_encryption_key.bin --address 0x10000 --output build/arduino-lib-builder-encrypted.bin build/arduino-lib-builder.bin

esptool.py -b 115200 write_flash 0x0 build/arduino-lib-builder-encrypted.bin

idf.py monitor

`
But this is the result i get continuously in monitor this messages:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

roysG, 2021-09-09:
Alos i checked in menuconfig that flash encyption is disabled

ESP-Marius, 2021-09-09:
Actually if FLASH_CRYPT_CNT is an even number of bits then flash encryption is disabled.

Just flashing the plaintext with `idf.py flash` should work fine now that the  encryption efuse is disabled.

You can either:

-  Flash a project where the `SECURE_FLASH_ENC_ENABLED` config is **not** enabled. Project should now boot and run without flash encryption
-  Flash a project where the `SECURE_FLASH_ENC_ENABLED` config **is** enabled. Project should now boot, encrypt the binary, burn efuse to enable flash encryption.

roysG, 2021-09-09:
"Flash a project where the SECURE_FLASH_ENC_ENABLED config is enabled. Project should now boot, encrypt the binary, burn efuse to enable flash encryption."

You mean, change in the menuconfig the option to allow flash encryption and the run command: "idf.py flash monitor", right?

ESP-Marius, 2021-09-09:
Yes, but be aware that on bootup it will burn efuses and there is no way to undo this.

roysG, 2021-09-09:
I enabled it as production and this is what i get in the monitor:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

roysG, 2021-09-09:
This is the result from summary:

 espefuse.py -p /dev/cu.usbserial-3 summary
Connecting........____
Detecting chip type... ESP32
espefuse.py v3.2-dev
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1128 R/W (0b00100)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 51 R/W (0x33)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 128 R/W (0x0080)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 1 R/W (0x1)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = ac:67:b2:2c:0e:20 (CRC 0x27 OK) R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 39 R/W (0x27)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = False R/W (0b0)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 0 R/W (0b000)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 3 R/W (0b0000011)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                         Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

ESP-Marius, 2021-09-09:
And what is the value of `FLASH_CRYPT_CNT` now?

I think you are probably still burning a pre-encrypted binary to a chip where flash enc is not enabled.

Could you please post the whole log from when you are flashing?

roysG, 2021-09-09:
Yes of course, 
Do you want i will put here the output of "idf.py flash" ?

ESP-Marius, 2021-09-09:
Yes please

roysG, 2021-09-09:
$ idf.py flash
Executing action: flash
Serial port /dev/cu.usbserial-3
Connecting........

__
Detecting chip type... ESP32
Running ninja in directory /Users/roysabah/esp/esp32-arduino-lib-builder/build
Executing "ninja flash"...
[1/5] cd /Users/roysabah/esp/esp32-arduino-lib-builder...sp32-arduino-lib-builder/build/arduino-lib-builder.bin
arduino-lib-builder.bin binary size 0xad550 bytes. Smallest app partition is 0x100000 bytes. 0x52ab0 bytes (32%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/roysabah/esp/esp32-arduino-lib-builder/build/bootloader/esp-idf/esptool_py && /Users/roysabah/.espressif/python_env/idf4.4_py3.9_env/bin/python /Users/roysabah/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x80000 bootloader 0x1000 /Users/roysabah/esp/esp32-arduino-lib-builder/build/bootloader/bootloader.bin
Bootloader binary size 0x5570 bytes. 0x79a90 bytes (2278%) free.
[2/3] cd /Users/roysabah/esp/esp-idf/components/esptoo...sp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32 -p /dev/cu.usbserial-3 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x90000 arduino-lib-builder.bin 0x80000 partition_table/partition-table.bin
esptool.py v3.2-dev
Serial port /dev/cu.usbserial-3
Connecting........____
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:2c:0e:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00090000 to 0x0013dfff...
Flash will be erased from 0x00080000 to 0x00080fff...
Compressed 21872 bytes to 15226...
Writing at 0x00001000... (100 %)
Wrote 21872 bytes (15226 compressed) at 0x00001000 in 0.8 seconds (effective 214.6 kbit/s)...
Hash of data verified.
Compressed 709968 bytes to 450136...
Writing at 0x00090000... (3 %)
Writing at 0x0009c550... (7 %)
Writing at 0x000a5e3f... (10 %)
Writing at 0x000b076d... (14 %)
Writing at 0x000b6275... (17 %)
Writing at 0x000bbb88... (21 %)
Writing at 0x000c1312... (25 %)
Writing at 0x000c6ecc... (28 %)
Writing at 0x000cc50d... (32 %)
Writing at 0x000d1cda... (35 %)
Writing at 0x000d75c8... (39 %)
Writing at 0x000dce5f... (42 %)
Writing at 0x000e3827... (46 %)
Writing at 0x000e8b61... (50 %)
Writing at 0x000eddb7... (53 %)
Writing at 0x000f30e7... (57 %)
Writing at 0x000f873b... (60 %)
Writing at 0x000fdd04... (64 %)
Writing at 0x001037ab... (67 %)
Writing at 0x00109160... (71 %)
Writing at 0x0010ec85... (75 %)
Writing at 0x001145a4... (78 %)
Writing at 0x0011af49... (82 %)
Writing at 0x0012279e... (85 %)
Writing at 0x00129be3... (89 %)
Writing at 0x0012f959... (92 %)
Writing at 0x00135572... (96 %)
Writing at 0x0013aa52... (100 %)
Wrote 709968 bytes (450136 compressed) at 0x00090000 in 11.1 seconds (effective 511.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 104...
Writing at 0x00080000... (100 %)
Wrote 3072 bytes (104 compressed) at 0x00080000 in 0.1 seconds (effective 245.0 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
Done

ESP-Marius, 2021-09-09:
Hmm, looks OK to me. Could you try just flashing our hello world example?

roysG, 2021-09-09:
Yes, Please send me the link or code

roysG, 2021-09-09:
This is the output for the hello_world:

Executing action: flash
Running cmake in directory /Users/roysabah/esp/esp-idf/examples/get-started/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/roysabah/esp/esp-idf/examples/get-started/hello_world"...
-- Found Git: /usr/local/bin/git (found version "2.23.0")
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/roysabah/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/roysabah/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/roysabah/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /Users/roysabah/esp/esp-idf/examples/get-started/hello_world/sdkconfig
-- Found PythonInterp: /Users/roysabah/.espressif/python_env/idf4.4_py3.9_env/bin/python (found version "3.9.6")
-- Found Perl: /usr/bin/perl (found version "5.18.4")
-- App "hello-world" version: v4.4-dev-1849-g8e3e65a47
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/examples/get-started/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /Users/roysabah/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/roysabah/esp/esp-idf/components/app_trace /Users/roysabah/esp/esp-idf/components/app_update /Users/roysabah/esp/esp-idf/components/asio /Users/roysabah/esp/esp-idf/components/bootloader /Users/roysabah/esp/esp-idf/components/bootloader_support /Users/roysabah/esp/esp-idf/components/bt /Users/roysabah/esp/esp-idf/components/cbor /Users/roysabah/esp/esp-idf/components/cmock /Users/roysabah/esp/esp-idf/components/coap /Users/roysabah/esp/esp-idf/components/console /Users/roysabah/esp/esp-idf/components/cxx /Users/roysabah/esp/esp-idf/components/driver /Users/roysabah/esp/esp-idf/components/efuse /Users/roysabah/esp/esp-idf/components/esp-tls /Users/roysabah/esp/esp-idf/components/esp32 /Users/roysabah/esp/esp-idf/components/esp_adc_cal /Users/roysabah/esp/esp-idf/components/esp_common /Users/roysabah/esp/esp-idf/components/esp_eth /Users/roysabah/esp/esp-idf/components/esp_event /Users/roysabah/esp/esp-idf/components/esp_gdbstub /Users/roysabah/esp/esp-idf/components/esp_hid /Users/roysabah/esp/esp-idf/components/esp_http_client /Users/roysabah/esp/esp-idf/components/esp_http_server /Users/roysabah/esp/esp-idf/components/esp_https_ota /Users/roysabah/esp/esp-idf/components/esp_https_server /Users/roysabah/esp/esp-idf/components/esp_hw_support /Users/roysabah/esp/esp-idf/components/esp_ipc /Users/roysabah/esp/esp-idf/components/esp_lcd /Users/roysabah/esp/esp-idf/components/esp_local_ctrl /Users/roysabah/esp/esp-idf/components/esp_netif /Users/roysabah/esp/esp-idf/components/esp_phy /Users/roysabah/esp/esp-idf/components/esp_pm /Users/roysabah/esp/esp-idf/components/esp_ringbuf /Users/roysabah/esp/esp-idf/components/esp_rom /Users/roysabah/esp/esp-idf/components/esp_serial_slave_link /Users/roysabah/esp/esp-idf/components/esp_system /Users/roysabah/esp/esp-idf/components/esp_timer /Users/roysabah/esp/esp-idf/components/esp_websocket_client /Users/roysabah/esp/esp-idf/components/esp_wifi /Users/roysabah/esp/esp-idf/components/espcoredump /Users/roysabah/esp/esp-idf/components/esptool_py /Users/roysabah/esp/esp [... cut ...]
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 1, 2MB external flash
Minimum free heap size: 294328 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
ets Jun  8 2016 00:22:57

ESP-Marius, 2021-09-09:
Thanks, this means flash encryption is not enabled. So flashing the file we tried earlier should work fine. 

Try again, but before flashing delete the build folder and rebuild.  This should ensure that any pre-encrypted files you could be mistakenly flashing is gone :)

roysG, 2021-10-21:
Sorry, i did not understand what had been fixed?

ESP-Marius, 2021-10-21:
There were some commands that should only be used on ESP32-S2/C3 (`burn_key BLOCK my_flash_encryption_key.bin XTS_AES_128_KEY`) present in the ESP32 docs. These should now be gone.

I couldn't see any other issues in the docs.

roysG, 2021-10-21:
In case I want to protect on the device with both protection:
secure boot and flash encryption.

Do i need to do in the first place the steps for secure boot and then the
flash encryption?

I will also write here the commands I tried to run with no success.

I follow the steps in production according to documentation of the boot
menu for esp32.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html

1. I enabled the secure boot in the menuconfig to One-time Flash.
2. I generated key for the secure boot

` openssl ecparam -name prime256v1 -genkey -noout -out
my_secure_boot_signing_key.pem `

3. Run command:  idf.py bootloader

4.Run command: idf.py flash

5. restart device esp32

Now i following the steps for flash encryption in production mode

1. enable flash encryption as production.

2. run command: idf.py flash monitor

3. restart device

Please approve my steps, thank you.

ESP-Marius, 2021-10-22:
No this won't work. Flash encryption is enabled by the bootloader and if you choose " One-time Flash" you will not be able to reflash the bootloader.

Can you please describe what you are trying to achieve? How are you planning to update your app after enabling these features?

roysG, 2021-10-30:
@ESP-Marius ?