espressif/esp-idf #7538: Random DNS lookup failures when connected with Wi-Fi

ac-frg, 2021-09-08:
When using the Wi-Fi configuration that is stored in the NVS i get the following ip information:

`// Fails
I (284186) NETWORK_WIFI: WIFI GOT IP
I (284188) NETWORK_WIFI_EVENT: Ip: 192.168.0.18
I (284193) NETWORK_WIFI_EVENT: Netmask: 255.255.255.0
I (284199) NETWORK_WIFI_EVENT: Gateway: 192.168.0.194
I (284205) NETWORK_WIFI_EVENT: DNS 1: 192.168.0.194
I (284210) NETWORK_WIFI_EVENT: DNS 2: 0.0.0.0

// Works
I (8443) NETWORK_WIFI: WIFI GOT IP
I (8445) NETWORK_WIFI_EVENT: Ip: 192.168.0.210
I (8450) NETWORK_WIFI_EVENT: Netmask: 255.255.255.0
I (8455) NETWORK_WIFI_EVENT: Gateway: 192.168.0.1
I (8461) NETWORK_WIFI_EVENT: DNS 1: 195.130.130.4
I (8467) NETWORK_WIFI_EVENT: DNS 2: 195.130.131.4`

Note that i always get one of the above ip address info.
When i get the the one where DNS 2 == 0.0.0.0 i get always the following error:

`E (30371) TRANS_TCP: DNS lookup failed err=202 res=0x0                                          
E (30371) HTTP_CLIENT: Connection failed, sock < 0                                           
E (30372) NETWORK_TIME: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT     `

I am using esp-idf version 4.2.2
Any idea why this occurs?

ESP-YJM, 2021-09-09:
@ac-frg When we get IP from router by DHCP, we also will get DNS server information. For your WI-FI repeater, i think it responses the DNS server as itself IP(192.168.0.194). This action is OK. And we send DNS request to this IP, it resolves failed. Could you please help capture the WIFI packet and open the  LWIP DNS debug. Or you can set DNS server manually by API `esp_netif_set_dns_info`. The more details you can refer to https://github.com/espressif/esp-idf/blob/master/examples/protocols/static_ip/main/static_ip_example_main.c

ac-frg, 2021-09-09:
How do i open the LWIP DNS debug? Is it an option in the menuconfig?

ESP-YJM, 2021-09-09:
@ac-frg I think you should first open `LWIP_DEBUG` which is an option in menuconfig and then modify [file](https://github.com/espressif/esp-lwip/blob/2195f7416fb3136831babf3e96c027a73075bd4f/src/include/lwip/opt.h#L3522) to `#define DNS_DEBUG                       LWIP_DBG_ON`

ac-frg, 2021-09-09:
Output with dns debugging enabled:

D (23655) NETWORK_TIME: Initializing time using HTTP                                                                                    
dns_enqueue: "time.azurewebsites.net": use DNS entry 0                                                                     
dns_enqueue: "time.azurewebsites.net": use DNS pcb 0                                                                       
dns_send: dns_servers[0] "time.azurewebsites.net": request                                                                 
sending DNS request ID 8801 for name "time.azurewebsites.net" to server 0                                                  
dns_tmr: dns_check_entries                                                                                                              
dns_send: dns_servers[0] "time.azurewebsites.net": request                                                                 
sending DNS request ID 8801 for name "time.azurewebsites.net" to server 0                                                                                                      
dns_tmr: dns_check_entries                                                                                                              
dns_send: dns_servers[0] "time.azurewebsites.net": request                                                                 
sending DNS request ID 8801 for name "time.azurewebsites.net" to server 0                                                  
dns_tmr: dns_check_entries                                                                                                              
dns_tmr: dns_check_entries                                                                                                              
dns_send: dns_servers[0] "time.azurewebsites.net": request
sending DNS request ID 8801 for name "time.azurewebsites.net" to server 0
dns_tmr: dns_check_entries
dns_tmr: dns_check_entries
dns_tmr: dns_check_entries
dns_check_entry: "time.azurewebsites.net": timeout
E (30613) TRANS_TCP: DNS lookup failed err=202 res=0x0
E (30616) HTTP_CLIENT: Connection failed, sock < 0
E (30621) NETWORK_TIME: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
W (30629) NETWORK_TIME: Failed to synchronize time using HTTP

ESP-YJM, 2021-09-09:
@ac-frg It seems the DNS server not reply the DNS request. Could you please set DNS server(114.114.114.144 is ok) manually by method i told before.

ac-frg, 2021-09-09:
As main or backup DNS server?

ESP-YJM, 2021-09-09:
Main is ok.

AxelLin, 2021-09-09:
> @ac-frg When we get IP from router by DHCP, we also will get DNS server information. For your WI-FI repeater, i think it responses the DNS server as itself IP(192.168.0.194). 

But it is not ok if the behavior is inconsistent between tests.

Why **sometimes** it got
I (284205) NETWORK_WIFI_EVENT: DNS 1: 192.168.0.194
I (284210) NETWORK_WIFI_EVENT: DNS 2: 0.0.0.0
rather than
I (8461) NETWORK_WIFI_EVENT: DNS 1: 195.130.130.4
I (8467) NETWORK_WIFI_EVENT: DNS 2: 195.130.131.4

ESP-YJM, 2021-09-09:
> But it is not ok if the behavior is inconsistent between tests.
>
> Why **sometimes** it got
> I (284205) NETWORK_WIFI_EVENT: DNS 1: 192.168.0.194
> I (284210) NETWORK_WIFI_EVENT: DNS 2: 0.0.0.0
> rather [...]

I think the device connect to different AP and DHCP server maybe different.

ESP-YJM, 2021-09-09:
> 
> 
> I always connect to the same SSID.

Is there another DHCP server in your environment?  Maybe you need enable DHCP debug in menuconfig.

ac-frg, 2021-09-09:
I enabled:

 /**
 * DHCP_DEBUG: Enable debugging in dhcp.c.
 */
#if !defined DHCP_DEBUG || defined __DOXYGEN__
#define DHCP_DEBUG                      LWIP_DBG_ON
#endif

But nothing is shown in the logs.

ESP-YJM, 2021-09-09:
> 
> 
> I enabled:
> 
> /**
> 
>     * DHCP_DEBUG: Enable debugging in dhcp.c.
>       */
>       #if !defined DHCP_DEBUG || defined **DOXYGEN**
>       #define DHCP_DEBUG                      LWIP_DBG_ON
>       #endif
> 
> 
> But nothing is shown in the logs.

For DHCP, you should enable it in menuconfig.

ac-frg, 2021-09-09:
What is the name of the option i can not find it.

AxelLin, 2021-09-09:
> For DHCP, you should enable it in menuconfig.

esp-idf version 4.2.2 does not has this setting in menuconfig.

ESP-YJM, 2021-09-09:
@ac-frg @AxelLin Sorry that i not check it in v4.2.2. Maybe you need modify file https://github.com/espressif/esp-idf/blob/v4.2.2/components/lwip/port/esp32/include/lwipopts.h#L904-L905

ac-frg, 2021-09-09:
dhcp_start(): starting DHCP configuration
dhcp_discover()                                                                                                                         
dhcp_discover(): dhcp state is DISCOVER                                                                                                 
dhcp_select: could not allocate DHCP request                                                                                            
transaction id xid(baee3372)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 500 msecs
dhcp_coarse_tmr()
dns_tmr: dns_check_entries
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(baee3372)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 1000 msecs
dhcp_recv(pbuf = 0x3f81b0c8) from DHCP server 192.168.0.194 port 67
pbuf->len = 300
pbuf->tot_len = 300
skipping option 28 in options
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_OFFER received in DHCP_STATE_SELECTING state
dhcp_handle_offer(netif=0x3ffdf33c) st1
dhcp_handle_offer(): server 0xc200a8c0
dhcp_handle_offer(): offer for 0x1200a8c0
dhcp_select(netif=0x3ffdf33c) st1
dhcp_select(): dhcp state is REQUESTING
transaction id xid(baee3372)
dhcp_select: REQUESTING
dhcp_select(): set request timeout 2000 msecs
D (7264) BLE_GATTS_EVENT: ESP_GATTS_SET_ATTR_VAL_EVT, attr hndl 57, svc hndl 40, status 0
dhcp_recv(pbuf = 0x3f81b0c8) from DHCP server 192.168.0.1 port 67
pbuf->len = 548
pbuf->tot_len = 548
skipping option 12 in options
skipping option 15 in options
skipping option 23 in options
skipping option 42 in options
searching DHCP_OPTION_MESSAGE_TYPE
dhcp_recv(pbuf = 0x3f81b0c8) from DHCP server 192.168.0.194 port 67
pbuf->len = 300
pbuf->tot_len = 300
skipping option 28 in options
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_ACK received
dhcp_check(netif=0x3ffdf33c) st
dhcp_check(): dhcp state is CHECKING
dhcp_check(): set request timeout 500 msecs
D (7464) BLE_GATTS_EVENT: ESP_GATTS_SET_ATTR_VAL_EVT, attr hndl 60, svc hndl 40, status 0
dhcp_recv(pbuf = 0x3f81b0c8) from DHCP server 192.168.0.194 port 67
pbuf->len = 300
pbuf->tot_len = 300
skipping option 28 in options
searching DHCP_OPTION_MESSAGE_TYPE
dhcp_recv(pbuf = 0x3f81b0c8) from DHCP server 192.168.0.1 port 67
pbuf->len = 548
pbuf->tot_len = 548
skipping option 56 in options
searching DHCP_OPTION_MESSAGE_TYPE
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_check(netif=0x3ffdf33c) st
dhcp_check(): dhcp state is CHECKING
dhcp_check(): set request timeout 500 msecs
dhcp_coarse_tmr()
dns_tmr: dns_check_entries
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_bind(netif=0x3ffdf33c) st1
dhcp_bind(): t0 renewal timer 3600 secs
dhcp_bind(): set request timeout 3600000 msecs
dhcp_bind(): t1 renewal timer 1800 secs
dhcp_bind(): set request timeout 1800000 msecs
dhcp_bind(): t2 rebind timer 3150 secs
dhcp_bind(): set request timeout 3150000 msecs
dhcp_bind(): IP: 0x1200a8c0 SN: 0x00ffffff GW: 0xc200a8c0
dhcp_bind(): dhcp state is BOUND
I (8173) esp_netif_handlers: sta ip: 192.168.0.18, mask: 255.255.255.0, gw: 192.168.0.194
I (8181) NETWORK_WIFI: WIFI GOT IP
I (8185) NETWORK_WIFI_EVENT: Ip: 192.168.0.18
I (8190) NETWORK_WIFI_EVENT: Netmask: 255.255.255.0
I (8196) NETWORK_WIFI_EVENT: Gateway: 192.168.0.194
I (8201) NETWORK_WIFI_EVENT: DNS 1: 192.168.0.194
I (8206) NETWORK_WIFI_EVENT: DNS 2: 0.0.0.0

ESP-YJM, 2021-09-09:
@ac-frg It seems your environment has two DHCP server, one is 192.168.0.194 and another is 192.168.0.1. Please check the device 192.168.0.194.

ac-frg, 2021-09-09:
There was indeed a DHCP running on my laptop that was assigning the 192.168.0.194 ip address. Is there a way to only allow the dhcp server of the AP?

ESP-YJM, 2021-09-09:
@ac-frg No, unless you stop the DHCP service on 192.168.0.194.

closed 2021-09-15 by ac-frg