espressif/esp-idf #7539: interrupt from touch TO non-touch (release)?

mahesh2000, 2021-09-08:
hi, is there a way to trigger an interrupt if the touch sensor goes from touch to release of touch? the reverse of normal operation? thanks!

https://github.com/espressif/esp-idf/blob/dd491ee8513689191421943b737a31c159c91d17/examples/peripherals/touch_pad_interrupt/main/esp32/tp_interrupt_main.c#L1

o-marshmallow, 2021-09-09:
Hello @mahesh2000 ,

The [documentation for Touch Sensor](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/touch_pad.html) states that it is possible to trigger the interrupt when your sensor goes **below** your threshold. Here is the function:
```
esp_err_t touch_pad_set_trigger_mode(touch_trigger_mode_t mode)
```
So you can call it with:
```
touch_pad_set_trigger_mode(TOUCH_TRIGGER_BELOW);
```

This will let you detect when a release has occurred

mahesh2000, 2021-09-17:
hi @o-marshmallow, thank you. i will try this.

mahesh2000, 2021-09-21:
@o-marshmallow, that WORKED!! i used pin `T3`. 2 questions more: **1:** can _some_ pins trigger on `TOUCH_TRIGGER_ABOVE` and *others* on `TOUCH_TRIGGER_BELOW` (the default)? i can only do ALL above or ALL below right now. **2:** what's the diff between `touch_pad_config` and `touch_pad_set_thresh`? i'm using both to be safe. thanks!! 

code snippets below. works on Arduino, too.

```
touch_pad_config(T3, 800);
touch_pad_filter_start(TOUCHPAD_FILTER_TOUCH_PERIOD);
touch_pad_set_trigger_mode(TOUCH_TRIGGER_ABOVE);
touch_pad_set_thresh( T3, 800);
```

more details:
```
#include "driver/touch_pad.h"

uint16_t value = 0;

static void tp_example_rtc_intr(void *arg)
{

  uint32_t pad_intr = touch_pad_get_status();
  //clear interrupt
  touch_pad_clear_status();
  touch_pad_read_filtered(T3, &value);
}

void main() {
  touch_pad_init();
  touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);
  touch_pad_set_voltage(TOUCH_HVOLT_2V7, TOUCH_LVOLT_0V5, TOUCH_HVOLT_ATTEN_1V);
  touch_pad_config(T3, 800);
  touch_pad_filter_start(TOUCHPAD_FILTER_TOUCH_PERIOD);
  touch_pad_set_trigger_mode(TOUCH_TRIGGER_ABOVE);
  touch_pad_set_thresh( T3, 800);
  touch_pad_isr_register(tp_example_rtc_intr, NULL);
  touch_pad_intr_enable();
}

```

o-marshmallow, 2021-09-24:
@mahesh2000 Glad it worked!
1: Unfortunately, it is not possible to change the trigger on a per-pin basis, it is only global
2: Looking at ESP32 source code for [touch_pad_config](https://github.com/espressif/esp-idf/blob/master/components/driver/esp32/touch_sensor.c#L246), we can see that this function does more than only setting the threshold, it also initializes the given touch pad (including setting the slope, initial voltage, clearing the mask)

mahesh2000, 2021-09-24:
@o-marshmallow, that's too bad. .Also, I didn't quite understand SET1 and SET2 in touch. Are there actually 20 touch inputs, and you can select one or the other? What would be the benefit of that? 

Also, is there any difference between touch_pad_config(...) and touch_pad_thresh(...)?

Thanks!

o-marshmallow, 2021-09-26:
@mahesh2000 
Short answer: No, it doesn't let you have 20 sensors, both sets share the same 10 sensors and they are used as a wakeup source for the chip when in sleep mode.

Explanations:
Set1 and Set2 are used for waking up the chip when sleeping. Both sets are represented as bitmaps, so they are a 10-bit value, as we have 10 possible touch sensors.
What does it mean? Let's take an example to understand:
```
uint16_t set1 = 0b100; // Only bit 2 is set
uint16_t set2 = 0b001; // Only bit 0 is set
uint32_t enable = 0b101; // Both bit 0 and 2 are set
```
For the given values, Set1 only contains touch sensor 2 (GPIO2), whereas Set2 contains touch sensor 0 (GPIO4). It is possible to have multiple sensors within one set, so it is possible to have the `set1 = 0b11101` which means that Set1 contains touch sensors 0, 2, 3 and 4. Same goes for the enable variable, which is also a bitmap. In the second case, the set is considered as "TOUCHED" if at least one of its sensors is touched.

Then, it is possible to configure which set(s) must be touched to wake up the device, two possibilities: set1 only or both sets.
In fact, the functions `touch_pad_set_group_mask()` and `touch_pad_clear_group_mask()` let you reset or set the values for `set1`, `set2` and `enable` masks.

About `touch_pad_thresh`, once again, its goal is to modify the threshold at runtime, where `touch_pad_config` is used for initialization. Moreover, as I said earlier, it does more than setting the threshold.