espressif/esp-idf #7541: Add client mac and requested hostname to ip_event_ap_staipassigned_t

ToMe25, 2021-09-09:
**Is your feature request related to a problem? Please describe.**

Currently the only way, that I am aware of, to associate the IP address from [ip_event_ap_staipassigned_t](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/include/esp_netif_types.h#L139) to a specific device is to record the mac addresses of connected devices in [wifi_event_ap_staconnected_t](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi_types.h#L674) then iterate over them in [ip_event_ap_staipassigned_t](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/include/esp_netif_types.h#L139) and check using [dhcp_search_ip_on_mac](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L1291) which one corresponds to the new IP address.

This is a lot of extra effort if you for example just want to react differently depending on whether a device connects for the first time.

Having access to the hostname for identification purposes, and to have a human readable device name, would be nice too.

**Describe the solution you'd like**

The most simple solution for this would be to add the client mac address and requested hostname to the dhcp server callback ([dhcps_cb](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L105), [dhcps_set_new_lease_cb](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L1130]), and [esp_netif_dhcps_cb](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/lwip/esp_netif_lwip.c#L670)) and [ip_event_ap_staipassigned_t](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/include/esp_netif_types.h#L139).

At the callback [call site](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L693) the client mac address is available as `m->chaddr` and the requested hostname, if sent, as part of `m->options`.
It would be some extra effort to extract the hostname from the options, but I don't think it should be too hard to do.

This is a relatively small change, however it would still be an API break.

**Describe alternatives you've considered**

One alternative I considered was to add the full [dhcps_msg](https://github.com/espressif/esp-idf/blob/master/components/lwip/include/apps/dhcpserver/dhcpserver.h#L36) struct instead of just the client mac address and requested hostname.
This has the advantage of giving even more info to the user.
It would also remove the need to do extra work to extract the requested hostname.
However it isn't nearly as clean of a solution so I don't think this solution is preferable.

**Additional context**

I don't think I have any additional context for this.

david-cermak, 2021-09-12:
Hi @ToMe25 

Thanks for this feature proposal, I think it makes sense to provide the additional information, like the mac address of the client. Not sure about the hostname, as this seems a little complex.
Are you planning on using this API directly from dhcpserver or via esp-netif? Is it okay if we target this for `v5.0`? This would be easier as we might alter the API, but, there's also a way to make this change backward compatible.

TBH, I would like the alternative of using the `dhcps_msg` as this is a low level API and an additional context could be retrieved if needed. The issue is that the structure is re-used in the code to store the response (ACK message) so we could have to either allocate and copy the original message or invoke the callback before sending ACK -- none of which seems correct. The same applies to the `hostname` option -- here we could *somehow* save the field, but the code doesn't look ideal -- see the WIP version below:
[lwip-WIP-Add-more-client-s-infor-to-dhcp-server-cb.patch.txt](https://github.com/espressif/esp-idf/files/7150208/lwip-WIP-Add-more-client-s-infor-to-dhcp-server-cb.patch.txt)

But, since you're saying that having a hostname would be nice, I would prefer not to include it in the callback parameters.

ToMe25, 2021-09-12:
Thanks for your reply @david-cermak 

> Are you planning on using this API directly from dhcpserver or via esp-netif?

I am using [arduino-esp32](https://github.com/espressif/arduino-esp32) `WiFi#onEvent`, so thats esp-netif I think.

>  Is it okay if we target this for `v5.0`?

I don't know when a 5.0 release is going to happen so, sure, I guess?

> But, since you're saying that having a hostname would be nice, I would prefer not to include it in the callback parameters.

I don't understand this line at all, sorry.
The only way I could interpret this sentence at all, but even that seems like a stretch from the phrasing, would be that me wanting a hostname being a reason not to add the hostname parameter, but I don't think thats the intended message.

About this implementation, wouldn't it make more sense to wrap the callback in case the old method gets called rather then storing both callbacks?
That way it wouldn't have to call two callbacks, which should be faster, and there shouldn't be situations where something that was meant to override is treated as an addition instead(addition meaning both the old and the new callback being called, rather then just the newly set one).
If extracting the hostname has a significant performance penalty a bool flag set in the callback set method could be used to skip the hostname extraction if the callback is set using the old method and thus doesn't receive it.

david-cermak, 2021-09-13:
@ToMe25 

> But, since you're saying that having a hostname would be nice, I would prefer not to include it in the callback parameters.

Sorry, this wasn't clear at all, what I wanted to say is:
```cpp
if (you_say("hostname is nice-to-have") && I_say("extracting hostname is complex")) {
    do_not_extract_hostname();
}
```
I would prefer not extracting it at all and adding only the mac address as extra parameters to the callback. So, in other words, are you okay if we solve this issue by adding only mac address to the event?

> If extracting the hostname has a significant performance penalty 

Not a performance loss I think, but I would like to avoid any extra allocation if possible. That's why posted the patch: to show how *hacky* it is to store the hostname without allocation.

> wouldn't it make more sense to wrap the callback in case the old method gets called rather then storing both callbacks?

This won't be necessary if the target is `v5.0`, but that's an option, too, I think.

ToMe25, 2021-09-13:
@david-cermak 
One of the main things I planned to do with this, tho there is a significant chance I will never get around to doing that, was to make a DNS server library that automatically handles the requested hostnames to allow devices in an esp32 ap to address each other by their hostname.
For this it would be necessary to have access to the hostname.

I'm not sure if that makes it easier at all, but adding the whole options array rather then just the hostname would be fine too.

About your explanation for the line that I didn't understand, I think I get it now, but I'm pretty sure that contains more then one syntax error :P

david-cermak, 2021-09-13:
> For this it would be necessary to have access to the hostname.

Okay, so as it seems that acquiring the hostname is *required* for this feature request (my original assumption was that the mac address was required, but the hostname was *optional*, good to have).

> I'm not sure if that makes it easier at all, but adding the whole options array rather then just the hostname would be fine too.

It wouldn't, as said before, as the option array gets erased and populated by the response packet *before* the callback. We could either allocate (which I was trying to avoid since hundreds thousands usecases it won't be used) or we  could copy it somewhere (as I did in the patch above). Of course we could introduce a new config option or a flag to _enable_ this, but all comes at a cost (maintenance, more RAM)

ToMe25, 2021-09-13:
Then how about keeping it as two set callback methods, or maybe even adding a third one with only ip and mac,
having the one(s?) without the options parameter wrap the callback so there is only ever one callback,
and having them set a flag for whether or not to copy the options?
The flag could be automatically set depending on which method was called.
As long as the callback doesn't use the options that should only come at a RAM cost of a few bytes max, maybe even less.
But it still allows using the hostname for things like my dns idea.
This would still require some additional maintenance, but I don't think it would be that much.
Also in this case using the options might be better since its less code to maintain and should run a bit faster.

I know this still isn't perfect, but its the best I could come up with to try and make this as clean as possible.

david-cermak, 2021-09-14:
I'm afraid we cannot find a cleaner way if we care about allocations, memory consumption and consistency.

Your last proposal sounds (almost) good, just a flag, few bytes, minimal maintenance effort.
yet these couple of bytes are always taken (reserved in the global space, as the dhcp server is still a singleton). Then, we'd need another flag in `esp-netif` to enable it and implement two callbacks on this layer. We have to reserve a space in the event struct (312 bytes!) which will be allocated every time we post this event (even if the flag is disabled, alternatively we could maintain two separate events). Then we have to think about setting this flag from a user application, probably a new config option to `esp_netif_new()` and properly document it and explain that the event data will only get populated only if the flag is enabled)
This is doable, even easy to implement, memory cost acceptable per default config (though not ideal if counting every byte in the static space and not greatly efficient considering extra allocations, but still acceptable). The maintenance outlook does not sound negligible to me, though.
And all this for your "DNS idea"? that you might implement? one day, perhaps?

Apparently I feel a bit reluctant to implement this change. My idea is to keep this issue open to see if anybody else comments/votes for this feature.

As a clean and efficient alternative, might I offer an extension of the dhcp server API only, providing a completely separate and independent callback (or better yet some kind of an `LWIP_HOOK...`) to the `DHCPREQUEST` packet. User application could then get notified whenever a request arrives to store the incoming options and match it with the upcoming event after the request is accepted.
This is of course more complicated from user apps, but opens the flexibility to retrieve whatever information any incoming packet contains  (+ is in line with the way how the lwip hooks usually work). But more importantly, has absolutely no impact on common default confutation.

ToMe25, 2021-09-14:
@david-cermak 

> And all this for your "DNS idea"? that you might implement? one day, perhaps?

Well, I also wanted it for another project that I am already working on, but since its not strictly necessary there, and I will also likely never publish that project I didn't want to use it as an example.

> As a clean and efficient alternative, might I offer an extension of the dhcp server API only, providing a completely separate and independent callback (or better yet some kind of an LWIP_HOOK...) to [...]

As long as `ip_event_ap_staassigned_t` contains ip and mac that would be fine for what I want to do.
I need access to the hostname and/or options of the packet, but whether thats in the event, or in an user app callback doesn't matter, as long as the callback also contains the ip or better yet the mac address, so I can link the event and the callback call.

I have not yet looked into `LWIP_HOOK` stuff, so I can't say for sure whether that would work for me, but it would probably be fine too, again as long as I have the mac address and the hostname/options in one place.

I thought extending the existing event is less of a change then adding a new callback or something, so thats why I suggested adding it there.

Having the mac address in `ip_event_ap_staassigned_t` seems like I'd be useful for many usecases, so that part of the suggestion still stands.

david-cermak, 2021-09-14:
@ToMe25 Okay, so let's wait then if anyone comments and if not I'd add the mac address to the event and elaborate on the `LWIP_HOOK`'s.

> I thought extending the existing event is less of a change then adding a new callback or something, so thats why I suggested adding it there.

So it seems, but those options are no longer valid at the time of the callback.

> I have not yet looked into LWIP_HOOK stuff,

Hope we can get it working in arduino somehow, as it uses precompiled IDF and those hooks are just means to inject macros into the lwip code.

> Having the mac address in ip_event_ap_staassigned_t seems like I'd be useful for many usecases, so that part of the suggestion still stands.

This is a very simple change and will be added in any case.

david-cermak, 2022-02-01:
Hi @ToMe25 

The discussed changes have been pushed to master:
* https://github.com/espressif/esp-idf/commit/e3d71c984af3dcfe99bf94ccb3ceb572d0bc3057 adds client's MAC to the event
* https://github.com/espressif/esp-idf/commit/05911fd4a10f8296811eff3fac858b52759951e8 supports user `LWIP_HOOK` to access DHCP messages
* Example of usage:
https://github.com/espressif/esp-idf/blob/05911fd4a10f8296811eff3fac858b52759951e8/components/lwip/include/apps/dhcpserver/dhcpserver.h#L56-L57

Could you please close the issue if this solves it for you, or comment if not?

ToMe25, 2022-02-01:
@david-cermak Thanks for implementing this :)

While I cannot test this rn because I don't have the time to fully understand how to set up an esp-idf workspace, it looks like this should work perfectly for what I want to do.

I don't know how to use this with arduino, but I'm sure I'll find a some way to do that.

closed 2022-02-01 by ToMe25