espressif/esp-idf #7543: OTA delta update

pegahnikbakht, 2021-09-09:
Hi,

Does OTA delta update have been implemented for ESP32? I have been searching and did not find a clear answer.

Thanks in advance,
Pegah

igrr, 2021-09-09:
Hi @pegahnikbakht, there is no implementation as part of IDF yet. We have done some studies and found a couple promising approaches to this, though. If this is required for your product, please get in touch with Espressif business support and our application engineering team might help to get this implemented in your project.

negativekelvin, 2021-09-09:
@igrr did your study explore what binary size reductions could be expected from a delta update across, for example, a bugfix release?

igrr, 2021-09-09:
We didn't check a lot of cases yet, but taking `wifi_prov_mgr` example and upgrading it from 4.3 to 4.3.1, the compressed delta update was 200 kB while the base binary was 970 kB.

laukik-hase, 2021-09-16:
@pegahnikbakht, I ported [detools](https://github.com/eerimoq/detools)  to ESP-IDF in this project [ESP32 Compressed Delta OTA Updates](https://github.com/laukik-hase/esp32_compressed_delta_ota_update) and got some good results. It is a still a work in progress and I have not tested it out much, so it is **not recommended for production** yet.  But, take a look and see if this what you wanted and if it helps you.

igrr, 2021-09-16:
Nice work @laukik-hase!
I think we can close this issue, as the solution (after additional testing) will likely be distributed via the [component manager](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html).

closed 2021-09-16 by igrr