espressif/esp-idf #7545: esp_partition_iterator_release not usable after esp_partition_next

norventa, 2021-09-10:
The following code snippet shows the issue whereby esp_partition_iterator_release throws a wobbly:

`esp_partition_iterator_t iter = esp_partition_find(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, 0);

    if (iter != 0)
    {
        do
        {
            const esp_partition_t *pPartition = esp_partition_get(iter);

            ESP_LOGI(c_szTag, "Partition: %s", pPartition->label);
            ESP_LOGI(c_szTag, "type: %u", pPartition->type);
            ESP_LOGI(c_szTag, "subtype: 0x%02X", pPartition->subtype);
            ESP_LOGI(c_szTag, "address: 0x%08X", pPartition->address);
            ESP_LOGI(c_szTag, "size: 0x%08X", pPartition->size);
            ESP_LOGI(c_szTag, "encrypted: %u\n", pPartition->encrypted);

            esp_partition_iterator_t it = esp_partition_next(iter);

            if (it == 0)
                break;
        } while (true);

        esp_partition_iterator_release(iter);
    }`

I would say esp_partition_next should return a const esp_partition_t *.

igrr, 2021-09-10:
The code is almost correct, you just need to use the iterator returned from esp_partition_next, instead of discarding it.
```c++
iter = esp_partition_next(iter);
```

See also the example here: https://github.com/espressif/esp-idf/blob/dd491ee8513689191421943b737a31c159c91d17/examples/storage/partition_api/partition_find/main/main.c#L97-L109

norventa, 2021-09-10:
Hi igrr, The problem with that is when esp_partition_next returns null. Then you can't release the iterator.

negativekelvin, 2021-09-10:
It is released automatically

https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/spi_flash/partition.c#L104-L111

norventa, 2021-09-10:
I see. So the call to esp_partition_iterator_release is doubly wrong in the example. Wrong to pass it a null and wrong to call it in the first place. Thank you for making all this clear. Perhaps the header file's comments could be improved.

igrr, 2021-09-10:
Sorry, indeed esp_partition_erase implementation allows `NULL` pointer (which is a common pattern, `free` also allows a `NULL` argument) but the doxygen comment says that the iterator must be non-NULL.

Will fix the comment, thank you for noticing this discrepancy!

norventa, 2021-09-10:
The comments for esp_partition_find: "...Iterator obtained through this function has to be released using esp_partition_iterator_release when not used any more." might be updated to reflect it can get released by calling esp_partition_next.

The comments for esp_partition_iterator_release: "Iterator obtained using esp_partition_find. Must be non-NULL." might be updated to: "Iterator obtained using esp_partition_find or esp_partition_next."

And something similar for esp_partition_next.

Do I need to close this issue or is that something you do?

igrr, 2021-09-10:
This will get closed automatically when the commit with comment updates gets merged.

closed 2021-10-12 by igrr