espressif/esp-idf #7546: A fatal error occurred: Failed to connect to ESP32: Invalid head of packet

Francois-Belanger, 2021-09-10:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-2977-gdd491ee851
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [macOS 11.5.2 MacBook Pro (16-inch, 2019), 2.3 GHz 8-Core Intel Core i9 32 GB 2667 MHz DDR4]
- Using an IDE?: [No]
- Power Supply:         [Battery]

## Problem Description

Since updating to a newer version, cannot connect to my board anymore.  Was working great, was having an issue with the wifi driver, decide to update to latest version... cannot connect anymore !  I can still see all my log in idf.py monitor

```
esptool.py v3.2-dev
Serial port /dev/cu.usbserial-0001
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x4F)
CMake Error at run_serial_tool.cmake:56 (message):
  .../.espressif/python_env/idf4.4_py3.6_env/bin/python
  .../esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32 failed
```

The Invalid packet is always changing... I tried rebooting...

## Debug Logs

This is my older code still running... I can also see that the reset pin is working... 
```
rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7080
load:0x40078000,len:16592
load:0x40080400,len:4536
0x40080400: _init at ??:?

entry 0x400806b8
I (27) boot: ESP-IDF 5aa4ae4-dirty 2nd stage bootloader
I (27) boot: compile time 06:20:07
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) qio_mode: Enabling default flash chip QIO
I (42) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (51) boot.esp32: SPI Flash Size : 16MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00011000 00004000
I (80) boot:  1 otadata          OTA data         01 00 00015000 00002000
I (87) boot:  2 phy_init         RF data          01 01 00017000 00001000
I (94) boot:  3 factory          factory app      00 00 00020000 00200000
I (102) boot:  4 ota_0            OTA app          00 10 00220000 00200000
I (109) boot:  5 ota_1            OTA app          00 11 00420000 00200000
I (117) boot:  6 littlefs         Unknown data     01 82 00620000 00900000
I (125) boot: End of partition table
I (129) boot: Defaulting to factory image
I (134) boot_comm: chip revision: 1, min. application chip revision: 0
I (141) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=416f4h (268020) map
I (223) esp_image: segment 1: paddr=0006171c vaddr=3ffbdb60 size=0478ch ( 18316) load
I (229) esp_image: segment 2: paddr=00065eb0 vaddr=40080000 size=0a168h ( 41320) load
I (243) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=104a5ch (1067612) map
I (537) esp_image: segment 4: paddr=00174a84 vaddr=4008a168 size=10bfch ( 68604) load
I (560) esp_image: segment 5: paddr=00185688 vaddr=50000000 size=00010h (    16) load
I (573) boot: Loaded app from partition at offset 0x20000
I (573) boot: Disabling RNG early entropy source...
LOG: !!!!!!!!!  Booting !!!!!!!!!!!
...
```

radimkarnis, 2021-09-12:
Hello @Francois-Belanger,

ESP-IDF uses [esptool](https://github.com/espressif/esptool) as its submodule for flashing and communicating with the chip. Updating IDF also resulted in an esptool update, which might be ultimately causing your connection problems.

The only recent change in the connection process which I can think of is https://github.com/espressif/esptool/commit/2d72f56ce6b6103d908194f6d4237a2032ce9b89, but that should apply only to ESP32 revision 0 chips (your attached boot log identifies your chip as revision 1).

There might be other factors causing this issue, would you mind pulling esptool independently and going through the [troubleshooting list](https://github.com/espressif/esptool#troubleshooting)? If this doesn't help, please try older versions of esptool (starting with [v3.1](https://github.com/espressif/esptool/tree/v3.1)). This should help us to identify any regressions!

Thank you,
Radim

Francois-Belanger, 2021-09-12:
Hi, yes going back to version 3.1 works :-). I'll try to help by providing the troubled version.

Francois-Belanger, 2021-09-12:
Found it to be the commit:

radimkarnis Remove esp32r0_delay 2d72f56 on Jul 26

radimkarnis, 2021-09-13:
Thank you @Francois-Belanger for confirming! 
Until now, the mentioned `esp32r0_delay` covered the fact, that revisions 1 also require special timing to successfully connect. 

As a temporary patch, it is sufficient to change this https://github.com/espressif/esptool/blob/master/esptool.py#L597 value to `0.5`. We will issue a fix in esptool and update the submodule in IDF.

dobairoland, 2021-09-20:
We will fix this and then close this issue.

closed 2021-09-24 by espressif-bot