espressif/esp-idf #7548: sdkconfig: Needs convenient way to serial flash into an OTA slot and not alter the partition table.

chipweinberger, 2021-09-10:
Currently in Visual Studio Code when you click 'flash' it always writes a new partition table and the currently compiled application takes OTA_0 or Factory.

This is not flexible enough. **When I'm working on the Factory binary, I want it to be flashed to the Factory partition, and when I'm working on the Main Application binary, I want it to be flashed to a partition of my choosing.**

**I think this could be accomplished pretty simply with a sdkconfig option:**
  ✓  Flash To Existing Partition __________ (& Dont Overwrite Existing Partition Table)

chipweinberger, 2021-09-10:
You can do something like this:

`esptool.py --chip esp32 --port com9 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 0x100000 /home/ble_test/build/ota.bin`

But surely there must be something more user friendly? This seems like a common task.

chipweinberger, 2021-09-10:
```
# Write to partition 'factory' the contents of a file named 'factory.bin'
parttool.py --port "/dev/ttyUSB1" write_partition --partition-name=factory "factory.bin"
```

This is getting closer.  IDE support seems worthwhile.

brianignacio5, 2021-09-13:
In the [vscode extension for ESP-IDF](https://github.com/espressif/vscode-esp-idf-extension) we have something related to this. For that follow these steps:

1) Run the `ESP-IDF: Refresh partition table` command. You can choose the partition table offset from the sdkconfig or input it manually. You will see the partitions of your device in the `ESP-IDF Explorer` view in the `Device Partition Explorer`. 

2) Right click to any .bin file and choose to `Flash binary to partition...`

[image]
[image]

chipweinberger, 2021-09-13:
Thanks for the tutorial! I'll give this a shot. I think it's good others might be able to find this now too :)

In terms of suggestion:

1. it would be great if this was a permanent setting. Maybe it could live in a "ESP-IDF: flash settings" command that would affect the behavior of the "flash " button in the IDE. You could choose which bins on the filesystem to flash, where to flash them, etc.

brianignacio5, 2021-09-14:
Usually the IDEs will consume the `build/flasher_args.json` generated from the build directory to flash binaries to each partition based on your sdkconfig for your project. This makes for a simple flash command.

I think we could add such custom flashing use case as:

1) Either a user defined `tasks.json`'s task where the user specifies `esptool.py --port COM4 write_flash 0x00000 my_app.elf-0x00000.bin 0x40000 my_app.elf-0x40000.bin`
2) Add some UI with a form to create such custom task. This could generate the previous task in your project tasks.json.

What do you think?

PD: Insert a custom script ? Can you explain a bit more about this ?

boarchuz, 2021-09-14:
There was some discussion here but it seems to have fizzled out: https://github.com/espressif/esp-idf/pull/6613

@brianignacio5 Would it be possible to integrate building beforehand and monitoring afterwards (ie. a custom or overridden "Build, Flash and Monitor")?

I think there are still benefits to IDF supporting this, eg.
1. Skip building and flashing other partitions (eg. bootloader, partition table, ota_data)
2. Optionally generate and flash a custom ota_data binary to boot from the selected partition in development

chipweinberger, 2021-09-14:
> Insert a custom script ?

Many IDE's support custom scripts to be called at build time. Usually they support "before build starts" and "after build competes". Use cases: generate and flash a unique serial number, automatically runs tests, etc.

> What do you think?

`tasks.json` sounds a little generic. I'm not familiar enough with the build chain to make good suggestions tho.

**The use case is this:**
In production, my ESP32 will ship with 2 applications installed. A Factory app, and a OTA app both pre-installed. The Factory App is very small and can only do firmware updates. The OTA app has the real application in it.

Visual Studio Code does not support this use case very well. The flash button needs more flexibility. When I click flash, I want it to flash a real production device for me, with both the Factory app and the OTA app flashed. Ideally, I want to flash my 'serial number' partition with a unique serial number as well, set the boot partition to the OTA app, etc.

Yes, I can make a custom script for all of this, but IMO the 'flash' button needs this flexibility.

I think the UI would go like this:

Go to 'ESP-IDF: flash settings'

1. you see a list of what gets flashed where, and where the Bins are coming from on the filesystem:
```
Partition Table: /my_custom_partition_table.csv

'partition name',  'bin'
partition_table  /build/partition_table.bin`
factory_app     /build/my_factory.bin`
ota_0         /build/super_app.bin`
```

2. the ability to customize that above table ^. 

2. ability to set the boot partition (i.e OTA Data partition)
`Boot Partition: ota_0`

3. also ability to run custom scripts: 
`Run this script before building: ../../generate-serial-number-bin`
`Run this script after flashing: ../../post-flash-test-device`

chipweinberger, 2021-09-14:
Of course, this is just a suggestion how it would work. #6613 has its own thing too that looks okay.

chipweinberger, 2021-10-29:
Maybe even simpler...

**what if we just allowed the user to replace the IDE flash button with their own flashing script?**

I have a script that generates and flashes the multiple partitions I need, but i rarely use it because the "Flash" button is so convenient to push.

Doing this seems like a better first step because you can cover many more use cases with less complicated code on your side.

brianignacio5, 2021-12-10:
The pre/post build and flash custom tasks will be added in https://github.com/espressif/vscode-esp-idf-extension/pull/585 and the custom flasher args will come a bit later !

chipweinberger, 2021-12-10:
This is great! I'm excited to use it!

appreciate your work.

chipweinberger, 2021-12-10:
Also, when it lands, would be great to get it documented here. Otherwise no one will know about it!

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html

chipweinberger, 2021-12-11:
Oh last question, will this allow me to specify which partition the opened project will be flashed to?

brianignacio5, 2021-12-13:
To override the partition to flash we talked about a way to customize flasher_args.json right?

Basically there is going to be an UI you can use to specify connection method, partitions and offset to flash and you can use this `custom` mode instead of previous `JTAG`, `UART` and `DFU`.

Do you need a custom build task? Our current build task execute 2 things:

`cmake -G Ninja ..` where you can modify arguments (-G Ninja ..) with `idf.cmakeCompilerArgs`
`ninja` where you can modify additional arguments with `idf.ninjaArgs`

chipweinberger, 2021-12-24:
> customize flasher_args.json

will it get wiped when you do a "Full Clean"?

> Do you need a custom build task? 

Yes. My script flashes 3 partitions (2 binaries, and 1 data), and a custom serial number.  So, idf.customTask sounds super useful. Looking forward  to the next vs-code extension update.

> idf.ninjaArgs
> idf.cmakeCompilerArgs
> idf.customTask

How do I modify these things?

chipweinberger, 2021-12-24:
[image]

Found them =)

- On Windows/Linux - File > Preferences > Settings
- On macOS - Code > Preferences > Settings

chipweinberger, 2022-01-07:
Just wanted to report back, I'm using the custom task and it works great =) So useful. Thank you.

brianignacio5, 2022-01-07:
Thanks @chipweinberger ! Always looking forward to make it better for all users!

If you have a few minutes, could you leave a review in the [marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension&ssr=false#overview) ?

If you have an issue or some other features you want or that might be cool, please open a issue in the [extension GitHub repository](https://github.com/espressif/vscode-esp-idf-extension)

closed 2022-01-26 by chipweinberger