espressif/esp-idf #7549: externalproject_add not working

LanFly, 2021-09-11:
## Environment

- Development Kit:         ESP32-Wrover-Kit
- Kit version v4
- Module or chip used:  ESP32-WROOM-32
- IDF version:                  v4.3
- Build System:               CMake
- Compiler version:         xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:       macOS
- Using an IDE?:               No
- Power Supply:               USB

## Problem Description

externalproject_add not working. but working on version ESP_IDF v3.2.

**CMakeList.txt**

```
set(COMPONENT_ADD_INCLUDEDIRS "")
set(COMPONENT_SRCDIRS .)

set(COMPONENT_PRIV_INCLUDEDIRS
    ${JERRYSCRIPT_SOURCE}/jerry-core/include
    ${JERRYSCRIPT_SOURCE}/jerry-ext/include
    ${JERRYSCRIPT_SOURCE}/jerry-port/default/include)

register_component()

# Xtensa processor architecture optimization
set(EXTERNAL_COMPILE_FLAGS -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w)
string(REPLACE ";" "|" EXTERNAL_COMPILE_FLAGS_ALT_SEP "${EXTERNAL_COMPILE_FLAGS}")

externalproject_add(jerryscript_build
  PREFIX ${COMPONENT_DIR}
  SOURCE_DIR ${JERRYSCRIPT_SOURCE}
  BUILD_IN_SOURCE 0
  BINARY_DIR jerryscript
  INSTALL_COMMAND "" # Do not install to host
  LIST_SEPARATOR | # Use the alternate list separator
  CMAKE_ARGS
    -DJERRY_GLOBAL_HEAP_SIZE=${JERRY_GLOBAL_HEAP_SIZE}
    -DJERRY_CMDLINE=OFF
    -DENABLE_LTO=OFF # FIXME: This option must be turned off or the cross-compiler settings will be overwritten
    -DCMAKE_C_COMPILER_WORKS=true # cross-compiler
    -DCMAKE_SYSTEM_NAME=Generic
    -DCMAKE_SYSTEM_PROCESSOR=xtensa
    -DCMAKE_C_COMPILER=${CMAKE_C_COMPILER}
    -DEXTERNAL_COMPILE_FLAGS=${EXTERNAL_COMPILE_FLAGS_ALT_SEP}
    -DCMAKE_EXE_LINKER_FLAGS=${CMAKE_EXE_LINKER_FLAGS}
    -DCMAKE_LINKER=${CMAKE_LINKER}
    -DCMAKE_AR=${CMAKE_AR}
    -DCMAKE_NM=${CMAKE_NM}
    -DCMAKE_RANLIB=${CMAKE_RANLIB}
    -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=NEVER
)

add_dependencies(${COMPONENT_LIB} jerryscript_build)

set(COMPONENT_BUILD_PATH ${CMAKE_CURRENT_BINARY_DIR}/jerryscript)

target_link_libraries(${COMPONENT_LIB}
                      jerry-core
                      jerry-ext
                      jerry-port-default-minimal)
```

**output**

``` bash
Scanning dependencies of target jerryscript_build
[ 98%] Performing build step for 'jerryscript_build'
make[3]: *** No targets specified and no makefile found.  Stop.
make[2]: *** [../main/src/jerryscript_build-stamp/jerryscript_build-build] Error 2
make[1]: *** [esp-idf/main/CMakeFiles/jerryscript_build.dir/all] Error 2
make: *** [all] Error 2
```

### Expected Behavior

Execute the build of the jerryscript project.

### Actual Behavior

no build was performed.

### Steps to reproduce

1. git clone jerryscript project.
2. externalproject_add. See above for detail.

### Code to reproduce this issue

## Debug Logs

```
Scanning dependencies of target jerryscript_build
[ 98%] Performing build step for 'jerryscript_build'
make[3]: *** No targets specified and no makefile found.  Stop.
make[2]: *** [../main/src/jerryscript_build-stamp/jerryscript_build-build] Error 2
make[1]: *** [esp-idf/main/CMakeFiles/jerryscript_build.dir/all] Error 2
make: *** [all] Error 2
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2021-09-14:
@LanFly is this the complete CMakeLists.txt file you are using? It seems that the variable `JERRYSCRIPT_SOURCE` isn't set anywhere.

LanFly, 2021-09-20:
> @LanFly is this the complete CMakeLists.txt file you are using? It seems that the variable `JERRYSCRIPT_SOURCE` isn't set anywhere.

This is not the complete CMakeLists.txt file. The variable `JERRYSCRIPT_SOURCE ` points to the external `jerryscript` path.

igrr, 2021-09-20:
Could you please post the complete file to reproduce, or give a link to the repository with a reproducer?

As is, it looks like you are having issues with `externalproject_add`, which is a standard CMake feature, not ESP-IDF specific. I'm afraid you will have to provide more information for us to help you troubleshoot the issue. The reproducer would help a lot. Also please specify the version of CMake you are using in each of the cases (IDF 3.2 and 4.3).

Alvin1Zhang, 2021-11-22:
@LanFly Thanks for reporting, would you please help share if any updates for the issue? Thanks.

LanFly, 2022-01-08:
> @LanFly Thanks for reporting, would you please help share if any updates for the issue? Thanks.

@Alvin1Zhang @igrr thanks you, I solved the problem because of my own mistakes.

closed 2022-01-08 by LanFly

Alvin1Zhang, 2022-01-10:
@LanFly Thanks for sharing updates.