espressif/esp-idf #7550: ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x400902bb

Prabhuelectro, 2021-09-11:
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x400902bb
0x400902bb: _esp_error_check_failed_without_abort at /home/proz/work/esp32/esp-idf/components/esp_system/esp_err.c:37

file: "IDF/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c" line 371
func: npl_freertos_callout_deinit
expression: esp_timer_stop(co->handle)

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.4-dev-3042-g220590d599
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
Reboot at the error

## Debug Logs

```
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x400902bb
0x400902bb: _esp_error_check_failed_without_abort at /home/proz/work/esp32/esp-idf/components/esp_system/esp_err.c:37

file: "IDF/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c" line 371
func: npl_freertos_callout_deinit
![Screenshot from 2021-09-11 17-38-28](https://user-images.githubusercontent.com/32390411/132947370-f7b5b2db-3a24-4966-a7c5-f756811aebf8.png)

expression: esp_timer_stop(co->handle)
I (40155) wifi_prov_mgr: Provisioning stopped
I (40155) wifi_prov_scheme_ble: BTDM memory released
```

rahult-github, 2021-09-11:
Hi @Prabhuelectro ,

The print is a friendly indication that a timer stop function was called but the timer was not started. It should return from the function instead of processing any wrong timer functionality. This should not cause reboot or crash. Are you observing a crash ? 

Thanks,
 Rahul

rahult-github, 2021-09-12:
Hi @Prabhuelectro ,
 Please confirm if attached changes works ok. 
[correct_warning_print.txt](https://github.com/espressif/esp-idf/files/7149063/correct_warning_print.txt)

Thanks,
 Rahul