espressif/esp-idf #7552: BLE 和 WIFI 共存问题出现任务看门狗触发重启

gl-zhongshangwen, 2021-09-13:
# 环境：

- Module or chip used: ESP32-WROOM-32U
- Build System: CMake
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB
- IDF Version :  ESP-IDF v4.2    --commit c40f2590bf759ff60ef122afa79b4ec04e7633d2

# 数据流向：

 BLE扫描接收广播包   》有效数据过滤 》 环形缓冲器 》轮询任务打包数据 》MQTT发布数据  （单包数据在 20KB以内，平均10KB）

# 问题描述：

BLE 主动广播扫描 和 WIFI STA同时运行时，运行一段时间后出现任务占用率急剧上升，且任务看门狗会复位

# 工程配置信息：

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
[... 1969 lines cut ...]
13-11:19:01   I (1751) wifi:state: auth -> assoc (0)
13-11:19:01   I (1821) wifi:state: assoc -> run (10)
13-11:19:01   I (1861) wifi:connected with MY_AP, aid = 1, channel 6, 40D, bssid = e4:95:6e:47:8c:d4
13-11:19:01   I (1861) wifi:security: WPA2-PSK, phy: bgn, rssi: -31
13-11:19:01   I (1861) wifi:pm start, type: 1
13-11:19:01   
13-11:19:01   ../main/src/app_dev_info.c[130]: WIFI_EVENT_STA_CONNECTED
13-11:19:01   ../components/gl_dev_ble/src/gl_dev_ble.c[104]: Enable GAP
13-11:19:02   ../main/src/app_mqtt_info.c[152]: p_gl_mqtt_client_userConfig->host:134.175.4.252
13-11:19:02   ../main/src/app_mqtt_info.c[153]: p_gl_mqtt_client_userConfig->port:1884
13-11:19:02   ../main/src/app_mqtt_info.c[154]: p_gl_mqtt_client_userConfig->username:
13-11:19:02   ../main/src/app_mqtt_info.c[155]: p_gl_mqtt_client_userConfig->password:
13-11:19:02   ../main/src/app_mqtt_info.c[54]: Other event id:7
13-11:19:02   E (1931) TRANS_TCP: [sock=54] connect() error: Host is unreachable
13-11:19:02   E (1931) MQTT_CLIENT: Error transport connect
13-11:19:02   ../main/src/app_mqtt_info.c[51]: MQTT_EVENT_ERROR
13-11:19:02   ../main/src/app_mqtt_info.c[30]: MQTT_EVENT_DISCONNECTED
13-11:19:02   I (1941) wifi:AP's beacon interval = 102400 us, DTIM period = 1
13-11:19:02   ble cmd list create
13-11:19:02   mqtt cmd list create
13-11:19:02   use esp timer !!!
13-11:19:02   I (1961) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
13-11:19:03   ../main/src/app_dev_info.c[35]: IP_EVENT_STA_GOT_IP
13-11:19:03   I (2911) esp_netif_handlers: sta ip: 192.168.8.202, mask: 255.255.255.0, gw: 192.168.8.1
```

gl-zhongshangwen, 2021-09-13:
在做待机测试时出现，CPU占用率异常

```
任务名             状态  ID    优先级  堆栈    CPU核   CPU使用率
gl_freeStuPrint     R     27    10      884      1	 75%
gl_ota_task         R     23    1       3560     1	t<1%
IDLE0               R     6     0       928      0	13580%
IDLE1               R     7     0       928      1	12509%
led_indicator_t     B     13    1       1372     0	  2%
Tmr Svc             B     8     1       1312     0	  3%
gl_mqtt_data_pu     B     24    1       1700     1	343%
mqtt_task           B     21    5       4224    -1	 16%
main                B     5     1       1792     0	t<1%
tiT                 B     14    18      1888    -1	 53%
emac_rx             S     16    15      3560    -1	t<1%
ipc1                B     3     24      452      1	t<1%
gl_execute_ble_     B     25    2       3316     0	t<1%
gl_execute_mqtt     B     26    2       2620     0	t<1%
hciT                B     19    22      1480     1	266%
BTU_TASK            B     20    20      2984     1	 86%
BTC_TASK            B     18    19      1108     1	 24%
esp_timer           B     1     22      3276     0	 44%
wifi                B     15    23      1052     0	269%
btController        B     17    23      2004     1	582%
ipc0                B     2     24      452      0	t<1%
sys_evt             B     12    20      916      0	t<1%
任务状态: r-运行  R-就绪  B-阻塞  S-挂起  D-删除
Total free psram memory: 3941.34KB internal ram memory:98.86KB

```

gl-zhongshangwen, 2021-09-14:
[日志.txt](https://github.com/espressif/esp-idf/files/7160951/default.txt)
[gl_s10_v2_2.zip](https://github.com/espressif/esp-idf/files/7160981/gl_s10_v2_2.zip)
这是源码以及重启时的完整日志

xiewenxiang, 2021-09-29:
@gl-zhongshangwen 
不要在扫描结果返回的回调函数里，做任何的耗时操作，打印也尽可能的避免，将受到的数据抛到另外一个 task 中处理，让回调函数尽快返回。

gl-zhongshangwen, 2021-10-14:
> @gl-zhongshangwen 不要在扫描结果的返回函数里，任何的耗时操作，打印也很容易避免，将收到的数据抛到另一个任务中处理，让函数做功能尽快返回。

嗯，谢谢的回复，按照你的建议，我将adv数据直接丢入环形数组，然后在另一个任务处理，确实改善了很多，重启变少了，后来我将CPU主频由 240Mhz  降低到 160Mhz ，重启问题得到了彻底解决。

gl-zhongshangwen, 2021-10-14:
> @gl-zhongshangwen 不要在扫描结果的返回函数里，任何持续的操作，也很容易避免，将收到的数据抛到另一个任务中处理，让函数打印功能尽快返回。

嗯，谢谢你的回复，按照你的建议，我将adv数据直接丢入网络数组，然后在另一个任务处理，确实改善了很多，重启少了，以后我将CPU主频由240Mhz降低到160Mhz，重启问题得到了解决。

xiewenxiang, 2021-10-21:
@gl-zhongshangwen 
If this issue was fixed, could you please close it?

closed 2021-10-21 by gl-zhongshangwen