espressif/esp-idf #7553: How to install esp-idf-v4.1 on Ubuntu 20.04?

mianqi2016, 2021-09-13:
----------------------------- Delete below -----------------------------

----------------------------- Delete above -----------------------------

## Environment

- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
1.I have installed and run esi-idf-v4.3, and tried "Blink" example. Since I want to use esp-adf, and I found that esp-adf wad based on esp-idf, but esp-adf nowadays was not supported by esp-idf-v4.3, so I downloaded and want to install esp-idf-v4.1. Python 3 and pip were all past testing by esp-idf-v4.3, while, esp-idf-v4.1 couldn't be installed on the same machine run esp-idf-v4.3 well.
2.when I run ``xtensa-esp32-elf-gcc --version``, it was prompted:"xtensa-esp32-elf-gcc: command not found"
 
### Expected Behavior
When I typed "esp-idf.py menuconfig", it would shown the config menu.
### Actual Behavior
1.https://ibb.co/NxvpZ1x
2.https://ibb.co/vw0ws0p
### Steps to reproduce

1. Download and unzip esp-idf-v4.1
2. ./install.sh

dobairoland, 2021-09-13:
Hi @mianqi2016 Please use the latest release: v4.1.2. It will prefer the use of python3. And make sure that you have installed the packages from https://docs.espressif.com/projects/esp-idf/en/release-v4.1/get-started/linux-setup.html#install-prerequisites.

If you cannot switch to v4.1.2 then installing the `python-pip-whl` (https://packages.ubuntu.com/focal/python-pip-whl) package could be a workaround. I don't recommend this solution.

dobairoland, 2021-09-13:
BTW, the master branch of ESP-ADF seems to be based on ESP-IDF v3.3 and not v4.1 as you think. cc @jason-mao

mianqi2016, 2021-09-16:
I'm sure after several days trying in different time, I have downloaded esp-idf-v3.3.5 rightly just now, but when I run "./install.sh", it was said:
`Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@1.22.0-97-gc752ad5-5.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/mq/.espressif/python_env/idf3.3_py2.7_env
Installing virtualenv
/usr/bin/python: No module named pip
Traceback (most recent call last):
  File "/home/mq/esp/esp-idf-v3.3.5/tools/idf_tools.py", line 1358, in <module>
    main(sys.argv[1:])
  File "/home/mq/esp/esp-idf-v3.3.5/tools/idf_tools.py", line 1354, in main
    action_func(args)
  File "/home/mq/esp/esp-idf-v3.3.5/tools/idf_tools.py", line 1158, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1
` 
how to fix it? I am sure I have pip installed since I have installed esp-idf-v4.3.

mianqi2016, 2021-09-16:
When I typed:pip --version
It was shown:
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
while, in the upper comment, it was said:/usr/bin/python: No module named pip
Should I add pip here - /usr/bin/python, or I need to change the directory from "/usr/bin/python" to "/usr/lib/python3/dist-packages/"?

dobairoland, 2021-09-16:
v4.3 used Python 3 for which you have `pip`. When you switch to older releases then Python 2 is used by default for which you don't have `pip`. 

You can install `pip` for your old python manually:

```sh
wget https://bootstrap.pypa.io/pip/2.7/get-pip.py          # you can download this with any tool you prefer
/usr/bin/python get-pip.py --user        # --user ensures that it will be installed in you home directory (no sudo is required)
```

There is also a possibility to install the `python-is-python3` package which will make `python` be linked to `python3`. This setting might affect other software in your computer since it will be set for everything. But it is better to use Python 3 because Python 2.7 is no longer supported (by the Python developers) and some packages could be incompatible.

mianqi2016, 2021-09-17:
Right, I went to ". ./export.sh", but there was still errors, I'm seeking for solution. Thanks!

chegewara, 2021-09-17:
https://github.com/espressif/esp-idf/issues?q=is%3Aissue+ubuntu+python+alternative+is%3Aclosed
https://github.com/espressif/esp-idf/issues/5908#issuecomment-701517116

dobairoland, 2021-09-17:
Yes, that update-alternative does the same as installing `python-is-python3`. It will link "python" to "python3".

What have you tried @mianqi2016? Export have reported the same error as before?

mianqi2016, 2021-09-17:
This was the error prompt:https://ibb.co/5jZrxZz
I found the hint was:"https://ibb.co/5jZrxZz", I found the solution was:
1. add "export IDF_PATH=~/esp/esp-idf-v3.3.5" into "~/.profile"
2. comment current "export IDF_PATH=~/esp/esp-idf-v4.3"
3. Restart.
Then, it was OK - https://ibb.co/1Twj3bk

mianqi2016, 2021-09-17:
Why there was a erase line on my first step description？I haven't mark it.

mianqi2016, 2021-09-17:
The newest error:"ImportError: No module named click" after I typed "idf.py menuconfig".

mianqi2016, 2021-09-20:
I found it searched tools under v4.3 directory, while, I want it in v3.3.5 directory(https://ibb.co/Ht4MQDT), that means I want command "idf.py" point to "esp-idf-v3.3.5" other than "esp-idf-v4.3", how to do it?

dobairoland, 2021-09-20:
@mianqi2016 Please check the content of `$IDF_PATH` before running export.

```sh
echo $IDF_PATH      # set it to the right version of ESP-IDF if it is incorrect
. $IDF_PATH/export.sh    # don't forget the dot in beginning

mianqi2016, 2021-09-23:
I went to "make menuconfig" in "esp-adf", I used esp-idf-v3.3.5, and could run "make flash", while, it was said:
"cc1: some warnings being treated as errors"

1. https://ibb.co/NrPhBHk
2. https://ibb.co/rZHRB1S
3. https://ibb.co/j6G8tN6

I tried the 2 examples in "get-started" - "play_mp3" and "play_mp3_control", both the same, Are these errors of source code or of compiler?

dobairoland, 2021-09-23:
@mianqi2016 Please report ESP-ADF related issues in https://github.com/espressif/esp-adf/issues. You will get better help there. The build issue you linked appears in the ESP-ADF project.

closed 2021-09-23 by dobairoland