espressif/esp-idf #7555: AES encryption fails sometimes

pranavdhage, 2021-09-14:
## Environment

- Development Kit : ESP32 devkitv1
- Kit version : v1
- Module or chip used : ESP32-WROOM-32D
- IDF version : v4.3.1-29-ge493a4c30e
- Build System : idf.py
- Compiler version : 8.4.0
- Operating System:  Linux
- Using an IDE?: Yes  (vs code extension)
- Power Supply: USB

## Problem Description
I am working on payload encryption and I want to encrypt the data using AES256 cbc. for that I am using the attached code snippet. I am expecting the data to be encrypted and for the most of the time it is encrypting the data properly, but sometime it fails to do so. After further analysis I found that the `mbedtls_aes_crypt_cbc` is not encrypting it properly. As the input data is of x bytes I am suppose to get x bytes as output as well, but this is not happening consistently. Outof 10 messages in 2 I am observing this. To minimise the failure I have added retries.

```cpp

esp_err_t mqtt_payload_encryption_standalone(unsigned char *output_data, size_t *output_data_length, unsigned char *input_data, int input_data_length, unsigned char *encryption_key, unsigned char *initialization_vector)
{
    int error;
    mbedtls_aes_context aes_context;
    size_t input_data_length_remainder = (input_data_length % 16) == 0 ? 0 : 1;
    size_t input_data_length_quotient = (input_data_length / 16);
    size_t input_data_length_final = (input_data_length_quotient + input_data_length_remainder + 1) * 16;
    unsigned char initialization_vector_local[16] = {0};
    unsigned char encrypted_intermediate_data[1024] = {0};
    esp_fill_random(initialization_vector, 16);
    memcpy((char *)initialization_vector_local, (char *)initialization_vector, 16);
    mbedtls_aes_init(&aes_context);
    int retry_count = 0;
    do
    {
        error = mbedtls_aes_setkey_enc(&aes_context, encryption_key, 256);
        if (error != 0)
            return ESP_FAIL;
        memcpy((char *)initialization_vector_local, (char *)initialization_vector, 16);
        error = mbedtls_aes_crypt_cbc(&aes_context, MBEDTLS_AES_ENCRYPT, input_data_length_final, (unsigned char *)initialization_vector_local, input_data, &encrypted_intermediate_data[0]);
        retry_count++;
    } while ((input_data_length_final != strlen((char *)encrypted_intermediate_data)) && (retry_count < 5));

    if (error != 0)
        return ESP_FAIL;
    error = mbedtls_base64_encode(output_data, 512, output_data_length, &encrypted_intermediate_data[0], strlen((char *)encrypted_intermediate_data));
    if (error != 0)
        return ESP_FAIL;
    mbedtls_aes_free(&aes_context);
    if (input_data_length_final != strlen((char *)encrypted_intermediate_data))
        return ESP_FAIL;
    return ESP_OK;
}
```
I have tested this while WiFi STA and MQTT is turned on.

igrr, 2021-09-14:
It seems like you are using `strlen((char *)encrypted_intermediate_data)`, but `encrypted_intermediate_data` array may contain bytes equal to zero. `strlen` basically returns the number of bytes until the first zero byte. This may be either less or more than the expected length of your data. The length of the encrypted buffer should be the same as `input_data_length_final`. Can you try to replace `strlen` with the input length and try again?

pranavdhage, 2021-09-14:
Thanks, it's working.
Though I have other question about `input_data_length_final`, I am padding the data with the extra block. If I don't do so the data is getting truncated.

closed 2021-09-15 by pranavdhage