espressif/esp-idf #7556: BT legacy pairing wrong pin code error

rpalma0, 2021-09-14:
Hi all,
I'm working with ESP32 Dev Kit 32E and i'm using the "bt_app_acceptor" template provided in esp-idf examples.
Like in this post [https://github.com/espressif/esp-idf/issues/6497](url) the problem is the authentication.
I'm using legacy pairing so when a device tryes to connect to the ESP32 a pin code is request. If the pin code is insert correctly everything works fine but when a wrong pin code in inserted the application crash and reboot showing the following error.

Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3ffb9a34  PS      : 0x00060730  A0      : 0x800e1fd0  A1      : 0x3ffd02b0
A2      : 0x3ffbb394  A3      : 0x0000000a  A4      : 0x00000000  A5      : 0x00000005
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x800e1818  A9      : 0x00000000
A10     : 0x3ffbb3b4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x0000000a
A14     : 0x000000a5  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x00000002
EXCVADDR: 0x3ffb9a34  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

Backtrace:0x3ffb9a31:0x3ffd02b0 |<-CORRUPTED

Probably the exception is called between ESP_BT_GAP_PIN_REQ_EVT and ESP_BT_GAP_AUTH_CMPL_EVT events of the 
esp_bt_gap_cb(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *).

I'll appreciate any suggestion.

Thanks, 
Riccardo

Wth-Esp, 2021-09-14:
Hi, @rpalma0 

Could you please provide basic information (commit of release branch and reproduce method...) about the issue like the templates?

And the issue is about crashing, you can use `xtensa-esp32-elf-addr2line -pfiaC -e [elf path] [backtrace]` to show the function that error occurs and it's very appreciated if you can provide `.elf` file and `sdkconfig`.

Thank you

rpalma0, 2021-09-14:
Hi,
i'm a newbie so i don't know if i can provide you all the necessary information. Anyway this should be the commit.

> commit bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
> Author: Ivan Grokhotkov <ivan@espressif.com>
> Date:   Wed Aug 17 23:08:22 2016 +0800

I started devoloping my application from esp-idf\examples\bluetooth\bluedroid\classic_bt\bt_spp_acceptor, then i disabled the ssp int the sdk to use legacy pairing.
Now my code looks like the following.

```
#include <stdbool.h>
#include <stdint.h>
#include <string.h>
#include <stdio.h>
#include "nvs.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"
#include "freertos/timers.h"
#include "esp_log.h"
#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_gap_bt_api.h"
#include "esp_bt_device.h"
#include "esp_spp_api.h"

#include "time.h"
#include "sys/time.h"

#include "driver/spi_master.h"
#include "esp_spi_flash.h"

#include "driver/gpio.h"
#include "esp_intr_alloc.h"

#define SPP_TAG "SIS_LED_DEMO"
#define SPP_SERVER_NAME "SPP_SERVER"
#define EXAMPLE_DEVICE_NAME "SIS_LED_DEMO"
#define SPP_SHOW_DATA 0
#define SPP_SHOW_SPEED 1
#define SPP_SHOW_MODE SPP_SHOW_DATA

static const esp_spp_mode_t esp_spp_mode = ESP_SPP_MODE_CB;
static const esp_spp_sec_t sec_mask = ESP_SPP_SEC_AUTHENTICATE;
static const esp_spp_role_t role_slave = ESP_SPP_ROLE_SLAVE;

void esp_bt_gap_cb(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    ESP_LOGI(SPP_TAG, " gap event: %d", event);
    int ConnectionIndex;
    switch (event) {
    case ESP_BT_GAP_AUTH_CMPL_EVT:{
        if (param->auth_cmpl.stat == ESP_BT_STATUS_SUCCESS) {
            ESP_LOGI(SPP_TAG, "authentication success: %s", param->auth_cmpl.device_name);
            esp_log_buffer_hex(SPP_TAG, param->auth_cmpl.bda, ESP_BD_ADDR_LEN);
        } else {
            ESP_LOGE(SPP_TAG, "authentication failed, status:%d", param->auth_cmpl.stat);
        }
        break;
    }
    case ESP_BT_GAP_PIN_REQ_EVT:{
       ESP_LOGI(SPP_TAG, "ESP_BT_GAP_PIN_REQ_EVT min_16_digit:%d", param->pin_req.min_16_digit);
        if (param->pin_req.min_16_digit) {
            ESP_LOGI(SPP_TAG, "Input pin code: 0000 0000 0000 0000");
            esp_bt_pin_code_t pin_code = {0};
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 16, pin_code);
        } else {
            ESP_LOGI(SPP_TAG, "Input pin code: 1234");
            esp_bt_pin_code_t pin_code;
            pin_code[0] = '1';
            pin_code[1] = '2';
            pin_code[2] = '3';
            pin_code[3] = '4';
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 4, pin_code);
        }
        break;
    }
    default: {
        break;
    }
    }
    return;
}

static void esp_spp_cb(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
{
    ESP_LOGI(SPP_TAG, " cb event: %d", event);
    int AuthIndex;
    switch (event) {
    case ESP_SPP_INIT_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_INIT_EVT");
        esp_bt_dev_set_device_name(EXAMPLE_DEVICE_NAME);
        esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
        esp_spp_start_srv(sec_mask,role_slave, 0, SPP_SERVER_NAME);
        break;
    case ESP_SPP_DISCOVERY_COMP_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_DISCOVERY_COMP_EVT");
        break;
    case ESP_SPP_OPEN_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_OPEN_EVT");
        break;
    case ESP_SPP_CLOSE_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_CLOSE_EVT");
        xSemaphoreTake(connectionMutex,portMAX_DELAY);
        break;
    case ESP_SPP_START_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_START_EVT");
        break;
    case ESP_SPP_CL_INIT_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_CL_INIT_EVT");
        break;
    case ESP_SPP_DATA_IND_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_DATA_IND_EVT len=%d handle=%d",
                 param->data_ind.len, param->data_ind.handle);

		snprintf(bt_rx_data, (size_t)param->data_ind.len, (char *)param->data_ind.data);
        break;
    case ESP_SPP_CONG_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_CONG_EVT");
        break;
    case ESP_SPP_WRITE_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_WRITE_EVT");
        break;
    case ESP_SPP_SRV_OPEN_EVT:
        ESP_LOGI(SPP_TAG, "ESP_SPP_SRV_OPEN_EVT");
        break;
    default:
        break;
    }
}

void app_main(void)
{
	nvs_flash_init();

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if (esp_bt_controller_init(&bt_cfg) != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s initialize controller failed\n", __func__);
        return;
    }

    if (esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT) != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s enable controller failed\n", __func__);
        return;
    }

    if (esp_bluedroid_init() != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s initialize bluedroid failed\n", __func__);
        return;
    }

    if (esp_bluedroid_enable() != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s enable bluedroid failed\n", __func__);
        return;
    }

    if (esp_spp_register_callback(esp_spp_cb) != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s spp register failed\n", __func__);
        return;
    }
    if ( esp_bt_gap_register_callback(esp_bt_gap_cb) != ESP_OK)
    {
        ESP_LOGE(SPP_TAG, "%s gap register failed: \n", __func__);
        return;
    }
    if (esp_spp_init(esp_spp_mode) != ESP_OK) {
        ESP_LOGE(SPP_TAG, "%s spp init failed\n", __func__);
        return;
    }

//      Set default parameters for Legacy Pairing
//      Use variable pin, input pin code when pairing

     esp_bt_pin_type_t pin_type =ESP_BT_PIN_TYPE_VARIABLE;
     esp_bt_pin_code_t pin_code;
     esp_bt_gap_set_pin(pin_type, 0, pin_code);

	while(1)
	{
		/*
		*
		*/
	}
}

```
In the attached files you can find elf and sdk files.
Hoping i've been exhaustive.+

[files.zip](https://github.com/espressif/esp-idf/files/7161838/files.zip)

BetterJincheng, 2021-09-15:
I tried to reproduce the crash by using bt_spp_initiator with wrong pin code but failed. Here is my LOGs:

```c
I (8836) SPP_ACCEPTOR_DEMO: Input pin code: 1234
E (8946) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5
E (8946) SPP_ACCEPTOR_DEMO: authentication failed, status:9
```

Could you please provide some more information on how to produce the crash ?

rpalma0, 2021-09-15:
Hi @BetterJincheng ,
i'm using bt_spp_acceptor not initiator, i dont know if it is related.
Anyway i tried to download esp-idf master branch (development), i thought i have corrupted something in my old version, but nothing change..
This time i create a new espressif project starting from the esp-idf\examples\bluetooth\bluedroid\classic_bt\bt_spp_acceptor, after disabling ssp in sdk config i have built and flashed the project.
The result doesn't have changed, when i try to pair my OnePlus 8T inserting a wrong pin code the application crashes and reboots.

What version of esp-idf do you use?

dizcza, 2021-09-27:
Let me shine a light on this issue.

I've encountered the same problem while running the bt_spp_acceptor example on an ESP32 board. The ESP-IDF version is the master branch on Sep 6, commit [316988bd2](https://github.com/espressif/esp-idf/tree/316988bd2d3379912e40c159de953318afe5050e).

I've included more verbose logging:

```
I (44338) SPP_ACCEPTOR_DEMO: ESP_BT_GAP_PIN_REQ_EVT min_16_digit:0
I (44348) SPP_ACCEPTOR_DEMO: Input pin code: 1234
I (44428) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 4859290b2641

I (44438) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 4859290b2641

I (44438) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 4859290b2641

Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3ffb639c  PS      : 0x00060e30  A0      : 0x800e2414  A1      : 0x3ffceb00  
A2      : 0x3ffb73b4  A3      : 0x0000000a  A4      : 0x00000000  A5      : 0x00000005  
A6      : 0x3f405de8  A7      : 0x00000000  A8      : 0x800e1a44  A9      : 0x00000000  
A10     : 0x3ffb73d4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x0000000a  
A14     : 0x0000000f  A15     : 0x400f7ee0  SAR     : 0x00000010  EXCCAUSE: 0x00000002  
0x400f7ee0: btu_hcif_command_complete_evt_on_task at /media/dizcza/media/esp-idf/components/bt/host/bluedroid/stack/btu/btu_hcif.c:1153

EXCVADDR: 0x3ffb639c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x3ffb6399:0x3ffceb00 |<-CORRUPTED
```

`addr2line` doesn't help:

```
$ xtensa-esp32-elf-addr2line -pfiaC -e build/bt_spp_acceptor_demo.elf  0x3ffb6399:0x3ffceb00
0x3ffb6399: ?? ??:0
```

The only mention of the backtrace is in the current thread registers:

```
================== CURRENT THREAD REGISTERS ===================
a1             0x3ffceb00	1073539840
```

sdkconfig, elf, and the coredump are attached.

[coredump.log](https://github.com/espressif/esp-idf/files/7234893/coredump.log)
[sdkconfig](https://www.dropbox.com/s/xh70azwb6nl4k6s/sdkconfig?dl=1)
[bt_spp_acceptor_demo.elf](https://www.dropbox.com/s/73xlih0d5izdk77/bt_spp_acceptor_demo.elf?dl=1)

BetterJincheng, 2021-09-27:
Hi, @rpalma0, @dizcza

A fix is done and waiting for approval.

Thanks!

rpalma0, 2021-09-29:
Thanks everybody for helping, looking forward to the new version.

gmelai, 2021-10-06:
Hi,

I encountered the same problem. Which commit should I refer to for downloading the fixed version?

Thanks in advance!