espressif/esp-idf #7558: No esp_adc_cal_esp32s3.c ?

UnexpectedMaker, 2021-09-14:
Hi folks...   I'm on the latest 4.4.x (downloaded today) and I'm working on getting CircuitPython running on the ESP32-S3 - and I can't seem to find `esp_adc_cal_esp32s3.c ` or it's replacement/equivalent?

Is the ADC support for the S3 still a WIP? Or am I supposed to be using one of the other esp_adc_cal_esp32xxx.c files? The ADCs are not compatible with the S2, so I'm not sure how to move forward, but this is a blocker in getting CircuitPython to compile.

Any suggestions?

Thanks!
Seon
Unexpected Maker

Icarus113, 2021-09-15:
Hi @UnexpectedMaker , it will be sync here soon

Alvin1Zhang, 2021-11-22:
Thanks for reporting, this is available at https://github.com/espressif/esp-idf/commit/13b63cd9d28cda2a7ee0d1b39b75d8f7c89367cb, and will be part of v4.4 release, thanks.

closed 2021-11-22 by Alvin1Zhang