espressif/esp-idf #7559: &I2S0 not found ESP32c3 & Arduino + Arduno

mrfaptastic, 2021-09-14:
Is I2S0 support completed yet? I can see reference to IS20 in i2s_struct.h for the esp32c3, but using the latest arduino framework produces:

```
#include <soc/i2s_struct.h>
#include <soc/i2s_reg.h>

static i2s_dev_t* i2s_device          = &I2S0; // Device to use for this library, change if you want.

void setup() {
  // put your setup code here, to run once:

  i2s_device->reserved_0 = 0;

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
=

```
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\ANONYM~1\AppData\Local\Temp\arduino_build_905298\sketch\sketch_sep14a.ino.cpp.o: in function `setup()':
C:\Users\ANONYM~1\AppData\Local\Temp\arduino_modified_sketch_844489/sketch_sep14a.ino:16: undefined reference to `I2S0'
c:/users/anonymous/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\ANONYM~1\AppData\Local\Temp\arduino_modified_sketch_844489/sketch_sep14a.ino:16: undefined reference to `I2S0'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32C3 Dev Module.
```

atanisoft, 2021-09-14:
this likely should have been filed over on https://github.com/espressif/arduino-esp32/ but even then it would be a duplicate of https://github.com/espressif/arduino-esp32/issues/5578 and https://github.com/espressif/arduino-esp32/issues/5597 and https://github.com/espressif/esp-idf/issues/7114

o-marshmallow, 2021-09-28:
Hello @mrfaptastic ,
As stated by @atanisoft , this is not linked to ESP-IDF but to arduino-esp32.
In fact, ESP-IDF does define `I2S0` for ESPC3:
```
PROVIDE ( I2S0  = 0x6002d000 );
```
in [esp32c3.peripherals.ld](https://github.com/espressif/esp-idf/blob/3e370c4296247b349aa3b9a0076c05b9946d47dc/components/soc/esp32c3/ld/esp32c3.peripherals.ld#L10) file.

This issue can be closed.

closed 2021-09-28 by mrfaptastic