espressif/esp-idf #7561: IDF V4.3.1 examples a2dp-source no audio transmitted

wibbler, 2021-09-15:
Using esp-v4.3.1 examples/ bluetooth/bluedroid/classic_bt/a2dp_source No audio is transmitted to the audio sink.
I had changed main.c so device name for ESP_SPEAKER test is ignored, and any BT device is used.
I have tested using 4.1.1 and this works correctly transmits the random noise.
With  v4.3.1 I there is no audio.

so, with identical code 4.1.1 works, 4.3.1  fails.
(I'm using the example as  test, I am having this problem on my production code after changing esp-idf version).

The v4.3.1 console log:

I (31233) BT_AV: bt_app_av_sm_hdlr state 1, evt 0xff00
I (35373) BT_AV: Scanned device: 00:23:02:32:14:f7
I (35383) BT_AV: --Class of Device: 0x240404
I (35383) BT_AV: --RSSI: -64
I (35383) BT_AV: Found a target device, address 00:23:02:32:14:f7, name MA650 Wireless
I (35393) BT_AV: Cancel device discovery ...
I (35393) BT_AV: Device discovery stopped.
I (35403) BT_AV: a2dp connecting to peer: MA650 Wireless
W (35403) BT_APPL: reset flags
I (35413) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
E (37203) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (37493) BT_APPL: new conn_srvc id:18, app_id:0
I (37493) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (37493) BT_AV: a2dp connected
I (39303) RCCT: AVRC conn_state evt: state 1, [00:23:02:32:14:f7]
I (39303) RCCT: AVRC remote features 24b, TG features 2
I (39313) RCCT: remote rn_cap: count 1, bitmask 0x2000
I (41233) BT_AV: bt_app_av_sm_hdlr state 5, evt 0xff00
I (41233) BT_AV: a2dp media ready checking ...
I (41233) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (41233) BT_AV: a2dp media ready, starting ...
W (41243) BT_APPL: new conn_srvc id:18, app_id:1
I (41283) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x824b

I (41283) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (41283) BT_AV: a2dp media start successfully.
I (41283) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x1
I (51233) BT_AV: bt_app_av_sm_hdlr state 5, evt 0xff00

nopnop2002, 2021-09-18:
esp_bt_gap_cancel_discovery function does not work properly on ESP-IDF V4.4.   

- ESP-IDF v4.3-dev-1901-g178b122c1-dirty
esp_bt_gap_cancel_discovery is running on 4631.
ESP_BT_GAP_DISC_STATE_CHANGED_EVT is notified on 4631.   

```
I (0) cpu_start: App cpu up.
I (893) cpu_start: Pro cpu start user code
I (893) cpu_start: cpu freq: 240000000
I (893) cpu_start: Application information:
I (897) cpu_start: Project name:     a2dp-source
I (903) cpu_start: App version:      1
I (907) cpu_start: Compile time:     Sep 18 2021 09:47:42
I (913) cpu_start: ELF file SHA256:  2955c14e416a2128...
I (919) cpu_start: ESP-IDF:          v4.3-dev-1901-g178b122c1-dirty
I (926) heap_init: Initializing. RAM available for dynamic allocation:
I (933) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (939) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (945) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (952) heap_init: At 3FFC9330 len 00016CD0 (91 KiB): DRAM
I (958) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (964) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (971) heap_init: At 40096400 len 00009C00 (39 KiB): IRAM
I (978) spi_flash: detected chip: generic
I (981) spi_flash: flash io: dio
I (987) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1041) BTDM_INIT: BT controller compile version [f060711]
I (1051) system_api: Base MAC address is not set
I (1051) system_api: read default base MAC address from EFUSE
I (1141) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (1531) BT_AV: Starting device discovery...
W (1531) BT_BTC: btc_avrc_tg_set_rn_supported_evt failed: AVRC TG not yet initialized

W (1531) BT_BTC: A2DP Enable with AVRC
I (1551) BT_AV: event: 10
I (1551) BT_AV: event: 10
I (1551) BT_AV: event: 10
I (1551) BT_AV: event: 10
I (1561) BT_AV: Discovery started.
I (4611) BT_AV: Scanned device: 83:21:5d:9a:62:bd
I (4611) BT_AV: --Class of Device: 0x340404
I (4611) BT_AV: --RSSI: -53
I (4611) BT_AV: Checking potential device, address 83:21:5d:9a:62:bd, name TG-148
I (4621) BT_AV: Found a target device, address 83:21:5d:9a:62:bd, name TG-148
I (4631) BT_AV: Cancel device discovery ...
I (4631) BT_AV: Device discovery stopped.
I (4641) BT_AV: a2dp connecting to peer: TG-148
E (4641) BT_APPL: reset flags
I (4651) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
E (6101) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (6151) BT_APPL: new conn_srvc id:18, app_id:0
I (6151) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (6151) BT_AV: a2dp connected
I (11531) BT_AV: bt_app_av_sm_hdlr state 5, evt 0xff00
I (11531) BT_AV: a2dp media ready checking ...
I (11531) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (11531) BT_AV: a2dp media ready, starting ...
W (11541) BT_APPL: new conn_srvc id:18, app_id:1
I (11551) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x824b

I (11551) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (11561) BT_AV: a2dp media start successfully.
I (11561) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x1
```

- ESP-IDF v4.4-dev-2825-gb63ec47238-dirty
esp_bt_gap_cancel_discovery is running on 4216.   
__ESP_BT_GAP_DISC_STATE_CHANGED_EVT is not notified.__   
[... 56 lines cut ...]
I (11556) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (11556) BT_AV: a2dp media ready, starting ...
W (11566) BT_APPL: new conn_srvc id:18, app_id:1
I (11576) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x824b

I (11576) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x3
I (11586) BT_AV: a2dp media start successfully.
I (11586) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x1
I (17176) BT_AV: Device discovery failed, continue to discover...
W (17176) BT_APPL: [0]: state:1, info:x20, avoid_rs 0
I (17186) BT_AV: Discovery started.
E (18656) BT_L2CAP: l2cab is_cong_cback_context
E (18666) BT_L2CAP: l2cab is_cong_cback_context
E (18686) BT_L2CAP: l2cab is_cong_cback_context
E (18706) BT_L2CAP: l2cab is_cong_cback_context
E (18716) BT_L2CAP: l2cab is_cong_cback_context
E (18736) BT_L2CAP: l2cab is_cong_cback_context
E (18756) BT_L2CAP: l2cab is_cong_cback_context
E (18776) BT_L2CAP: l2cab is_cong_cback_context
E (18796) BT_L2CAP: l2cab is_cong_cback_context
E (18806) BT_L2CAP: l2cab is_cong_cback_context
E (18826) BT_L2CAP: l2cab is_cong_cback_context
E (18836) BT_L2CAP: l2cab is_cong_cback_context
I (21556) BT_AV: bt_app_av_sm_hdlr state 5, evt 0xff00
```

BetterJincheng, 2021-09-23:
Hi @nopnop2002,

Thanks for your reporting! 
We have fixed this problem. You can access the fixed version as soon as my MR is merged.

About the wrong behavior of `esp_bt_gap_cancel_discovery`, here are some of my suggestions.
- As we all known, LOG(printf) is a time-consuming instruction. The task will be blocked when there are too many LOGs. 
- Comment some of the LOGs in `filter_inquiry_scan_result` can be helpful.

Thanks!