espressif/esp-idf #7562: ESP32C3 can not reconnect with iPhone

jigneshpatel47, 2021-09-15:
Hi,

I have build and flash example [example_ble_sec_gatts_demo.c](https://github.com/espressif/esp-idf/blob/80c5547b971128fdecbbcf6d8178196d35a15b73/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c) on ESP32C3-DevKitM-1 v1.0. 
I am using application nRF Connect rev 2.4.12 on iPhone 6S, software revision 14.7.1 with esp32c3 to connect with ESP.

My mobile is able to connect first time with esp with the first time. If I disconnect and connect, It can not connect with ESP32C3.

I am facing the issue: iPhone can not connect with esp if it is already paired.

Below is the error log of ESP3C3 while trying to connect from mobile: 

> 
> I (33297) SEC_GATTS_DEMO: advertising start success
> W (34967) BT_HCI: btu_hcif_disconnection_comp_evt DiscCmpl evt: hdl=1, rsn=0x3d
> E (34967) BT_HCI: btu_hcif_hdl_command_complete CC evt: op=0x2022, status=0x2
> I (34977) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
> I (34977) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
> E (34987) BT_APPL: bta_dm_set_encryption, not find peer_bdaddr or peer_bdaddr connection state error
> I (34997) SEC_GATTS_DEMO: advertising start success
> W (36117) BT_HCI: btu_hcif_disconnection_comp_evt DiscCmpl evt: hdl=1, rsn=0x3d
> E (36117) BT_HCI: btu_hcif_hdl_command_complete CC evt: op=0x2022, status=0x2
> I (36127) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
> I (36127) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
> E (36137) BT_APPL: bta_dm_set_encryption, not find peer_bdaddr or peer_bdaddr connection state error
> 

Note: the same thing is working fine with ESP32 module.
I am using ESP-IDF v4.3-beta3-dirty for ESP32C3 and ESP32

Thank you,
Jignesh Patel

xiewenxiang, 2021-09-29:
Please remove the bonding info both in the phone and esp32c3.

jigneshpatel47, 2021-09-30:
Thank you for your response,

After updating the ESP-IDF version to v4.3.1 working fine.

BR,
Jignesh Patel

closed 2021-09-30 by jigneshpatel47