espressif/esp-idf #7564: Freertos's prvTaskIsTaskSuspended function code error

zhangyang-hxd, 2021-09-15:
There is no problem when running, but as below maybe not correct.
### Code to reproduce this issue
components/freertos/tasks.c
```cpp

#if ( INCLUDE_vTaskSuspend == 1 )

    static BaseType_t prvTaskIsTaskSuspended( const TaskHandle_t xTask )
    {
            ...
            if( listIS_CONTAINED_WITHIN( &xPendingReadyList[xPortGetCoreID()], &( pxTCB->xEventListItem )) ||
                listIS_CONTAINED_WITHIN( &xPendingReadyList[!xPortGetCoreID()], &( pxTCB->xEventListItem ))  == pdFALSE )
            {

It should be as below
    static BaseType_t prvTaskIsTaskSuspended( const TaskHandle_t xTask )
    {
            ...
            if( listIS_CONTAINED_WITHIN( &xPendingReadyList[xPortGetCoreID()], &( pxTCB->xEventListItem ))  == pdFALSE ||
                listIS_CONTAINED_WITHIN( &xPendingReadyList[!xPortGetCoreID()], &( pxTCB->xEventListItem ))  == pdFALSE )
            {

```

negativekelvin, 2021-09-15:
Should also be generalized to arbitrary number of processors like

https://github.com/espressif/esp-idf/blob/e493a4c30e1b2f7c54a376327388b9e12262cbbf/components/freertos/tasks.c#L619-L624

Dazza0, 2021-09-27:
@zhangyang-hxd Good catch. It should actually be the following:

```c
    static BaseType_t prvTaskIsTaskSuspended( const TaskHandle_t xTask )
    {
            ...
            if( listIS_CONTAINED_WITHIN( &xPendingReadyList[xPortGetCoreID()], &( pxTCB->xEventListItem ))  == pdFALSE &&
                listIS_CONTAINED_WITHIN( &xPendingReadyList[!xPortGetCoreID()], &( pxTCB->xEventListItem ))  == pdFALSE )
            {
```
The check is to make sure that the task is in neither of the pending ready lists, thus ensuring that it is still suspended. I'll push a fix for it shortly.

> Should also be generalized to arbitrary number of processors like
@negativekelvin We are aware of this issue. There are actually quite a few more places in the SMP implementation that relies of `!xPortGetCoreID()`, thus the current SMP implementation can only support two cores. We have plans to expand SMP support to N cores, thus the generalization to an arbitrary number of processor will be tackled then.

AxelLin, 2021-11-17:
@Dazza0 
release/v4.3 also needs the fix.

AxelLin, 2021-12-24:
v4.3 branch still does not include this fix.

Alvin1Zhang, 2021-12-27:
@AxelLin Thanks for the attention, we have back ported this fix to 4.3, but still under internal reviewing, will sync up here once the fix is available on GitHub.