espressif/esp-idf #7570: Ethernet basic example broke with IDF version 4.3.1

dabbinavo, 2021-09-16:
## Environment

- Development Kit: **none**
- Module or chip used: **ESP32-WROOM-32U**
- IDF version (run ``git describe --tags`` to find it): **v4.3.1**
- Build System: **CMake**
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): **(crosstool-NG esp-2021r1) 8.4.0**
- Operating System: **Windows**
- (Windows only) environment type: **ESP Command Prompt**
- Using an IDE?: **Yes - VSCode with Espressif IDF Plugin**
- Power Supply: **external 3.3V**

## Problem Description

Starting with ESP-IDF V4.3.1, the dhcp client in ethernet mode is not able to start.
It seems that the root cause for the issue is [this change to components/esp_eth/src/esp_eth.c](https://github.com/espressif/esp-idf/commit/03a6c4975c7fd7d0267f26b93f43b5b391ad2de4#diff-f70b7e7eb060881f32460033b9c62d65cf93a019645367af6745d17e45c62f43). When undoing this change, everything works fine again.

Investigations showed, that the error message `esp_netif_lwip: dhcp client start failed` is generated, because the `dhcp_start()` function is called before the [`ethernet_low_level_init()`](https://github.com/espressif/esp-idf/blob/v4.3.1/components/lwip/port/esp32/netif/ethernetif.c#L77) function for some reason ([which sets netif->mtu to 1500](https://github.com/espressif/esp-idf/blob/v4.3.1/components/lwip/port/esp32/netif/ethernetif.c#L83)). Therefore in `dhcp_start()`, the [check if the MTU is large enough](https://github.com/espressif/esp-lwip/blob/2195f7416fb3136831babf3e96c027a73075bd4f/src/core/ipv4/dhcp.c#L822) fails. At this point, `netif->mtu` is still unitialized and therefore `0`.

@suda-morris Do you have any idea why the changes in the above mentioned commit lead to that behaviour?

### Expected Behavior
DHCP client does start.

### Actual Behavior
DHCP client is unable to start: `esp_netif_lwip: dhcp client start failed`

### Steps to reproduce
1. Prepare code (see [Code to reproduce this issue](#Code-to-reproduce-this-issue))
1. Open the basic ethernet example (`/esp-idf/examples/ethernet/basic`)
1. Change the sdkconfig to match your hardware
1. Build, run and monitor the example

### Code to reproduce this issue
The bug can be reproduced in (`/esp-idf/examples/ethernet/basic`).

Maybe the bug can be reproduced without any modifications to the esp-idf and the examples.

I just had a custom hardware for testing. So I added a modified version of an existing ethernet phy and added some comments inside the ESP-IDF for debugging purposes. So here is the modified version of the code.

1. Checkout ESP-IDF v4.3.1 and init all submodules
1. Apply patch [ethernet_basic_example.patch.txt](https://github.com/espressif/esp-idf/files/7176660/ethernet_basic_example.patch.txt)

## Debug Logs

```
I (0) cpu_start: App cpu up.
I (274) cpu_start: Pro cpu start user code
I (274) cpu_start: cpu freq: 160000000
I (274) cpu_start: Application information:
I (278) cpu_start: Project name:     ethernet_basic
I (284) cpu_start: App version:      v4.3.1-dirty
I (289) cpu_start: Compile time:     Sep 16 2021 10:21:01
I (295) cpu_start: ELF file SHA256:  c545828d205b3a96...
I (301) cpu_start: ESP-IDF:          v4.3.1-dirty
I (307) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (320) heap_init: At 3FFB43C8 len 0002BC38 (175 KiB): DRAM
I (326) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (333) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (339) heap_init: At 4008BB44 len 000144BC (81 KiB): IRAM
I (346) spi_flash: detected chip: generic
I (350) spi_flash: flash io: dio
I (355) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (375) system_api: Base MAC address is not set
I (375) system_api: read default base MAC address from EFUSE
I (395) esp_eth.netif.glue: 24:0a:c4:4e:af:cb
I (395) esp_eth.netif.glue: ethernet attached to netif
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_start(netif=0x3ffbb324) 0
netif->mtu: 0 vs DHCP_MAX_MSG_LEN_MIN_REQUIRED: 576
dhcp_start(): Cannot use this netif with DHCP: MTU is too small
E (2485) esp_netif_lwip: dhcp client start failed
I (2485) eth_example: Ethernet Link Up
I (2495) eth_example: Ethernet HW Addr 24:0a:c4:4e:af:cb
I (2495) ethernetif_init: called...
W (2505) ethernet_low_level_init: set ethernet mtu to 1500!
I (2515) eth_example: Ethernet Started
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
...
```

## Other items

- [ ] [sdkconfig file](https://github.com/espressif/esp-idf/files/7176742/sdkconfig.txt) (attach the sdkconfig file from your project folder)
- [ ] [elf file](https://github.com/espressif/esp-idf/files/7176752/ethernet_basic.elf.txt) in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

dabbinavo, 2021-09-16:
Ok, just saw that in commit 03a6c4975c7fd7d0267f26b93f43b5b391ad2de4 all `<phy>_negotiate()` functions in its corresponding `esp_eth_phy_<phy>.c` were modified. Except for the phy `ksz8041`. The hardware I am working on uses a `ksz8061`. For that phy, I just copied the sources of `ksz8061` and slightly modified it. Applying the changes from 03a6c4975c7fd7d0267f26b93f43b5b391ad2de4 (setting status to link down, extending delays, removing update of link/duplex/speed) to my sources of the phy `ksz8061` fixed the issue.

suda-morris, 2021-09-16:
I'm sorry for this kind of issue, we've backported several fixes from master to other release branches, this is somehow being left over. Thankfully you found this!