espressif/esp-idf #7571: Strange behavior after watchdog task reset

moefear85, 2021-09-16:
I'm using ESP-IDF 4.3.1 with VScode 1.60.1 on Windows 11. I have the following code at the top of my app_main():

`esp_reset_reason_t` reason = esp_reset_reason();
printf("\nReset Reason: %d. Execution halted.", reason);
fflush(stdout);
if (reason != ESP_RST_POWERON)
{
    sleep(10000);
}
`
Then I enter a tight loop reading the adc and printing it using:
`printf("\nRaw: %d\tVoltage: %dmV", adc_reading, voltage);`

However, the tight loop eventually triggers the task watchdog. So I see the following errors on restart:

> E (20421) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (20421) task_wdt:  - IDLE (CPU 0)
> E (20421) task_wdt: Tasks currently running:
> E (20421) task_wdt: CPU 0: main
> E (20421) task_wdt: Print CPU 0 (current core) backtrace

However, the above reset reason code never gets executed. It never shows any output. It jumps straight back to the loop which again triggers the watchdog, and it keeps doing so, even though it's supposed to print out a reset reason then go to sleep (which works however for other reset reasons (it always prints the reason and sleeps if it's not poweron).
Although it occured to me the chip might not necessarily be resetting after the error message, that still doesn't explain why it isn't printing out the reason nor sleeping if it's anything besides POWERON, yet still executing the rest of the app_main() function.

boarchuz, 2021-09-16:
https://github.com/espressif/esp-idf/blob/220590d599e134d7a5e7f1e683cc4550349ffbf8/components/esp_system/Kconfig#L381

ESP-Marius, 2021-09-17:
Hi, @moefear85 

As @boarchuz already mentioned, the task WDT doesnt reset the chip by default. It can be enabled by setting `ESP_TASK_WDT_PANIC` in menuconfig.

> Although it occured to me the chip might not necessarily be resetting after the error message, that still doesn't explain why it isn't printing out the reason nor sleeping if it's anything besides POWERON, yet still executing the rest of the app_main() function.

Not sure I completely understand you here. As you suspected, the chip is not resetting, but if the chip is not resetting it will never run `app_main` again. It will just keep running whatever it was running before the WDT triggered, i.e. your tight loop.