espressif/esp-idf #7572: MQTT WebSocket connection closed prematurely

olicooper, 2021-09-16:
## Environment

- Development Kit:      Lolin 32 lite
- Module or chip used:  ESP32
- IDF version Arduino 1.0.6 (IDF 4.4.3)
- Operating System:     Windows
- Using an IDE?: PlatformIO
- Power Supply:         USB

## Problem Description

I'm trying to update the MQTT connection URI. The Server logs show that the WebSocket connection is now being gracefully disconnected. 

The process is as follows:
- connect with WebSockets
- wait 15 seconds
- disconnect and wait for disconnect event
- stop mqtt, update the uri and start again

### Actual Behavior

Server logs:
```log
13:24:56.584+01:00 [DBG][44][Microsoft.AspNetCore.Server.Kestrel.Transport.Sockets] Connection id "0HMBP19PFQR4T" received FIN.
13:24:56.585+01:00 [DBG][52][Microsoft.AspNetCore.Server.Kestrel.Transport.Sockets] Connection id "0HMBP19PFQR4T" sending FIN because: "The client closed the connection."
13:24:56.585+01:00 [DBG][52][Microsoft.AspNetCore.Server.Kestrel] Connection id "0HMBP19PFQR4T" disconnecting.
13:24:56.586+01:00 [DBG][8][Microsoft.AspNetCore.Http.Connections.Internal.Transports.WebSocketsTransport] Socket connection closed prematurely.
System.Net.WebSockets.WebSocketException (0x80004005): The remote party closed the WebSocket connection without completing the close handshake.
   at System.Net.WebSockets.ManagedWebSocket.ThrowIfEOFUnexpected(Boolean throwOnPrematureClosure)
   at System.Net.WebSockets.ManagedWebSocket.EnsureBufferContainsAsync(Int32 minimumRequiredBytes, CancellationToken cancellationToken, Boolean throwOnPrematureClosure)
   at System.Net.WebSockets.ManagedWebSocket.ReceiveAsyncPrivate[TWebSocketReceiveResultGetter,TWebSocketReceiveResult](Memory`1 payloadBuffer, CancellationToken cancellationToken, TWebSocketReceiveResultGetter resultGetter)
   at Microsoft.AspNetCore.Http.Connections.Internal.Transports.WebSocketsServerTransport.StartReceiving(WebSocket socket)
13:24:56.586+01:00 [DBG][8][Microsoft.AspNetCore.Http.Connections.Internal.Transports.WebSocketsTransport] Waiting for the application to finish sending data.
13:24:56.593+01:00 [DBG][52][Microsoft.AspNetCore.Http.Connections.Internal.Transports.WebSocketsTransport] Socket closed.
```

### Code to reproduce this issue

```cpp
#define CORE_DEBUG_LEVEL ARDUHAL_LOG_LEVEL_VERBOSE

#include <Arduino.h>
#include <WiFi.h>
#include "sdkconfig.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "mqtt_client.h"
#include "esp_event.h"
#include "esp_log.h"

static const char *TAG = "MQTT_EXAMPLE";
esp_mqtt_client_handle_t mqttClient;
bool update = true;
bool setConfig = false;
bool canSetConfig = false;

static void log_error_if_nonzero(const char *message, int error_code)
{
    if (error_code != 0) {
        ESP_LOGE(TAG, "Last error %s: 0x%x", message, error_code);
    }
}
static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    ESP_LOGD(TAG, "Event dispatched from event loop base=%s, event_id=%d", base, event_id);
    esp_mqtt_event_handle_t event = (esp_mqtt_event_handle_t)event_data;
    switch ((esp_mqtt_event_id_t)event_id) {
    case MQTT_EVENT_CONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
        break;
    case MQTT_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
        if (setConfig) {
            ESP_LOGI(TAG, "MQTT SET URI 1");
            setConfig = false;
            canSetConfig = true;
        }
        break;
    case MQTT_EVENT_ERROR:
        ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
        if (event->error_handle->error_type == MQTT_ERROR_TYPE_TCP_TRANSPORT) {
            log_error_if_nonzero("reported from esp-tls", event->error_handle->esp_tls_last_esp_err);
            log_error_if_nonzero("reported from tls stack", event->error_handle->esp_tls_stack_err);
            log_error_if_nonzero("captured as transport's socket errno",  event->error_handle->esp_transport_sock_errno);
            ESP_LOGI(TAG, "Last errno string (%s)", strerror(event->error_handle->esp_transport_sock_errno));

        }
        break;
    default:
        ESP_LOGI(TAG, "Other event id:%d", event->event_id);
        break;
    }
}

void setup() {
    WiFi.mode(WIFI_MODE_STA);
    WiFi.begin("ssid", "pass");
    if (WiFi.waitForConnectResult() == WL_CONNECTED) {
        ESP_LOGI(TAG, "WIFI CONNECTED");

        esp_mqtt_client_config_t mqtt_cfg = {};
        mqtt_cfg.uri = "ws://user:pass@localhost:5000/mqtt";
        mqtt_cfg.client_id = "clientid";
        mqtt_cfg.buffer_size = 256;
        mqtt_cfg.network_timeout_ms = 5000;
        mqtt_cfg.keepalive = 60;
        mqttClient = esp_mqtt_client_init(&mqtt_cfg);
        esp_mqtt_client_register_event(mqttClient, (esp_mqtt_event_id_t)ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
        esp_mqtt_client_start(mqttClient);
    }
}

void loop() {
    if (update) {
        vTaskDelay(15000);
        setConfig = true;
        update = false;
        ESP_LOGI(TAG, "MQTT SET URI 0");
        
        esp_mqtt_client_disconnect(mqttClient);
    }
    if (canSetConfig) {
        ESP_LOGI(TAG, "MQTT SET URI 2");
        canSetConfig = false;
        
        esp_mqtt_client_stop(mqttClient);
        esp_mqtt_client_set_uri(mqttClient, "mqtt://user:pass@localhost:1883");
        esp_mqtt_client_start(mqttClient);
    }
}
```

## Debug Logs

```log
[I][main.cpp:127] setup(): WIFI CONNECTED
[D][main.cpp:50] mqtt_event_handler(): Event dispatched from event loop base=MQTT_EVENTS, event_id=7
[I][main.cpp:105] mqtt_event_handler(): Other event id:7
[D][main.cpp:50] mqtt_event_handler(): Event dispatched from event loop base=MQTT_EVENTS, event_id=1
[I][main.cpp:56] mqtt_event_handler(): MQTT_EVENT_CONNECTED
[I][main.cpp:147] loop(): MQTT SET URI 0
[D][main.cpp:50] mqtt_event_handler(): Event dispatched from event loop base=MQTT_EVENTS, event_id=2
[I][main.cpp:70] mqtt_event_handler(): MQTT_EVENT_DISCONNECTED
[I][main.cpp:72] mqtt_event_handler(): MQTT SET URI 1
[I][main.cpp:152] loop(): MQTT SET URI 2
[D][main.cpp:50] mqtt_event_handler(): Event dispatched from event loop base=MQTT_EVENTS, event_id=7
[I][main.cpp:105] mqtt_event_handler(): Other event id:7
[D][main.cpp:50] mqtt_event_handler(): Event dispatched from event loop base=MQTT_EVENTS, event_id=1
[I][main.cpp:56] mqtt_event_handler(): MQTT_EVENT_CONNECTED
```

david-cermak, 2021-09-24:
Hi @olicooper 

websocket close handshake was not supported on `v3.3`. Could you please switch to `v4.3` and test again?
It has been added in https://github.com/espressif/esp-idf/commits/5e9f8b52e7a87371370205a387b2d94e5ac6cbf9

olicooper, 2021-10-08:
Not able to switch to 4.3 right now because 3.3 is the version that Arduino 1.0.6 uses and this is what I'm using in the project I am working on. I will try to set up a IDF test project to test soon.

david-cermak, 2021-11-09:
Could you please post the packet capture for the connection closure then?

I've tested with the official broker implementation of `MQTTnet` and couldn't recreate this issue.
The broker log is the same with and without the patch and just says the connection been closed cleanly:
```
dbug: MQTTnet.Server.Scripting.PythonIOStream[0]
      Client 'ESP32_5f2CC4' is now disconnected (type = clean).
info: Microsoft.AspNetCore.Hosting.Diagnostics[2]
      Request finished HTTP/1.1 GET http://192.168.33.27:80/mqtt - - - 101 - - 4483.2060ms
```

But from the packet capture, I can see the connection wasn't closed according to WS handshake:
```
No.     Time           Source                Destination           Protocol Length Info
    236 21.384827      192.168.32.248        192.168.33.27         WebSocket 56     WebSocket Binary [FIN] [MASKED]
    237 21.385211      192.168.33.27         192.168.32.248        WebSocket 58     WebSocket Connection Close [FIN]
    238 21.390346      192.168.32.248        192.168.33.27         TCP      54     52988 â†’ 80 [RST, ACK] Seq=416 Ack=292 Win=28694 Len=0
```
When applying the patch to `send_disconnect_msg()` the ws close handshake looks correct:
```
 No.     Time           Source                Destination           Protocol Length Info
   1110 26.854812      192.168.32.248        192.168.33.27         WebSocket 56     WebSocket Binary [FIN] [MASKED]     (MQTT disconnect: 0000  e0 00    ..)
   1111 26.855566      192.168.33.27         192.168.32.248        WebSocket 58     WebSocket Connection Close [FIN]
   1112 26.875038      192.168.32.248        192.168.33.27         WebSocket 60     WebSocket Connection Close [FIN] [MASKED]
   1113 26.876670      192.168.33.27         192.168.32.248        TCP      54     80 â†’ 63104 [FIN, ACK] Seq=292 Ack=422 Win=64379 Len=0
   1114 26.880998      192.168.32.248        192.168.33.27         TCP      54     63104 â†’ 80 [ACK] Seq=422 Ack=293 Win=5452 Len=0
```

I'd like to understand how this "official" broker implementation differs from your own server implementation. Did you add any specific transport hooks? From the log, it looks like the server is actually sending some data after the ws closure.

Also, I've noticed this warning in the broker console:
```
The official MQTT server implementation of MQTTnet
Copyright (c) 2017-2020 The MQTTnet Team
https://github.com/chkr1011/MQTTnet

 ! THIS IS AN ALPHA VERSION! IT IS NOT RECOMMENDED TO USE IT FOR ANY DIFFERENT PURPOSE THAN TESTING OR EVALUATING!
```

Maksons, 2021-11-12:
@david-cermak  
Is this close behavior implemented only for ws_client ?
Could it be also implemented for ws server in esp_http_server ?

david-cermak, 2021-11-12:
@Maksons The close handshake is implemented in `tcp_transport` component and currently used only in `ws_client`. We'll probably add it to `mqtt_client` too.

> Could it be also implemented for ws server in esp_http_server ?

websocket server in the `esp_http_server` doesn't use this component, and is designed to interact with user handlers on very low level, so I'd recommend implementing this in user space by handling control packets.

david-cermak, 2021-11-19:
>  ! THIS IS AN ALPHA VERSION! IT IS NOT RECOMMENDED TO USE IT FOR ANY DIFFERENT PURPOSE THAN TESTING OR EVALUATING!

I'm considering closing this issue as *won't do*, at least until the official MQTT server implementation of MQTTnet, reaches maturity or some other broker requires this. The patch posted above solves the problem in my case, but it's apparently not a universal fix and, more importantly, it would reduce user experience with other standard brokers.

@olicooper Any comment?

Alvin1Zhang, 2021-11-22:
@olicooper Thanks for reporting, would you please help share if any updates for the issue? Thanks.

olicooper, 2021-11-22:
The only thing I can think of that may be different in my MQTTnet instance is that I publish a retained device 'online' status on connect and disconnect (`IMqttServerClientDisconnectedHandler`) events. It may be possible that the server publishes this message before the connection close event is fully handled? My MQTTnet instance is running on ASP.NET 5 in a Debian environment if that helps. I would argue that the software is beyond alpha stage at this point and that the developer is just limiting their liability.

As I don't think this issue is critical to the functioning of MQTT then I suppose it can be left for a while. Is it possible that this could cause memory leaks if left unhandled though? I have noticed a slight reduction in memory on each reconnect attempt.

I will try to get those Wireshark logs posted and do more testing in the mean time. Thanks for your help!

david-cermak, 2021-11-23:
> It may be possible that the server publishes this message before the connection close event is fully handled?

Maybe it could be the case, we will see from the wireshark capture.

>  I would argue that the software is beyond alpha stage

Yes, the ws close handshake is really *by the book* and makes perfect sense to be server initiated, as the server knows when the connection of the underlying transport closes (unlike e.g. for `mosquitto`s, where the ws close is implemented but expected to be client initiated and thus may cause races -- client cannot assure that the mqtt close and ws close packets arrive exactly one after another), but it is in fact quite impractical and effectively delays the closure.

>  Is it possible that this could cause memory leaks if left unhandled though?

You mean memory leaks in the embedded target? No, I don't think so, the connection (socket) is closed in both cases (without the handshake it's just closed sooner)

olicooper, 2021-11-26:
I am happy for you to close this issue for now as it sounds like it isn't a high priority issue, especially since the error is handled gracefully on both ends of the connection. I will try to post the wireshark logs when I can.
Thanks again for your investigation.

Alvin1Zhang, 2022-03-08:
@olicooper Thanks for reporting and sharing the updates, would you please help share if any updates for the issue? Could this issue be closed now? Thanks.

Alvin1Zhang, 2022-03-29:
Thanks for reporting, would close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2022-03-29 by Alvin1Zhang