espressif/esp-idf #7576: power_save example runs abnormally on esp32s2, when use DFS.

DaMiBear, 2021-09-17:
## Environment

- Module or chip used:  ESP32S2-MINI-1 & ESP32-WROOM
- IDF version : 4.3
   
- Build System:  idf.py

- Operating System: Windows

## Problem Description

I'm trying to use power_save example in IDF, I only modified the SSID and password in menuconfig and set the WiFi listen interval to 1. After flash into esp32s2-mini-1 and running for a period of time, the following errors will appear and restart several times. There is no problem putting the same code on esp32-wroom.
And I have tried to cancel the support for power management option on esp32s2, then this kind of error will not appear. Is it because esp32s2 is not supported? Or is there something wrong with my esp32s2?

## Debug Logs

```

I (530) wifi:new:<5,1>, old:<1,0>, ap:<255,255>, sta:<5,1>, prof:1
I (530) wifi:state: init -> auth (b0)
I (530) wifi:state: auth -> assoc (0)
I (530) wifi:state: assoc -> run (10)
W (540) wifi:<ba-add>idx:0 (ifx:0, 2e:15:e1:25:d2:22), tid:5, ssn:0, winSize:64
W (590) wifi:<ba-add>idx:1 (ifx:0, 2e:15:e1:25:d2:22), tid:0, ssn:2, winSize:64
I (590) wifi:connected with 109-2.4G, aid = 1, channel 5, 40U, bssid = 2e:15:e1:25:d2:22
I (590) wifi:security: WPA2-PSK, phy: bgn, rssi: -51
I (590) wifi:pm start, type: 1

I (610) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1390) esp_netif_handlers: sta ip: 192.168.123.100, mask: 255.255.255.0, gw: 192.168.123.1
I (1390) power_save: got ip: 192.168.123.100

abort() was called at PC 0x40024853 on core 0
0x40024853: lock_acquire_generic at E:/Espressif/esp-idf/components/newlib/locks.c:138

Backtrace:0x4002b70f:0x3ffd3050 0x4002c32d:0x3ffd3070 0x4003316a:0x3ffd3090 0x40024853:0x3ffd3100 0x4002499d:0x3ffd3130 0x4001a245:0x3ffd3150 0x40000baa:0x3ffd3170 0x40000a65:0x3ffd3220 0x4001a428:0x3ffd3270 0x4002daa1:0x3ffd32a0 0x40024de9:0x3ffd32c0 0x4002d8c2:0x3ffd32e0 0x4002f1ad:0x3ffd3300
0x4002b70f: panic_abort at E:/Espressif/esp-idf/components/esp_system/panic.c:356

0x4002c32d: esp_system_abort at E:/Espressif/esp-idf/components/esp_system/system_api.c:112

0x4003316a: abort at E:/Espressif/esp-idf/components/newlib/abort.c:46

0x40024853: lock_acquire_generic at E:/Espressif/esp-idf/components/newlib/locks.c:138

0x4002499d: _lock_acquire_recursive at E:/Espressif/esp-idf/components/newlib/locks.c:166

0x4002daa1: vTaskStepTick at E:/Espressif/esp-idf/components/freertos/tasks.c:2855 (discriminator 1)

0x40024de9: vApplicationSleep at E:/Espressif/esp-idf/components/esp_pm/pm_impl.c:647

0x4002d8c2: prvIdleTask at E:/Espressif/esp-idf/components/freertos/tasks.c:3882

0x4002f1ad: vPortTaskWrapper at E:/Espressif/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 8a2caa1c0a5cc90d

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x400241b0
0x400241b0: esp_restart_noos at E:/Espressif/esp-idf/components/esp32s2/system_api_esp32s2.c:109 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1800
load:0x4004c000,len:0x9e8
load:0x40050000,len:0x2b70
SHA-256 comparison failed:
Calculated: ac45cfcb2454d7e8d7c849d8aa0d37e69fb199f50323370616c52a4d38e094c9
[... 200 lines cut ...]
I (532) wifi:state: init -> auth (b0)
I (532) wifi:state: auth -> assoc (0)
I (532) wifi:state: assoc -> run (10)
W (542) wifi:<ba-add>idx:0 (ifx:0, 2e:15:e1:25:d2:22), tid:5, ssn:0, winSize:64
W (592) wifi:<ba-add>idx:1 (ifx:0, 2e:15:e1:25:d2:22), tid:0, ssn:2, winSize:64
I (592) wifi:connected with 109-2.4G, aid = 1, channel 5, 40U, bssid = 2e:15:e1:25:d2:22
I (592) wifi:security: WPA2-PSK, phy: bgn, rssi: -52
I (592) wifi:pm start, type: 1

I (602) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (632) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x000000ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1800
load:0x4004c000,len:0x9e8
load:0x40050000,len:0x2b70
SHA-256 comparison failed:
Calculated: ac45cfcb2454d7e8d7c849d8aa0d37e69fb199f50323370616c52a4d38e094c9
Expected: 922e3db13a75acd3c1eca495d3ae055e9c7cd03921b3a185ffb2add79d9f9e69
Attempting to boot anyway...
entry 0x4004c1ec

```

DaMiBear, 2021-09-19:
In menuconfig, change the  RTC clock source from `internal 90khz RC oscillator` to `internal 8MHz oscillator, by dividing by 256 (~ 32kHz)`. This error can be eliminated.

The log shows that there is a problem with the RTC clock, but I'm not very clear about the specific reason.

lennyJL, 2022-04-14:
Hi @DaMiBear. How long does `power_save example` take to reproduce this issue?

I tried with the following env:
- module: ESP32-S2-Saola-1_v1.2
- my cell phone: s2 conect my cell phone via WIFI (I don't have a router at hand)
- IDF ver: latest release/v4.3
- code: power_save example

menuconfig are:
- (1) WiFi listen interval
- RTC clock source (Internal 90kHz RC oscillator)

But I can not reproduce this issue with the code running for over two hours. Can you try your code on the latest release/v4.3? Or can you tell me what other specific configuration is required?

DaMiBear, 2022-04-15:
> Hi @DaMiBear. How long does `power_save example` take to reproduce this issue?
>
> I tried with the following env:
>
> * module: ESP32-S2-Saola-1_v1.2
> * my cell phone: s2 conect my cell phone via WIFI (I [...]

I'm sorry, I don't have an esp32s2 board during this time. It's been too long. I can't remember details, but it should appear in a few seconds, definitely not two hours . 
When I get esp32s2, I will try the latest IDF version.

xiaojie76, 2022-04-17:
Hi @DaMiBear, I also can not reproduce this issue in the latest v4.3 version,  please provide a more detailed idf version (commit id) with this issue, thank you.

DaMiBear, 2022-04-21:
Hi, @xiaojie76 @lennyJL  git log output：

> commit c9646ff0beffc86d2c6d1bfbad34da16e328e0e3 (HEAD, tag: v4.3)
Author: He Yin Ling <heyinling@espressif.com>
Date:   Thu Jun 10 14:20:45 2021 +0800
versions: Update version to 4.3.0

My board from [here](https://www.espressif.com/zh-hans/news/esp32-s2-audio-board)
I reproduce the same problem just now. 
Only modified the SSID, password and WiFi listen interval in menuconfig.
About 20s after this line of log
`I (1391) power_save: got ip: 192.168.2.221`
The following is the complete log output：
```
PS E:\Espressif\ESP_projects\power_save> E:/Espressif/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe E:\Espressif\esp-idf\tools\idf.py -p COM4 monitor
Executing action: monitor
Running idf_monitor in directory e:\espressif\esp_projects\power_save
Executing "E:\Espressif\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe E:/Espressif/esp-idf/tools/idf_monitor.py -p COM4 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- e:\espressif\esp_projects\power_save\build\power_save.elf -m 'E:\Espressif\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe' 'E:\Espressif\esp-idf\tools\idf.py' '-p' 'COM4'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM4 instead...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1800
load:0x4004c000,len:0x9e8
load:0x40050000,len:0x2b70
SHA-256 comparison failed:
Calculated: 0f9c9466764595ff1690cc0ad4f4ff011bb6a72fedb2c30842e61dfe89110eba
Expected: ad758c435914303675eb2ec0df54a5dfef51ebc24d33f74c64811f89ef2fb42b
Attempting to boot anyway...
entry 0x4004c1ec
I (40) boot: ESP-IDF v4.3 2nd stage bootloader
I (40) boot: compile time 22:33:40
I (40) boot: chip revision: 0
I (42) boot.esp32s2: SPI Speed      : 80MHz
I (47) boot.esp32s2: SPI Mode       : DIO
I (52) boot.esp32s2: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=16f80h ( 94080) map
I (128) esp_image: segment 1: paddr=00026fa8 vaddr=3ffc8bf0 size=03ed0h ( 16080) load
I (132) esp_image: segment 2: paddr=0002ae80 vaddr=40022000 size=05198h ( 20888) load
I (139) esp_image: segment 3: paddr=00030020 vaddr=40080020 size=6c350h (443216) map
I (237) esp_image: segment 4: paddr=0009c378 vaddr=40027198 size=11a4ch ( 72268) load
I (257) esp_image: segment 5: paddr=000addcc vaddr=50000000 size=00010h (    16) load
I (269) boot: Loaded app from partition at offset 0x10000
I (269) boot: Disabling RNG early entropy source...
I (281) ca�f�~��nstruction cache       : size 8KB, 4Ways, cache line size 32Byte
I (281) cpu_start: Pro cpu up.
I (295) cpu_start: Pro cpu start user code
I (295) cpu_start: cpu freq: 160000000
I (295) cpu_start: Application information:
I (299) cpu_start: Project name:     power_save
I (305) cpu_start: App version:      1
I (309) cpu_start: Compile time:     Apr 21 2022 22:32:46
I (315) cpu_start: ELF file SHA256:  d5482b992729a62b...
I (321) cpu_start: ESP-IDF:          v4.3
I (326) heap_init: Initializing. RAM available for dynamic allocation:
I (333) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (339) heap_init: At 3FFD11E8 len 0002AE18 (171 KiB): DRAM
I (346) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
[... 495 lines cut ...]
I (432) wifi_init: tcp rx win: 5744
I (442) wifi_init: tcp mss: 1440
I (442) wifi_init: WiFi IRAM OP enabled
I (442) wifi_init: WiFi RX IRAM OP enabled
I (442) wifi_init: WiFi SLP IRAM OP enabled
I (452) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
I (542) wifi:mode : sta (7c:df:a1:97:42:10)
I (542) wifi:enable tsf
I (542) power_save: esp_wifi_set_ps().
I (542) wifi:Set ps type: 1

I (542) wifi:new:<4,1>, old:<1,0>, ap:<255,255>, sta:<4,1>, prof:1
I (542) wifi:state: init -> auth (b0)
I (552) wifi:state: auth -> assoc (0)
I (552) wifi:state: assoc -> run (10)
I (602) wifi:connected with 411, aid = 1, channel 4, 40U, bssid = 28:d1:27:8f:fa:77
I (602) wifi:security: WPA2-PSK, phy: bgn, rssi: -36
I (602) wifi:pm start, type: 1

W (612) wifi:<ba-add>idx:0 (ifx:0, 28:d1:27:8f:fa:77), tid:6, ssn:2, winSize:64
I (632) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (1262) wifi:<ba-add>idx:1 (ifx:0, 28:d1:27:8f:fa:77), tid:0, ssn:3, winSize:64
I (1392) esp_netif_handlers: sta ip: 192.168.2.221, mask: 255.255.255.0, gw: 192.168.2.1
I (1392) power_save: got ip: 192.168.2.221
```

xiaojie76, 2022-04-22:
@DaMiBear Unfortunately I still can't reproduce the issue on this version. Although it is not the same board, but from your log, it should be caused by software, can you provide power_save.bin, power_save.elf and sdkconfig?

DaMiBear, 2022-04-22:
Hi, @xiaojie76 
Here are power_save.bin, power_save.elf and sdkconfig:
[issues_files.zip](https://github.com/espressif/esp-idf/files/8536344/issues_files.zip)

xiaojie76, 2022-04-22:
@DaMiBear  The real reason for the issue that it is asserted here in `vTaskStepTick`:
`configASSERT( ( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime );`
It appears to be sleeping longer than expected.
Can you get the value of `xNextTaskUnblockTime` and `(xTickCount + xTicksToJump) ` `if( xTickCount + xTicksToJump ) > xNextTaskUnblockTime )` ?

DaMiBear, 2022-04-22:
@xiaojie76  I'm new to RTOS. I tried add the code in `vTaskStepTick`:
```
if( ( xTickCount + xTicksToJump ) > xNextTaskUnblockTime )
{
	printf("xNextTaskUnblockTime: %d\n", xNextTaskUnblockTime);
	printf("(xTickCount + xTicksToJump): %d\n", (xTickCount + xTicksToJump));
}
```
But it`s not work, and reboots are more frequent.😟
It seems that debugging cannot be performed without JTAG.
Sorry about my poor knowledge. Can you please provide some ways?

xiaojie76, 2022-04-22:
@DaMiBear use `ets_printf`

DaMiBear, 2022-04-22:
@xiaojie76 Thanks for you help.
This is partial log output:
First reboot:
```
I (1386) power_save: got ip: 192.168.2.221
W (10466) rESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
```
Second:
```
I (1387) power_save: got ip: 192.168.2.221
W (1387) wifi:<ba-add>idx:1 (ifx:0, 28:d1:27:8f:fa:77), tid:0, ssn:2, winSize:64
W (3757) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
xNextTaskUnblockTime: 351
(xTickCount + xTicksToJump): -1913021300
```
Third:
```
W (669) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000
xNextTaskUnblockTime: 51
(xTickCount + xTicksToJump): 1678990186
```
Full log:
[idf_log.txt](https://github.com/espressif/esp-idf/files/8537622/idf_log.txt)

xiaojie76, 2022-04-22:
@DaMiBear For more detailed information, temporarily define `sleep_time_us` and `sleep_us` in vApplicationSleep (pm_impl.c) as global variables and print them in `vTaskStepTick`.
`
extern int64_t sleep_time_us;
extern int64_t slept_us;
void vTaskStepTick( const TickType_t xTicksToJump )
{
	/* Correct the tick count value after a period during which the tick
	was suppressed.  Note this does *not* call the tick hook function for
	each stepped tick. */
	taskENTER_CRITICAL(&xTaskQueueMutex);
	if( ( xTickCount + xTicksToJump ) > xNextTaskUnblockTime ){
		ets_printf("xTickCount:%u, xTicksToJump:%u, xNextTaskUnblockTime: %u\n", xTickCount, xTicksToJump, xNextTaskUnblockTime);
		ets_printf("sleep_time_us:%llu, slept_us:%llu\n\n", sleep_time_us, slept_us);
	};
	xTickCount += xTicksToJump;
	traceINCREASE_TICK_COUNT( xTicksToJump );
	taskEXIT_CRITICAL(&xTaskQueueMutex);
}
`

DaMiBear, 2022-04-22:
@xiaojie76
```
xTickCount:286, xTicksToJump:1564845560, xNextTaskUnblockTime: 301
sleep_time_us:150000, slept_us:1889741912
```
```
xTickCount:543, xTicksToJump:1203939361, xNextTaskUnblockTime: 551
sleep_time_us:80000, slept_us:600284726
```
[full_idf_log.txt](https://github.com/espressif/esp-idf/files/8538337/idf_log.txt)

xiaojie76, 2022-04-22:
@DaMiBear  There is a strange place:
xTicksToJump = slept_ticks = slept_us / (10000),
Obviously your result is wrong, can you find the reason?

xiaojie76, 2022-04-29:
@DaMiBear  Do you have any progress on this issue? Because even using the bin file you gave cannot be reproduced in my environment, it may be a hardware problem.
Can you confirm a few questions:
1. Your result does not satisfy `xTicksToJump = slept_ticks = slept_us / (10000)` , you can print the value of slept_ticks to see what went wrong.
2. In your result, `sleep_us >>sleep_time_us`, the sleep time does not match the set time. You can test the timer wakeup separately, the code is as follows:

3. Is there still with this issue when WiFi listen interval using the default configuration?

DaMiBear, 2022-04-30:
@xiaojie76 I apologize for not replying to you in time because of the busyness of college.
I came up with this issue to confirm whether it is my board problem or the software. It does look like a hardware problem for now😐.

1. I tried the idf-v.4.3.2 commit 8bf14a9238329954c7c5062eeeda569529aedf75 (HEAD, tag: v4.3.2).
It still keep rebooting(the cycle is about 90s), but there is no `ets_printf` output(I already changed the `vTaskStepTick`  discussed above)

```
I (1366) esp_netif_handlers: sta ip: 192.168.2.221, mask: 255.255.255.0, gw: 192.168.2.1
I (1366) power_save: got ip: 192.168.2.221
W (2476) wifi:<ba-add>idx:2 (ifx:0, 2e:15:e1:25:d2:22), tid:5, ssn:0, winSize:64
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x19ac
load:0x4004c000,len:0x9e4
load:0x40050000,len:0x2f04
entry 0x4004c1e8
I (21) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
```

2. Back to idf-v4.3 c9646ff0beffc86d2c6d1bfbad34da16e328e0e3
It still with this issue when WiFi listen interval using the default configuration
```
xTickCount:352, xTicksToJump:2613587116, xNextTaskUnblockTime: 402
sleep_time_us:46439, slept_us:995172984, slept_ticks:2613587116
```
3. This the output after I use the code you provide.
````
I (368) cpu_start: Starting scheduler on PRO CPU.
Entering light sleep for 2000000 us
Woke up from light sleep, sleep time: 1999980 us
```
But It seems work well?

xiaojie76, 2022-05-05:
@DaMiBear It looks more like your board has a problem, maybe you can try another board, if you bought it from the official online, use the after-sales service.

DaMiBear, 2022-05-08:
@xiaojie76  I‘ve already switched to esp32 on hand. Thank you very much for your patience and help. Have a great day!😃

closed 2022-05-08 by DaMiBear