espressif/esp-idf #7577: Large data transfer on BLE works differently on iOS than Android

silardg, 2021-09-17:
My code for reading the characteristic from the ESP32, which is longer than 22 bytes is as follows:
```
case 1: 
        {
            ESP_LOGI(TAG, "GATT_READ_EVT, conn_id %d, trans_id %d, handle %d\n", param->read.conn_id, param->read.trans_id, param->read.handle);

            // store response
            esp_gatt_rsp_t rsp;        

            // create space for the rsp
            memset(&rsp, 0, sizeof(esp_gatt_rsp_t));                                                                                            

            // dynamicjsondocument to hold the json
            DynamicJsonDocument json(1024);        

            // get the current credentials, and deserialize to json                               
            deserializeJson(json, Config::preferences->getString(CREDENTIALS_KEY, EMPTY_JSON));

            // keeping the password secret, pssttt
            json.remove("password");

            // convert the username to a string, so we can use it later to add it to the response
            std::string username = json["username"];

            // clear the json
            json.clear();

            // deserialize the network config part
            deserializeJson(json, Config::preferences->getString(NETWORK_CONFIG_KEY, EMPTY_JSON));         

            // add the previous decoded username                                
            json.add(username);

            // assign it a value
            json["username"] = username;

            // create a response string
            std::string test = "";

            // serialize it back to string
            serializeJson(json, test);

            // clear the json
            json.clear();

            // check the max length of our packet
            uint16_t payload_max_length = test.length();

            // is_long and offfset is always 0, when we are starting to read, that is why we are setting the position to 0
            if(!param->read.is_long && param->read.offset == 0) { currentReadPosition = 0; }

            // if we have more than 22 bytes left over, set the length to max value
            // if we have less than 22, we set the length to the max-currentpos
            if(payload_max_length - currentReadPosition >= 22) { rsp.attr_value.len = 22; } else { rsp.attr_value.len = payload_max_length - currentReadPosition; } 
            
            // loop through and assign it
            for (size_t i=0; i<rsp.attr_value.len; i++)
            {
                // assign it, depending on the position 
                // assigning is done with converting char to uint8_t, while adapting the offset
                rsp.attr_value.value[i] = (uint8_t)( test[i+param->read.offset]);

                // increase the position, this is where we can track the length of the entire packets success
                currentReadPosition++;
            }

            // set the handler
            rsp.attr_value.handle = param->read.handle;

            // send out the response
            esp_ble_gatts_send_response(gatts_if, param->read.conn_id, param->read.trans_id,
                                        ESP_GATT_OK, &rsp);

            break;
        }
```

I'm checking if the length is more than 22 bytes, if it is, send the packets until the size is less than 22 bytes, meaning it's the last packet I have to send. This works with android and the nRF Connect app, but it won't on iOS. 

The iOS is just showing one packet, the beginning of my json with a length of 22 bytes.

xiewenxiang, 2021-09-29:
can you provide a demo that can be compiled.

xulongzhuang, 2021-09-30:
I think this problem may be due to the MTU setting. the difference between nrfconnect in Android phones and IOS phones is that the latter exchanges MTU by default (I tested it on iPhone11 like this). The MTU of an IOS phone may be a number greater than 500. 
So if the IOS mobile phone reads data from the esp32 and the data length obtained is 22 but it is much smaller than the negotiated MTU value, it will consider that the data read is complete and will not use the` Read blob request` to read the remaining data, Therefore, nrfconnect on IOS phones only obtains a single data packet with a length of 22
You should consider swapping MTU and change your code logic, or you can also use the `esp_ble_gatt_set_local_mtu()` function to set the MTU value of esp32 to 23