espressif/esp-idf #7578: ftm : ESP32-C3 (rev.2) does not answer ftm requests

Sch-LikA, 2021-09-17:
## Environment

- Development Kit:      [ESP32-C3-DevKitM-1]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1.6]
- Module or chip used:  [ESP32-C3-MINI-1]
- IDF version (run ``git describe --tags`` to find it): 4.3.1
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp-2021r1-8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Yes (VSCode + esp-idf extension)]
- Power Supply:         [USB]

## Problem Description

I am testing the barebone "ftm" example between an EPS32-S2 and an ESP32-C3. The ESP32-S2 works as expected both in Initiator mode and AP mode. 

The ESP32-C3 only works in Initiator mode. I tested with both an Android smartphone and an ESP32-S2 and I get the same result. In AP mode, the ESP32-C3 advertises the FTM capability but never answers the queries, coming from both an ESP32-S2 and an Android smartphone.

The other way around, with an ESP32-S2 as AP and the ESP32-C3 as initiator, I get results.

### Expected Behavior

In AP mode, the ESP32-C3 should not only advertise FTM capability but also answer FTM queries.

### Actual Behavior

In AP mode, the ESP32-C3 only advertises FTM capability but does not answer FTM queries.

### Steps to reproduce

1. compile & flash "ftm" example on both an ESP32-C3 and an ESP32-S2
2. On the ESP32-C3, start an ap as in the example : `ap FTM password`
3. On the ESP32-S2, scan for ap's : `scan` : you should see the ESP32-C3 ap
4. On the ESP32-S2, Initiate the ftm process : `ftm -I -s FTM` : you should get distance estimate, instead you get an error.

### Code to reproduce this issue

Simply clone the "ftm" example.

## Debug Logs

```
ftm> scan
I (2451499) ftm_station: sta start to scan
I (2453599) ftm_station: [FTM][rssi=-45][FTM Responder]
I (2453619) ftm_station: sta scan done
ftm> ftm -I -s FTM
I (2654999) ftm_station: Requesting FTM session with Frm Count - 32, Burst Period - 200mSec (0: No Preference)
W (2654999) wifi:Starting FTM session with 7c:df:a1:42:f2:1d in 0.82 Sec
W (2655009) wifi:Mode: non-ASAP, Bursts: 8, FTM's per burst: 4, Burst Period: 200mSec, Burst Duration: 32000uSec
ftm> E (2656529) wifi:No valid FTM Measurements found!
I (2656529) ftm_station: FTM procedure with Peer(7c:df:a1:42:f2:1d) failed! (Status - 4)
```

nachiketkukade, 2021-09-29:
@Sch-LikA, do you have a ESP32-C3 (rev 3) module with you? Revision 2 and older modules are no longer supported with latest IDF v4.3.1.
Also we're not able to observe this issue on the Rev 3 boards. It could be older code related or module related, in either case moving to the Rev 3 module is a good idea.

AxelLin, 2021-09-29:
> @Sch-LikA, do you have a ESP32-C3 (rev 3) module with you? Revision 2 and older modules are no longer supported with latest IDF v4.3.1. 

Which commit removes rev < 3 ?
Below config is available in v4.3 and master branches:
https://github.com/espressif/esp-idf/blob/release/v4.3/components/esp32c3/Kconfig#L26-L42

nachiketkukade, 2021-09-29:
You're right, I wasn't aware this option was configurable. Using it and testing on Rev 1, I can see the issue. The sniffer shows that FTM measurement frames have zero for timestamps. The issue is that ESP32-C3 modules older than Rev 3 do not have the necessary ROM code to support FTM, so the values come out as zero. For ESP32-S2 this code is not in ROM so the support became available with latest code. So in conclusion, this goes back to my previous suggestion, that upgrading to Rev3 will be a good idea, hope it's clear, thanks.

Sch-LikA, 2021-09-30:
> You're right, I wasn't aware this option was configurable. Using it and testing on Rev 1, I can see the issue. The sniffer shows that FTM measurement frames have zero for timestamps. The issue is [...]

OK that is clear then. Too bad. I tried ordering some new ones but they are not deliverable anymore, probably due to global shortage. Too bad.

closed 2021-09-30 by Sch-LikA