espressif/esp-idf #7579: WiFi station crash when doing start/stop sequence

stefansobol, 2021-09-17:
Hi,
Long story short: I'm working on a project and I'm using a NodeMCU DevKitC v4 device and I need to use mqtt/lwip/wifi stack. I'm connecting and disconnecting periodically and I found that sometimes the function "esp_netif_receive" is sometimes called before esp_wifi_register_if_rxcb that will correctly set the s_wifi_netifs[WIFI_IF_STA] in file components/esp_wifi/src/wifi_netif.c

Please find bellow my log and some comments. Please note that the log time can be ignored since I'm heavily logging to find the cause of the crash.

let me know if more info required.

I (28) boot: ESP-IDF 4.3.0 2nd stage bootloader
I (28) boot: compile time 15:18:18
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (75) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (83) boot:  2 app0             OTA app          00 10 00010000 00300000
I (90) boot:  3 config           Unknown data     01 82 00310000 000f0000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 1, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=39a30h (236080) map
I (198) esp_image: segment 1: paddr=00049a58 vaddr=3ffb0000 size=044f0h ( 17648) load
I (205) esp_image: segment 2: paddr=0004df50 vaddr=40080000 size=020c8h (  8392) load
I (209) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=d9be8h (891880) map
I (517) esp_image: segment 4: paddr=00129c10 vaddr=400820c8 size=14310h ( 82704) load
I (551) esp_image: segment 5: paddr=0013df28 vaddr=50000000 size=00010h (    16) load
I (564) boot: Loaded app from partition at offset 0x10000
I (564) boot: Disabling RNG early entropy source...
I (575) cpu_start: Pro cpu up.
I (575) cpu_start: Starting app cpu, entry point is 0x40081974
I (561) cpu_start: App cpu up.
I (592) cpu_start: Pro cpu start user code
I (592) cpu_start: cpu freq: 160000000
I (592) cpu_start: Application information:
I (597) cpu_start: Project name:     firmware
I (602) cpu_start: App version:      0.0.1.0
I (607) cpu_start: Compile time:     Sep 16 2021 15:18:03
I (613) cpu_start: ELF file SHA256:  15b37092ddb5d405...
I (619) cpu_start: ESP-IDF:          4.3.0
I (624) heap_init: Initializing. RAM available for dynamic allocation:
I (631) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (637) heap_init: At 3FFB9BD8 len 00026428 (153 KiB): DRAM
I (643) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (649) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (656) heap_init: At 400963D8 len 00009C28 (39 KiB): IRAM
I (663) spi_flash: detected chip: generic
I (667) spi_flash: flash io: dio
I (677) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

----------------------------------------
Comment: connection triggered here

I (31441) wifi:wifi driver task: 3ffbf380, prio:23, stack:6656, core=0
I (31441) system_api: Base MAC address is not set
I (31441) system_api: read default base MAC address from EFUSE
I (31461) wifi:wifi firmware version: c7d0450
I (31461) wifi:wifi certification version: v7.0
I (31461) wifi:config NVS flash: enabled
I (31461) wifi:config nano formating: disabled
I (31471) wifi:Init data frame dynamic rx buffer num: 32
I (31471) wifi:Init management frame dynamic rx buffer num: 32
I (31471) wifi:Init management short buffer num: 32
I (31481) wifi:Init static tx buffer num: 16
I (31481) wifi:Init static rx buffer size: 1600
I (31491) wifi:Init static rx buffer num: 10
I (31501) wifi:Init dynamic rx buffer num: 32
I (31501) wifi_init: rx ba win: 6
I (31501) wifi_init: tcpip mbox: 32
I (31511) wifi_init: udp mbox: 6
I (31511) wifi_init: tcp mbox: 6
I (31511) wifi_init: tcp tx win: 5744
I (31511) wifi_init: tcp rx win: 5744
I (31521) wifi_init: tcp mss: 1440
I (31521) wifi_init: WiFi IRAM OP enabled
I (31541) wifi_init: WiFi RX IRAM OP enabled

----------------------------------------
Comment: log esp_netif pointer received by esp_netif_init_configuration (file components/esp_netif/lwip/esp_netif_lwip.c)
esp_netif = 0x3ffc2010

I (31541) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (31661) wifi:mode : sta (9c:9c:1f:e9:7f:cc)
I (31661) wifi:enable tsf
I (31671) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (32401) wifi:state: init -> auth (b0)
I (32431) wifi:state: auth -> assoc (0)
I (32451) wifi:state: assoc -> run (10)
I (32551) wifi:connected with Fastest, aid = 2, channel 1, 40U, bssid = 68:ff:7b:a5:a9:8c
I (32561) wifi:security: WPA2-PSK, phy: bgn, rssi: -75
I (32561) wifi:pm start, type: 1

I (32581) wifi:AP's beacon interval = 102400 us, DTIM period = 1

----------------------------------------
Comment: log esp_netif pointer received by esp_netif_init_configuration (file components/esp_netif/lwip/esp_netif_lwip.c)
esp_netif = 0x3ffc2010

----------------------------------------
Comment: esp_wifi_register_if_rxcb calls esp_wifi_internal_reg_rxcb (file components/esp_wifi/src/wifi_netif.c)
esp_wifi_internal_reg_rxcb

W (32801) wifi:<ba-add>idx:0 (ifx:0, 68:ff:7b:a5:a9:8c), tid:0, ssn:2, winSize:64
I (33611) esp_netif_handlers: sta ip: 192.168.100.85, mask: 255.255.255.0, gw: 192.168.100.1

----------------------------------------
Comment: trigger disconnect

I (38431) APP: MQTT stop
I (38431) MQTT_CLIENT: Client asked to disconnect
I (38611) wifi:state: run -> init (0)
I (38621) wifi:pm stop, total sleep time: 8715344 us / 59036953 us

W (38621) wifi:<ba-del>idx
I (38631) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1

----------------------------------------
Comment: esp_wifi_clear_default_wifi_driver_and_handlers removes s_wifi_netifs[STA] (file components/esp_wifi/src/wifi_default.c)
s_wifi_netifs[i] = NULL

I (38641) wifi:flush txq
I (38641) wifi:stop sw txq
I (38641) wifi:lmac stop hw txq
I (38641) wifi:Deinit lldesc rx mblock:10

----------------------------------------
Comment: connection triggered here

I (68401) wifi:wifi driver task: 3ffdbcd4, prio:23, stack:6656, core=0
I (68401) wifi:wifi firmware version: c7d0450
I (68401) wifi:wifi certification version: v7.0
I (68401) wifi:config NVS flash: enabled
I (68411) wifi:config nano formating: disabled
I (68421) wifi:Init data frame dynamic rx buffer num: 32
I (68421) wifi:Init management frame dynamic rx buffer num: 32
I (68431) wifi:Init management short buffer num: 32
I (68431) wifi:Init static tx buffer num: 16
I (68431) wifi:Init static rx buffer size: 1600
I (68431) wifi:Init static rx buffer num: 10
I (68441) wifi:Init dynamic rx buffer num: 32
I (68451) wifi_init: rx ba win: 6
I (68451) wifi_init: tcpip mbox: 32
I (68461) wifi_init: udp mbox: 6
I (68461) wifi_init: tcp mbox: 6
I (68461) wifi_init: tcp tx win: 5744
I (68471) wifi_init: tcp rx win: 5744
I (68471) wifi_init: tcp mss: 1440
I (68471) wifi_init: WiFi IRAM OP enabled
I (68481) wifi_init: WiFi RX IRAM OP enabled

----------------------------------------
Comment: log esp_netif pointer received by esp_netif_init_configuration (file components/esp_netif/lwip/esp_netif_lwip.c)
esp_netif = 0x3ffbf49c

I (68491) wifi:mode : sta (9c:9c:1f:e9:7f:cc)
I (68491) wifi:enable tsf
I (68511) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (69241) wifi:state: init -> auth (b0)
I (69251) wifi:state: auth -> assoc (0)
I (69271) wifi:state: assoc -> run (10)
I (69381) wifi:connected with Fastest, aid = 2, channel 1, 40U, bssid = 68:ff:7b:a5:a9:8c
I (69391) wifi:security: WPA2-PSK, phy: bgn, rssi: -78
I (69391) wifi:pm start, type: 1

I (69401) wifi:AP's beacon interval = 102400 us, DTIM period = 1

----------------------------------------
Comment: wifi_sta_receive called here (file components/esp_wifi/src/wifi_netif.c), but s_wifi_netifs[WIFI_IF_STA] still has the old value (0x3ffc2010), because probably wifi_default_action_sta_connected (file components/esp_wifi/src/wifi_default.c) still in progress and call to esp_wifi_register_if_rxcb was not completed

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Setting breakpoint at 0x40080400 and returning...

david-cermak, 2021-09-24:
Hi @stefansobol 

Thanks for the report. It seems we missed deinitialization of these statics in [`esp_wifi_destroy_if_driver()`](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/src/wifi_netif.c#L91-L93).

But, could you please share some portions your init/start/stop code? I would say it must be very hard to hit this issue if we init and start the netif *before* `esp_wifi_init()` and unregister and destroy it *after* the wifi itself.

stefansobol, 2021-09-28:
Hi @david-cermak,
Here is my simplified init/start/stop sequence:

Init (run once):
```
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
esp_wifi_init(&cfg);
esp_wifi_set_storage(WIFI_STORAGE_RAM);
esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &espEventHandler, this, &_esp_event_handler);
```

Start:
```
esp_wifi_set_mode(WIFI_MODE_STA);
wifi_config_t wifi_config = {};
wifi_config.sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;
wifi_config.sta.pmf_cfg.capable = true;
wifi_config.sta.pmf_cfg.required = false;
std::strcpy(reinterpret_cast<char *>(wifi_config.sta.ssid), ssid);
std::strcpy(reinterpret_cast<char *>(wifi_config.sta.password), password);
esp_wifi_set_config(WIFI_IF_STA, &wifi_config);
esp_netif_config_t netif_config = ESP_NETIF_DEFAULT_WIFI_STA();
esp_net_if = esp_netif_new(&netif_config);
esp_netif_attach_wifi_station(esp_net_if);
esp_wifi_set_default_wifi_sta_handlers();
esp_wifi_start();
```

Stop:
```
esp_wifi_stop();
esp_wifi_clear_default_wifi_driver_and_handlers(esp_net_if);
esp_netif_destroy(esp_net_if);
```

As you can see I was trying to create a new net_if every time Start is called and destroy it at Stop. My fix was to create the net_if in the Init function and destroy it in my Release function (created only once). But is clear to me that, as you stated, that is something missing from the deinitialization of the net if, or there is a race when starting a new interface.

david-cermak, 2021-09-30:
Thanks for sharing the code snippets.
This WIP patch should fix the race condition:
```patch
diff --git a/components/esp_wifi/src/wifi_netif.c b/components/esp_wifi/src/wifi_netif.c
index ba92759b4f4..0ac08d8e1d3 100644
--- a/components/esp_wifi/src/wifi_netif.c
+++ b/components/esp_wifi/src/wifi_netif.c
@@ -89,6 +81,11 @@ static esp_err_t wifi_driver_start(esp_netif_t * esp_netif, void * args)
 
 void esp_wifi_destroy_if_driver(wifi_netif_driver_t h)
 {
+    if (h) {
+        esp_wifi_internal_reg_rxcb(h->wifi_if, NULL);  // ignore the potential error
+                                                       // as the wifi might have been already uninitialized
+        s_wifi_netifs[h->wifi_if] = NULL;
+    }
     free(h);
 }
 ```

stefansobol, 2021-10-04:
I cannot reproduce it anymore. I think this patch works.
Thanks.

david-cermak, 2021-10-20:
Hi @stefansobol 

Yes this will be fixed in future releases ( both `v4.4` and `v5.0`). It hasn't been decided though, if this patch will be applied or if we fix it differently. The preliminary fix looks as per below:
[wifi-Fix-race-conditon-invoking-invalid-callback-on-.patch.txt](https://github.com/espressif/esp-idf/files/7381859/wifi-Fix-race-conditon-invoking-invalid-callback-on-.patch.txt)

closed 2021-11-25 by espressif-bot

AxelLin, 2021-12-27:
v4.3.2 still does not include this fix.

Alvin1Zhang, 2022-01-04:
@AxelLin Thanks for the pointer, we have back ported the fix to release/4.3, will sync up once the fix is available on GitHub.

AxelLin, 2022-01-25:
> @AxelLin Thanks for the pointer, we have back ported the fix to release/4.3, will sync up once the fix is available on GitHub.

@Alvin1Zhang @david-cermak 
This is still broken in v4.3 now.
(There are several updates on v4.3 branch recently but does not include this fix.)

AxelLin, 2022-02-20:
@Alvin1Zhang @david-cermak 
v4.3 still does not include this fix. (This is the third time to remind you guys about this missing fix. What's wrong with the backport process?)

Alvin1Zhang, 2022-02-21:
@AxelLin Thanks for the notes. The fix has been back ported to release/4.3, and is now under sync to GitHub, will update once the fix is available, thanks.