espressif/esp-idf #7580: Flash writes can clobber RTOS base priority (uxBasePriority) (with analysis)

themadsens, 2021-09-17:
## Environment

- Development Kit:     custom
- Module or chip used:  ESP32
- IDF version v3.2.2 + master
 
## Problem Description

The assigned priorities of tasks shifts after creation (in an upward fashion)

### Expected Behavior

Initial priorities of tasks should stick

### Actual Behavior

They don't

### Bug description

The analysis pointed toward uxBasePriority in the RTOS task TCB being overwritten.

Setting a data watchpoint with esp_set_watchpoint() pointed towards spi_flash_disable_interrupts_caches_and_other_cpu():

It obtains current task priority which may be elevated from tasks wanting locks we are holding (vTaskPriorityInherit() etc.)
It then elevates priority to max and finally restores both current **and** base priority with vTaskPrioritySet(old_prio)

## Fix

See associated PR

themadsens, 2021-09-17:
After reading the PR guidelines, I decided to just post a patch here for consideration / inspiration
```
diff --git a/components/freertos/tasks.c b/components/freertos/tasks.c
index b53fe5c75..07176905f 100644
--- a/components/freertos/tasks.c
+++ b/components/freertos/tasks.c
@@ -1581,6 +1581,42 @@ static void prvAddNewTaskToReadyList( TCB_t *pxNewTCB, TaskFunction_t pxTaskCode
 		return uxReturn;
 	}
 
+	UBaseType_t uxTaskBasePriorityGet( TaskHandle_t xTask )
+	{
+	TCB_t *pxTCB;
+	UBaseType_t uxReturn;
+
+		taskENTER_CRITICAL(&xTaskQueueMutex);
+		{
+			/* If null is passed in here then we are changing the
+			priority of the calling function. */
+			pxTCB = prvGetTCBFromHandle( xTask );
+			#if ( configUSE_MUTEXES == 1 )
+			uxReturn = pxTCB->uxBasePriority;
+			#else
+			uxReturn = pxTCB->uxPriority;
+			#endif
+		}
+		taskEXIT_CRITICAL(&xTaskQueueMutex);
+
+		return uxReturn;
+	}
+
+	void uxTaskBasePrioritySet( TaskHandle_t xTask, UBaseType_t prio )
+	{
+#if ( configUSE_MUTEXES == 1 )
+	TCB_t *pxTCB;
+		taskENTER_CRITICAL(&xTaskQueueMutex);
+		{
+			/* If null is passed in here then we are changing the
+			priority of the calling function. */
+			pxTCB = prvGetTCBFromHandle( xTask );
+            pxTCB->uxBasePriority = prio;
+		}
+		taskEXIT_CRITICAL(&xTaskQueueMutex);
+#endif
+	}
+
 #endif /* INCLUDE_uxTaskPriorityGet */
 /*-----------------------------------------------------------*/
 
diff --git a/components/spi_flash/cache_utils.c b/components/spi_flash/cache_utils.c
index 56038de37..c57268734 100644
--- a/components/spi_flash/cache_utils.c
+++ b/components/spi_flash/cache_utils.c
@@ -90,6 +90,8 @@ void IRAM_ATTR spi_flash_op_block_func(void* arg)
     xTaskResumeAll();
 }
 
+extern UBaseType_t uxTaskBasePriorityGet( TaskHandle_t xTask );
+extern void uxTaskBasePrioritySet( TaskHandle_t xTask, UBaseType_t prio );
 void IRAM_ATTR spi_flash_disable_interrupts_caches_and_other_cpu()
 {
     spi_flash_op_lock();
@@ -113,6 +115,7 @@ void IRAM_ATTR spi_flash_disable_interrupts_caches_and_other_cpu()
         // Temporarily raise current task priority to prevent a deadlock while
         // waiting for IPC task to start on the other CPU
         int old_prio = uxTaskPriorityGet(NULL);
+        int old_prio_base = uxTaskBasePriorityGet(NULL);
         vTaskPrioritySet(NULL, configMAX_PRIORITIES - 1);
         // Signal to the spi_flash_op_block_task on the other CPU that we need it to
         // disable cache there and block other tasks from executing.
@@ -127,6 +130,9 @@ void IRAM_ATTR spi_flash_disable_interrupts_caches_and_other_cpu()
         vTaskSuspendAll();
         // Can now set the priority back to the normal one
         vTaskPrioritySet(NULL, old_prio);
+        if (old_prio_base != old_prio) {
+            uxTaskBasePrioritySet(NULL, old_prio_base);
+        }
         // This is guaranteed to run on CPU <cpuid> because the other CPU is now
         // occupied by highest priority task
         assert(xPortGetCoreID() == cpuid);

```

AxelLin, 2021-11-04:
@Dazza0 

I'm wondering what would be the user visible impact of this issue.
Can you confirm this issue?

AxelLin, 2022-02-17:
@Dazza0 

I can confirm this bug on esp-idf v4.3.2-280-gbe19818db7.
I use vTaskList to show my task priority of all tasks.
Normally, the sys_evt task has priority 20 but after boot for some time I found the priority becomes 23.
My debug shows when sys_evt task was created, the priority is indeed 20.
But after spi_flash_disable_interrupts_caches_and_other_cpu is called, sys_evt priority may change.
ps. sys_evt is just an example, it also impacts priority of other tasks in my application.

Also confirm applying above changes fixes the task priority issue.

AxelLin, 2022-02-21:
BTW, this issue does not happen if CONFIG_FREERTOS_UNICORE=y.

AxelLin, 2022-04-08:
@Alvin1Zhang 
There is no response regarding this bug. What's your suggestion?

AxelLin, 2022-05-11:
This bug report is clear and even with a proposed fix, but noone steps in to review/fix it. Actually, no response at all.
It has been reported for very long time, can someone help to fix this issue? @Alvin1Zhang @igrr

Alvin1Zhang, 2022-05-11:
@AxelLin Thanks for the notes, sorry for the very slow turnaround, will check and provide feedback.

AxelLin, 2022-06-13:
> @AxelLin Thanks for the notes, sorry for the very slow turnaround, will check and provide feedback.

Even after waiting for yet another month, still no feedback at all.

AxelLin, 2022-07-05:
Is there any progress of fixing this issue?  It's really has been a long wait...
@Alvin1Zhang @Dazza0

KonstantinKondrashov, 2022-07-14:
Hi @AxelLin!
There was a discussion about exactly this case in freertos https://forums.freertos.org/t/bug-report-for-uxtaskpriorityget/13357. 
They disagree to change the freertos APIs because it might break inheritance/disinheritance logic.

One way to resolve this issue is to take the priority of task before a mutex from spi_flash_op_lock will be take.
[19037.zip](https://github.com/espressif/esp-idf/files/9109025/19037.zip)

But might be we can find another way to fix it.

closed 2022-09-05 by espressif-bot

KaeLL, 2022-09-05:
Will this be backported?

AxelLin, 2022-09-28:
@KonstantinKondrashov 
Will you fix this for v4.3?  The issue was reported on v4.3 quite long time ago: https://github.com/espressif/esp-idf/issues/7580#issuecomment-1042476721

Dazza0, 2022-09-29:
Hi @AxelLin, v4.3 has entered its [maintenance period since June 2022](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#support-periods), thus only fixes for "high severity issues or security issues" will be backported.

We would prefer if users could upgrade to v4.4 to receive this fix since this bug isn't a high sevrity issue. However, if this bug is a major issue for the user (e.g., something that stops production), we can also consider backporting this to v4.3, or giving a v4.3 based patch to the user.

@AxelLin  Please let me know if those options are OK with you.

Dazza0, 2022-09-29:
@AxelLin OK, the fix will be backported to v4.3