espressif/esp-idf #7581: espcoredump.py: AttributeError: 'EspGDB' object has no attribute 'p'

chipweinberger, 2021-09-17:
**ESP-IDF:** 4.3.1
**OS:** MacOS 11.5
**Board:** ESP32s2

When trying to run this command to look at the core dump from flash:

`python espcoredump.py --port /dev/cu.SLAB_USBtoUART --baud 921600 info_corefile /jamcorder/firmware/jamcorder_app/build/jamcorder_app.elf`

I get this error:

```
espcoredump.py v0.4-dev
INFO: Invoke parttool to read image.
INFO: esptool.py v3.1-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting.....
Detecting chip type... ESP32-S2
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:06:7d:b6
Uploading stub...
Running stub...
Stub running...
3072 (100 %)
3072 (100 %)
Read 3072 bytes at 0x8000 in 0.3 seconds (87.7 kbit/s)...
Hard resetting via RTS pin...
esptool.py v3.1-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP32-S2
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:06:7d:b6
Uploading stub...
Running stub...
Stub running...
65536 (100 %)
65536 (100 %)
Read 65536 bytes at 0xc000 in 5.8 seconds (89.7 kbit/s)...
Hard resetting via RTS pin...
Running /usr/local/opt/python@3.9/bin/python3.9 /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py --port /dev/cu.SLAB_USBtoUART read_flash 32768 3072 /var/folders/kc/8ph49dv51xg62gfz9z77dp9w0000gn/T/tmpigovj7v8...
Running /usr/local/opt/python@3.9/bin/python3.9 /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esptool_py/esptool/esptool.py --port /dev/cu.SLAB_USBtoUART read_flash 49152 65536 /var/folders/kc/8ph49dv51xg62gfz9z77dp9w0000gn/T/tmpgr49ahdp...
Read partition 'coredump' contents from device at offset 0xc000 to file '/var/folders/kc/8ph49dv51xg62gfz9z77dp9w0000gn/T/tmpgr49ahdp'

===============================================================
==================== ESP32 CORE DUMP START ====================
Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/espcoredump/espcoredump.py", line 295, in <module>
    info_corefile()
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/espcoredump/espcoredump.py", line 123, in info_corefile
    gdb = EspGDB(args.gdb, [rom_sym_cmd], core_filename, args.prog, timeout_sec=args.gdb_timeout_sec)
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/espcoredump/corefile/gdb.py", line 46, in __init__
    self.p = GdbController(gdb_path=gdb_path, gdb_args=gdb_args)
TypeError: __init__() got an unexpected keyword argument 'gdb_path'
Exception ignored in: <function EspGDB.__del__ at 0x103810940>
Traceback (most recent call last):
  File "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/espcoredump/corefile/gdb.py", line 56, in __del__
    self.p.exit()
AttributeError: 'EspGDB' object has no attribute 'p'
charlesweinberger@Chips-Macbook-Pro espcoredump % python espcoredump.py --port /dev/cu.SLAB_USBtoUART --baud 921600 info_corefile /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_app/build/jamcorder_app.elf
```

chipweinberger, 2021-09-17:
I've also tried reading the coredump partition myself manually, and then invoking:

`python espcoredump.py info_corefile --core /Users/charlesweinberger/Downloads/coredump-3.elf --core-format elf /jamcorder/firmware/jamcorder-firmware/jamcorder_app/build/jamcorder_app.elf`

or

`python espcoredump.py info_corefile --core /Users/charlesweinberger/Downloads/coredump-3.elf --core-format raw /jamcorder/firmware/jamcorder-firmware/jamcorder_app/build/jamcorder_app.elf`

But I get the same issue.

My partition table looks like this:

```
# Name,          Type, SubType,  Offset,   Size,  Flags
otadata,         data, ota,      ,           8K
phy_init,        data, phy,      ,           4K
coredump,        data, coredump, ,          64K
hardware,        data, 99,,           8K
nvs,             data, nvs,      ,         128K
nvs_key,         data, nvs_keys, ,           4K
jamcorder_app,    app, ota_0,    ,        2788K
factory_app,      app, ota_1,    ,         900K
factory_data,    data, 99,,           8K
extra1,          data, 99,,           8K
extra2,          data, 99,,           8K
```

And these are me sdkconfig settings:
[image]

chipweinberger, 2021-09-17:
I've also read through this issue thread already: https://github.com/espressif/esp-idf/issues/7283

chipweinberger, 2021-09-17:
Looks like this commit to pygdbmi might have broken it: 

https://github.com/cs01/pygdbmi/commit/449080a084583dc6ee6fd066d453072cf1ed065f#diff-e29bd3c69c0c2b3f6f7c89a6ea177b322af7411076e74606faddf68573a7630d

chipweinberger, 2021-09-17:
Fixed!!

needed to run:

```
pip3 uninstall pygdbmi
pip3 install pygdbmi==0.9.0.3  
```

Is there anyway ESP-IDF can be more forceful about the versions it requires? There should be a check in espcoredump that the version is the tested and known working version

dobairoland, 2021-09-18:
Hi @chipweinberger.

currently `pygdbmi<=0.9.0.2` (https://github.com/espressif/esp-idf/blob/release/v4.3/requirements.txt) is required. If you use the `install.sh` and work in an environment set up with `. export.sh` then right version should be available and enforced.

chipweinberger, 2021-09-21:
I suppose thats fair. I just around IDF versions quite a bit. somehow lost that enforcement

closed 2021-09-22 by dobairoland