espressif/esp-idf #7583: Cannot set console baud rate in sdkconfig.defaults

davidzuhn, 2021-09-19:
## Environment

- Development Kit:      TinyPICO 
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-3042-g220590d599 | v4.3.1
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:    USB

## Problem Description

I cannot set the console baud rate in the sdkconfig.defaults file.   

Using the console example application (standalone version can be found at https://github.com/davidzuhn/esp-idf-console-bug-example), I'm trying to set the console speed to 230400 in the sdkconfig.defaults file.   Using ESP-IDF v4.2, this example results in an sdkconfig file with an entry 'CONFIG_ESP_CONSOLE_UART_BAUDRATE=230400'.   Using ESP-IDF v4.3.1 or v4.4 (master), the sdkconfig file generated contains 'CONFIG_ESP_CONSOLE_UART_BAUDRATE=115200'.

In the sdkconfig.defaults file:

```
CONFIG_ESPTOOLPY_MONITOR_BAUD_230400B=y
CONFIG_ESPTOOLPY_MONITOR_BAUD=230400
CONFIG_ESP_CONSOLE_UART_BAUDRATE=230400
```

The generated binary uses the value that's in the sdkconfig file.

### Expected Behavior

```
$ grep CONSOLE_UART_BAUDRATE sdkconfig
CONFIG_ESP_CONSOLE_UART_BAUDRATE=230400
CONFIG_CONSOLE_UART_BAUDRATE=230400
```

### Actual Behavior

```
$ grep CONSOLE_UART_BAUDRATE sdkconfig
CONFIG_ESP_CONSOLE_UART_BAUDRATE=115200
CONFIG_CONSOLE_UART_BAUDRATE=115200
```
### Steps to reproduce

1. git clone https://github.com/davidzuhn/esp-idf-console-bug-example
2. cd esp-idf-console-bug-example
2. idf.py build

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

https://github.com/davidzuhn/esp-idf-console-bug-example

## Debug Logs

The error is not seen at runtime, except that the baud rate is not the expected value.

## Other items if possible

igrr, 2021-09-20:
I think the issue is that `CONFIG_ESP_CONSOLE_UART_BAUDRATE` option only gets enabled when `CONFIG_ESP_CONSOLE_UART_CUSTOM` is:

https://github.com/espressif/esp-idf/blob/d5f58ab13551cd883e8d8478ba367b6e4543ffec/components/esp_system/Kconfig#L310-L312

So in your sdkconfig.defaults file you need an additional line:
```
CONFIG_ESP_CONSOLE_UART_CUSTOM=y
CONFIG_ESP_CONSOLE_UART_BAUDRATE=230400
```

davidzuhn, 2021-09-21:
After testing, I see that this works.  It is, however, a change in behavior that's not documented in the Programming Guide, and the description for CONFIG_ESP_CONSOLE_UART_BAUDRATE mentions ranges based on PM_ENABLE, but doesn't indicate that the _CUSTOM setting applies to this as well.

ESP-Marius, 2024-03-24:
Yeah, @davidzuhn this `depends on` information is indeed not included in the kconfig  docs. I've added a ticket for this.

Closing this one, as this issue is not really related to just this UART option and the orginal question has been answered.

closed 2024-03-24 by ESP-Marius